fwupd/fwupd #2607: Fwupmgr cannot update latest dell 7390 firmware v 1.7.0

fredoche, 2020-11-19:
**Describe the bug**
I cant update my xps 13 7390 with the latest 1.7.0 firmware
A clear and concise description of what the bug is.
fwupdmgr update yelds an error

**Steps to Reproduce**

```
[root@grey-matter fred]# fwupdmgr update
Devices with no available firmware updates: 
 • Thunderbolt host controller
 • Touchpad
Devices with the latest available firmware version:
 • Micron 2200S NVMe 512GB
Upgrade available for System Firmware from 1.6.0 to 1.7.0
XPS 13 7390 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Téléchargement…          [***************************************]
Erreur lors de la lecture de données du connecteur TLS: Erreur de décodage du paquet TLS reçu.
```

**Expected behavior**
Should correctly download the file and update on reboot

**fwupd version information**
Please provide the version of the daemon and client.
```shell
client version:	1.5.1
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.1

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Fedora 33

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
[root@grey-matter fred]# fwupdmgr get-devices --show-all-devices
XPS 13 7390
│
├─Thunderbolt host controller:
│     Device ID:          bdf5329a388f2b37c11ab55433343b83ee6fbd6c
│     Résumé:             Unmatched performance for high-speed I/O
│     Version actuelle:   41.02
│     Vendor:             Dell (TBT:0x00D4)
│     GUIDs:              83f70f65-8de5-5520-93e2-a569615030bb ← THUNDERBOLT\VEN_00D4&DEV_0962&REV_00
│                         b067bd0b-045e-5b00-9948-f7c98fef1f72 ← THUNDERBOLT\VEN_00D4&DEV_0962
│                         7d538854-204d-51b2-8f9d-1fe881c70200 ← TBT-00d40962-native
│                         8a1cafa7-2d8b-5641-ba4a-140a6497a5e9 ← TBT-00d40962-native-controller0-0
│     Device Flags:       • Périphérique interne
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─Intel(R) Core™ i7-10510U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Version actuelle:   0x000000d6
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         d5092916-1868-5182-8c70-12f83dab6372 ← CPUID\PRO_0&FAM_06&MOD_8E&STP_C
│     Device Flags:       • Périphérique interne
│   
├─Micron 2200S NVMe 512GB:
│     Device ID:          c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│     Résumé:             NVM Express Solid State Drive
│     Version actuelle:   22001040
│     Vendor:             Micron Technology Inc (NVME:0x1344)
│     Serial Number:      193022D9A71A
│     GUIDs:              b29d3ccf-500b-5b69-9bb9-e46b4832225f ← STORAGE-DELL-108489
│                         d63070ee-eeed-11e8-8eb2-f2801f1b9fd1
│     Device Flags:       • Périphérique interne
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
│ │   Version actuelle:   1.6.0
│ │   Version minimum:    1.6.0
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              cbe49cca-492b-93e8-b0f4-f693501f271b
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       40d34c18-f6d5-54db-b81d-0aaa600c0540 ← UEFI\RES_{CBE49CCA-492B-93E8-B0F4-F693501F271B}
│ │   Device Flags:       • Périphérique interne
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Résumé:           UEFI Revocation Database
│       Version actuelle: 77
│       Version minimum:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 seconde
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Périphérique interne
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│     
├─TPM:
│ │   Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Version actuelle:   7.2.1.0
│ │   Vendor:             Nuvoton Technology (TPM:NTC)
│ │   GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
│ │                       fac1c8f3-73c8-5cd6-8330-07a3690b5140 ← TPM\VEN_NTC&DEV_0000
│ │                       8d94eb9a-78aa-513d-af50-319e3b9f5210 ← TPM\VEN_NTC&MOD_NPCT75x$rls
│ │                       e9ccc1dc-960a-5e09-afe9-e59a904b776d ← TPM\VEN_NTC&DEV_0000&VER_2.0
│ │                       f1c4013d-3f4c-5623-a40b-81ad96e23860 ← TPM\VEN_NTC&MOD_NPCT75x$rls&VER_2.0
│ │   Device Flags:       • Périphérique interne
│ │ 
│ └─Event Log:
│       Device ID:        58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff ← system-tpm-eventlog
│       Device Flags:     • Périphérique interne
│     
├─Touchpad:
│     Device ID:          b26933c085b020ecf84c490812458523aee710ac
│     Version actuelle:   1.5.2767034
│     Bootloader Version: 54.0.0
│     Vendor:             Synaptics (HIDRAW:0x06CB)
│     GUIDs:              f4384034-9243-5334-8075-a534be913e46 ← HIDRAW\VEN_06CB&DEV_76AF&REV_00
│                         424bd00e-9789-5cdf-a12a-3c81bc4676d6 ← HIDRAW\VEN_06CB&DEV_76AF
│                         140f4458-951b-5bb9-85e2-879bd5b02615 ← SYNAPTICS_RMI\TM3038-003
│                         b29d3c85-cd0e-503e-9c7e-f6731c1eaf2d ← SYNAPTICS_RMI\TM3038
│     Device Flags:       • Périphérique interne
│                         • Updatable
│   
└─UHD Graphics:
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Version actuelle:   02
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              3233ab48-faf4-524a-922a-670b4a02ea1a ← PCI\VEN_8086&DEV_9B41&SUBSYS_10280962&REV_02
                          26e2e22e-e2fa-5a5e-bb6c-dbc5623c1637 ← PCI\VEN_8086&DEV_9B41&SUBSYS_10280962
                          9779d679-0f31-5dd5-b3c2-f6ff275a9ca9 ← PCI\VEN_8086&DEV_9B41&REV_02
                          5589d313-88e5-54d3-8f4c-b2ee3882d4bc ← PCI\VEN_8086&DEV_9B41
      Device Flags:       • Périphérique interne
                          • Cryptographic hash verification is available
    

```

**Additional questions**
- Operating system and version:
Fedora 33
- Have you tried rebooting?
Yes
- Is this a regression?
I dont know
- Are you using an NVMe disk?
Yes
- Is secure boot enabled?
No

fredoche, 2020-11-19:
Follow up of https://github.com/fwupd/firmware-dell/issues/18#

fredoche, 2020-11-19:
wget is having a tough time downloading it as well
`wget https://fwupd.org/downloads/5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell%20XPS%2013%207390%20System%20BIOS_Ver.1.7.0.cab                                          ✔  10003  21:07:29
--2020-11-19 21:07:32--  https://fwupd.org/downloads/5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell%20XPS%2013%207390%20System%20BIOS_Ver.1.7.0.cab
Résolution de fwupd.org (fwupd.org)… 100.20.200.187, 44.242.100.1
Connexion à fwupd.org (fwupd.org)|100.20.200.187|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 12311158 (12M) [application/octet-stream]
Sauvegarde en : « 5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell XPS 13 7390 System BIOS_Ver.1.7.0.cab »

5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d  20%[===========================>                                                                                                              ]   2,39M  66,4KB/s    ds 35s     

2020-11-19 21:08:08 (69,7 KB/s) — Fermeture de la connexion à l’octet 2506752. Nouvel essai.

--2020-11-19 21:08:09--  (essai :  2)  https://fwupd.org/downloads/5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell%20XPS%2013%207390%20System%20BIOS_Ver.1.7.0.cab
Connexion à fwupd.org (fwupd.org)|100.20.200.187|:443… connecté.
requête HTTP transmise, en attente de la réponse… 206 PARTIAL CONTENT
Taille : 12311158 (12M), 9804406 (9,3M) restant [application/octet-stream]
Sauvegarde en : « 5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell XPS 13 7390 System BIOS_Ver.1.7.0.cab »

5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d  44%[++++++++++++++++++++++++++++=================================>                                                                            ]   5,28M  53,4KB/s    ds 3m 3s   

2020-11-19 21:11:13 (16,1 KB/s) — Fermeture de la connexion à l’octet 5533740. Nouvel essai.

--2020-11-19 21:11:15--  (essai :  3)  https://fwupd.org/downloads/5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell%20XPS%2013%207390%20System%20BIOS_Ver.1.7.0.cab
Connexion à fwupd.org (fwupd.org)|100.20.200.187|:443… connecté.
requête HTTP transmise, en attente de la réponse… 206 PARTIAL CONTENT
Taille : 12311158 (12M), 6777418 (6,5M) restant [application/octet-stream]
Sauvegarde en : « 5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell XPS 13 7390 System BIOS_Ver.1.7.0.cab »

5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d 100%[++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++===========================================================================>]  11,74M  73,6KB/s    ds 2m 6s   

2020-11-19 21:13:23 (52,3 KB/s) — « 5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell XPS 13 7390 System BIOS_Ver.1.7.0.cab » sauvegardé [12311158/12311158]
`

hughsie, 2020-11-19:
Could you get logs in en_US please.

fredoche, 2020-11-20:
I retried today with 
```
LANG=C wget https://fwupd.org/downloads/5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell%20XPS%2013%207390%20System%20BIOS_Ver.1.7.0.cab
--2020-11-20 10:07:40--  https://fwupd.org/downloads/5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell%20XPS%2013%207390%20System%20BIOS_Ver.1.7.0.cab
Resolving fwupd.org (fwupd.org)... 44.242.100.1, 100.20.200.187
Connecting to fwupd.org (fwupd.org)|44.242.100.1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12311158 (12M) [application/octet-stream]
Saving to: '5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell XPS 13 7390 System BIOS_Ver.1.7.0.cab.1'

5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d 100%[=========================================================================================================================================>]  11.74M  5.89MB/s    in 2.0s    

2020-11-20 10:07:43 (5.89 MB/s) - '5e154142e2214af942292517b1a547e69edf17abfb0009ac9962ef6813d49f01-Dell XPS 13 7390 System BIOS_Ver.1.7.0.cab.1' saved [12311158/12311158]
```

and it succeeded. The error was probably a transient issue. It's certainy not an issue for me anyway.
Thanks

closed 2020-11-20 by hughsie