fwupd/fwupd #2613: Failed to update metadata from Ubuntu Core18

woodrow-shen, 2020-11-23:
**Describe the question**
This only happened on Ubuntu Core using the confined fwupd (aka uefi-fw-tools)
We've found that executing fwupdmgr refresh will fail to get metadata due the following error:

```
Failed to update metadata for lvfs: signatures were required, but none supplied
```
More debugging info by --verbose:

```
Nov 23 08:03:50 localhost uefi-fw-tools.fwupd[7817]: 08:03:50:0835 FuMain               Called UpdateMetadata(lvfs,9,10)
Nov 23 08:03:50 localhost uefi-fw-tools.fwupd[7817]: 08:03:50:0853 Jcat                 got 1 PKCS7 signatures
Nov 23 08:03:50 localhost uefi-fw-tools.fwupd[7817]: 08:03:50:0860 Jcat                 signature failure: failed to verify data for O=Linux Vendor Firmware Project,CN=LVFS CA: Public key signature verification has failed. [-89]
Nov 23 08:03:50 localhost uefi-fw-tools.fwupd[7817]: 08:03:50:0861 Jcat                 Adding GnuPG public key /root/snap/uefi-fw-tools/x1/etc/pki/fwupd/GPG-KEY-Linux-Vendor-Firmware-Service
Nov 23 08:03:50 localhost uefi-fw-tools.fwupd[7817]: 08:03:50:0861 Jcat                 signature failure: failed to import /root/snap/uefi-fw-tools/x1/etc/pki/fwupd/GPG-KEY-Linux-Vendor-Firmware-Service: Unsupported protocol
```

However, we confirmed that it should work normally few weeks ago, so I'm not sure if any impacts from LVFS.

I also notice Jcat signing from https://github.com/hughsie/libjcat, but still I have no idea what's cause as fwupd snap works fine at the moment.

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.
```shell
$ fwupdmgr --version
client version:	1.5.1-5-g5e0c74c6
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.1-5-g5e0c74c6
```

hughsie, 2020-11-23:
I think you're missing something to do with gpg2, although I'm not sure exactly what.

woodrow-shen, 2020-11-24:
yes, after investigations, we indeed realize that gpg is not provided by core18, so that we have to make gpg available inside snap. I will close this. But I'll comment this into another open issue I raised before as it's related to Ubuntu Core integration.

closed 2020-11-24 by woodrow-shen