fwupd/fwupd #2618: WARNING: This package has not been validated, it may not work properly

haunt98, 2020-11-26:
**Describe the question**
Whenever I use fwupdmgr, It shows:

```txt
WARNING: This package has not been validated, it may not work properly.
```

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.
```shell
$ fwupdmgr --version
client version:	1.5.2
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.2
```

I use Arch Linux if it helps

haunt98, 2020-11-26:
It seem disappear after update to version `fwupd 1.5.2-2` in Arch Linux

closed 2020-11-26 by haunt98

prestevez, 2020-11-26:
I'm getting the same message, though I'm on pop OS and using the latest/stable snap installation of fwupd.

`client version:	1.5.2
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.2
`
Not sure if related, but I also get an error when trying to refresh:

`WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [***************************************]
Invalid compressed data
`
Haven't tried updating to 1.5.2-2 though.

hughsie, 2020-11-26:
> WARNING: This package has not been validated, it may not work properly.

We've done **zero** QA on the snap package, see https://blogs.gnome.org/hughsie/2020/11/23/fwupd-1-5-2/ for why it's marked as non-validated.

> Invalid compressed data

You're probably mixing the snap installed and library versions. See https://github.com/fwupd/fwupd/pull/2615 for the debugging.

prestevez, 2020-11-26:
Thanks @hughsie,

I removed the snap installation and fell back on the system version (1.3.11) and it now worked like a charm. It identified several upgrades that the snap version had missed.