fwupd/fwupd #2619: Installation Error Using Snap Store

onavsaber, 2020-11-26:
**Describe the question**
Trying to install from snap store, I get this error:
```
error: cannot perform the following tasks:
- Run install hook of "fwupd" snap if present (run hook "install": install: cannot stat '/snap/fwupd/2615/lib/modules-load.d/fwupd-msr.conf': No such file or directory)
```
I had fwupd installed. Somehow it got removed from my system after a system update and now it seems that I cannot get it installed again.

SeaMonkey82, 2020-11-28:
Started receiving a bunch of alerts that fwupd-refresh.service was failing in Linux Mint 20.  Tried `sudo snap install fwupd --classic` to see if it would fix the issue and received this error.

superm1, 2020-11-28:
This is a regression from https://github.com/fwupd/fwupd/commit/07d5aef2478a0b5d8573a071a20b13f5ee5b9527

barkanido, 2020-12-02:
Hey Just bumped into that and saw it was recently closed so tried:
`sudo snap info fwupd`
got:
```
...
channels:
  latest/stable:    1.5.2-4-g6f43c50e  2020-11-29 (2618) 31MB classic
  latest/candidate: 1.3.8              2020-02-13 (1625) 24MB classic
  latest/beta:      1.3.7              2020-01-31 (1563) 24MB classic
  latest/edge:      1.5.2-37-g6fca4697 2020-12-01 (2631) 31MB classic
```
then:
`sudo snap install fwupd --classic --channel latest/edge`
but still got:
```
- Run install hook of "fwupd" snap if present (run hook "install": install: cannot stat '/snap/fwupd/2631/lib/modules-load.d/fwupd-msr.conf': No such file or directory)
```

codingforfun, 2020-12-02:
Same issue for me on Ubuntu 16.04. The issue still persists as @barkanido documented.

hsanjuan, 2020-12-02:
Issue also happens in OpenSUSE Tumbleweed.

barkanido, 2020-12-03:
successfully updated `fwupd`
`fwupd (edge) 1.5.2-42-g014e5526 from Richard Hughes installed`
now how do I update the client?
```
➜ sudo fwupdmgr update    
Unsupported daemon version 1.5.2-42-g014e5526, client version is 1.2.10
```

superm1, 2020-12-03:
As mentioned in the wiki: https://github.com/fwupd/fwupd/wiki/fwupd-snap

You should remove the packaged version before installing snap.  If you need to keep side by side, you'll need to use the prefixed commands to the snap namespace.  IE `fwupd.fwupdmgr`

jajpater, 2021-01-12:
Why is this closed? I have the same problem.

superm1, 2021-01-12:
@jajpater please open a new issue with your exact details and we can determine if it's the same problems as discussed here.

zachleigh, 2021-01-21:
@barkanido 
How did you manage to update fwupd?

barkanido, 2021-01-21:
yeah, as the wiki linked explained above

zachleigh, 2021-01-21:
Thanks. The wiki instructions didnt do it for me so wondered if you had an alternate.

dimskiy, 2021-02-07:
The wiki instructions did not help - I was still getting "cannot create regular file '//usr/lib/modules-load.d/fwupd-msr.conf': No such file or directory".
Btw I managed to install fwupd after I manually created empty dir "/usr/lib/modules-load.d"

closed 2021-02-08 by superm1

mailsanchu, 2022-12-15:
I am still seeing the issue. Ubuntu 20.04 
`Linux svarkey-Precision-7760 5.14.0-1054-oem #61-Ubuntu SMP Fri Oct 14 13:05:50 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

superm1, 2022-12-16:
That's this bug: https://github.com/fwupd/fwupd/issues/5122

commit d34aeca2 (2020-11-28): trivial: snap: correct install hook
Fixes: #2619
commit 6f43c50e (2020-11-28): trivial: snap: correct install hook
Fixes: #2619
commit ab9e6560 (2020-12-02): trivial: snap: correct install path for kernel modules harder

The prefix used in the isn't actually used for kernel module
installation.

Fixes: #2619
commit 5bbb0b79 (2021-02-08): only copy files in hook if directory exists (Fixes: #2619)