fwupd/fwupd #2620: Failed to activate service 'org.freedesktop.fwupd': timed out

syscll, 2020-11-27:
**Describe the bug**
Since ~v1.5.0 I've had trouble using `fwupdmgr`:
```
$ fwupdmgr get-updates
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd:
Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

$ systemctl status fwupd
fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: inactive (dead)
       Docs: https://fwupd.org/

Nov 27 13:53:44 t14 systemd[1]: fwupd.service: Failed with result 'timeout'.
Nov 27 13:53:44 t14 systemd[1]: Failed to start Firmware update daemon.
Nov 27 13:54:36 t14 systemd[1]: Starting Firmware update daemon...
Nov 27 13:54:36 t14 fwupd[287500]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
Nov 27 13:54:38 t14 fwupd[287500]: 13:54:38:0175 FuEngine             failed to add USB device 17ef:a395: failed to read SPI chip ID: failed to read chip ID: endpoint stalled or request not supported
Nov 27 13:54:38 t14 fwupd[287500]: 13:54:38:0221 FuPluginDfu          04f2:b6d0 is missing download capability
Nov 27 13:54:38 t14 systemd[1]: Started Firmware update daemon.
```

**Steps to Reproduce**
Attempting to run any update related `fwupdmgr` commands always times out.

**Expected behavior**
I expected update related `fwupdmgr` commands to succeed, and not to see errors from the `fwupd` service.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.2
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20UDCTO1WW
│
├─USB3.1 Hub:
│ │   Device ID:          91c9a317b30acf7c0cd967f2ce6925d518ef19a7
│ │   Summary:            USB 3.x Hub
│ │   Current version:    13.24
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   GUIDs:              227d0f95-a764-598e-85de-3d2bd0a7372f
│ │                       e0c50668-5c2e-54df-8e17-c7c2472f3eaa
│ │                       c6381f16-fe6d-5c45-a4ad-b619af4f0f0e
│ │                       fc6cfb35-d4e9-5467-a014-8863de104969
│ │                       7ae3100f-9b01-571c-9fc2-20cd4544123a
│ │                       ef65710a-3fa0-5330-9a61-5ced5e976be1
│ │                       eaf17a36-88e9-5757-8e99-a14b76fb20f3
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │ 
│ ├─USB3.1 Hub:
│ │     Device ID:        05285288b47a52d357895905caed1c00e37e22f0
│ │     Summary:          USB 3.x Hub
│ │     Current version:  13.23
[... 192 lines cut ...]
│                         31bfb296-9f8b-5519-94ad-04dadf19d202
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          209551b815300e242a5521f54b82e0452f6e81f0
      Current version:    22552
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      GUIDs:              58ac1529-d706-459c-9f98-c711bb777410
                          98a6ece2-0b27-5ab5-b35b-4d8ff4f821da
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

**Additional questions**
- Operating system and version: `Arch Linux 5.9.10-arch1-1`
- Have you tried rebooting? Yes. Does not fix the issue
- Is this a regression? I think so. I've not had this problem before and I've been using fwupd since ~ v0.8.0

syscll, 2020-11-27:
Side note: is the `fwupd` service supposed to be enabled by myself?

```
$ sudo systemctl enable --now fwupd
The unit files have no installation config (WantedBy=, RequiredBy=, Also=,
Alias= settings in the [Install] section, and DefaultInstance= for template
units). This means they are not meant to be enabled using systemctl.

Possible reasons for having this kind of units are:
• A unit may be statically enabled by being symlinked from another unit's
  .wants/ or .requires/ directory.
• A unit's purpose may be to act as a helper for some other unit which has
  a requirement dependency on it.
• A unit may be started when needed via activation (socket, path, timer,
  D-Bus, udev, scripted systemctl call, ...).
• In case of template units, the unit is meant to be enabled with some
  instance name specified.
```

hughsie, 2020-11-27:
What is the output of the `sudo fwupd --verbose` log please?

syscll, 2020-11-27:
I don't have a `fwupd` command installed:

```
$ sudo fwupd --verbose
sudo: fwupd: command not found

$ sudo fwupdmgr --verbose
(fwupdmgr:365311): GLib-DEBUG: 15:00:16.970: setenv()/putenv() are not thread-safe and should not be used after threads are created
Command not found

Use fwupdmgr --help for help
```

hughsie, 2020-11-27:
It's probably in /usr/libexec/fwupd

syscll, 2020-11-27:
I don't have that file either, I have found `/usr/lib/fwupd/fwupd`:

```
$ sudo /usr/lib/fwupd/fwupd --verbose
16:22:32:0572 FuDebug              Verbose debugging enabled (on console 1)
16:22:32:0572 FuConfig             loading config values from /etc/fwupd/daemon.conf
16:22:32:0572 FuRemoteList         loading /usr/share/fwupd/metainfo
16:22:32:0572 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
16:22:32:0572 XbSilo               file: bb595ddb-78ed-4e73-ca43-b3925aa231b9, current:bb595ddb-78ed-4e73-ca43-b3925aa231b9, cached: (null)
16:22:32:0572 XbSilo               loading silo with file contents
16:22:32:0572 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
16:22:32:0572 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
16:22:32:0572 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
16:22:32:0573 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
16:22:32:0573 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
16:22:32:0573 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
16:22:32:0573 FuRemoteList         ignoring unfound remote fwupd
16:22:32:0573 FuRemoteList         ordering lvfs-testing=lvfs+1
16:22:32:0573 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:22:32:0573 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
16:22:32:0573 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
16:22:32:0573 Jcat                 loaded 1 certificates
16:22:32:0573 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
16:22:32:0573 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
16:22:32:0573 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:22:32:0573 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
16:22:32:0573 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
16:22:32:0573 Jcat                 loaded 1 certificates
16:22:32:0573 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
16:22:32:0573 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
16:22:32:0573 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
16:22:32:0577 FuEngine             client certificate exists and working
16:22:32:0577 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
16:22:32:0577 FuHistory            got schema version of 6
16:22:32:0577 FuIdle               setting timeout to 7200s
16:22:32:0577 FuHwids              smbios property Manufacturer=LENOVO
16:22:32:0577 FuHwids              smbios property EnclosureKind=a
16:22:32:0577 FuHwids              smbios property Family=ThinkPad T14 Gen 1
16:22:32:0577 FuHwids              smbios property ProductName=20UDCTO1WW
16:22:32:0577 FuHwids              smbios property ProductSku=LENOVO_MT_20UD_BU_Think_FM_ThinkPad T14 Gen 1
16:22:32:0577 FuHwids              smbios property BiosVendor=LENOVO
16:22:32:0577 FuHwids              smbios property BiosVersion=R1BET58W(1.27 )
16:22:32:0577 FuHwids              smbios property BiosMajorRelease=01
16:22:32:0577 FuHwids              smbios property BiosMinorRelease=1b
16:22:32:0577 FuHwids              smbios property BaseboardManufacturer=LENOVO
16:22:32:0577 FuHwids              smbios property BaseboardProduct=20UDCTO1WW
16:22:32:0581 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
16:22:32:0581 XbSilo               file: 05a643c0-5f3f-270b-812a-367d8d0881e5, current:05a643c0-5f3f-270b-812a-367d8d0881e5, cached: (null)
16:22:32:0581 XbSilo               loading silo with file contents
16:22:32:0581 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
16:22:32:0581 XbSilo               file: cdd1c840-a0ad-2f06-78d9-cbca340b4b11, current:cdd1c840-a0ad-2f06-78d9-cbca340b4b11, cached: (null)
16:22:32:0581 XbSilo               loading silo with file contents
16:22:32:0582 FuEngine             504 components now in silo
16:22:32:0590 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_synaptics_mst.so)
16:22:32:0590 FuPlugin             added udev subsystem watch of drm
16:22:32:0590 FuPlugin             added udev subsystem watch of drm_dp_aux_dev
16:22:32:0590 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_thunderbolt.so)
16:22:32:0590 FuPlugin             added udev subsystem watch of thunderbolt
16:22:32:0590 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_uefi_dbx.so)
16:22:32:0590 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_synaptics_prometheus.so)
16:22:32:0591 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_linux_lockdown.so)
16:22:32:0591 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_dell_dock.so)
16:22:32:0592 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_uefi.so)
16:22:32:0593 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_acpi_dmar.so)
16:22:32:0593 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_pci_mei.so)
16:22:32:0593 FuPlugin             added udev subsystem watch of pci
16:22:32:0593 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_nitrokey.so)
[... cut ...]
16:22:34:0622 XbSilo               ignoring for OR statement: indexed string '181168b9-f5c8-51c4-a8f2-958daab8af2d' was unfound
16:22:34:0633 GLib-GIO             Failed to initialize portal (GMemoryMonitorPortal) for gio-memory-monitor: Not using portals
16:22:34:0634 GLib-GIO             _g_io_module_get_default: Found default implementation dbus (GMemoryMonitorDBus) for ‘gio-memory-monitor’
16:22:34:0653 FuMain               Daemon ready for requests (locale en_US.UTF-8)
16:22:34:0655 FuMain               acquired name: org.freedesktop.fwupd
```

syscll, 2020-11-30:
After doing a bit of digging, I think this is related to the dock I'm using as `fwupd` works as expected when undocker- no timeouts.

syscll, 2020-12-07:
No longer seeing this issue on `Arch Linux 5.9.12-arch1-1`. Everything working as expected, so will close for now.

syscll, 2020-12-08:
Please ignore https://github.com/fwupd/fwupd/issues/2620#issuecomment-739989181, the issue still persists.

syscll, 2021-01-08:
Seems to be working again on `Arch Linux 5.10.5-arch1-1`

closed 2021-01-08 by syscll

adjamhub, 2021-01-20:
I'm having the same problem described here.
My error lists: "fwupd FuPluginDfu 04f2:b6d0 is missing download capability"
Strange thing is that checking that device with lsusb, it returns the integrated camera :)

Bus 001 Device 003: ID 04f2:b6d0 Chicony Electronics Co., Ltd Integrated Camera

So, what?

hughsie, 2021-01-20:
@adjamhub what exact problem are you having? I would guess `missing download capability` is correct for that camera device.

syscll, 2021-01-20:
Sorry to flip flop on this issue, but I am still experiencing it too. It definitely has something to do with having my machine being docked (Lenovo USB-C Dock Gen 2), as I don't get any timeouts when undocked.

adjamhub, 2021-01-20:
Hi Hughsie,
I'm having the same exact problem described here. 
ArchLinux machine (up-to-date).
$ fwupdmgr get-devices
Failed to connect to daemon: Error when calling StartServiceByName per org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

and journalctl -f said:
gen 20 15:32:39 dark dbus-daemon[337]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service' requested by ':1.58' (uid=1000 pid=3874 comm="fwupdmgr get-devices ")
gen 20 15:32:39 dark systemd[1]: Starting Firmware update daemon...
gen 20 15:32:39 dark fwupd[3882]: 14:32:39:0567 FuPluginDfu          04f2:b6d0 is missing download capability
gen 20 15:32:42 dark systemd[1]: fwupd.service: start operation timed out. Terminating.
gen 20 15:32:42 dark systemd[1]: fwupd.service: Failed with result 'timeout'.
gen 20 15:32:42 dark systemd[1]: Failed to start Firmware update daemon.

hughsie, 2021-01-20:
The ` 04f2:b6d0 is missing download capability` text is probably a red herring, the code is literally:

		g_warning ("%04x:%04x is missing download capability",
			   g_usb_device_get_vid (usb_device),
			   g_usb_device_get_pid (usb_device));

Can you get a backtrace of the fwupd daemon when it crashes like that please.

syscll, 2021-01-20:
```
$ journalctl -f
Jan 20 14:56:37 t14 polkitd[1469]: Registered Authentication Agent for unix-process:330108:1697879 (system bus name :1.13279 [/bin/pkttyagent --notify-fd 7 --fallback], object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8)
Jan 20 14:56:37 t14 dbus-daemon[805]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service' requested by ':1.13280' (uid=1000 pid=330108 comm="fwupdmgr update ")
Jan 20 14:56:54 t14 systemd[1]: fwupd.service: start operation timed out. Terminating.
Jan 20 14:56:54 t14 systemd[1]: fwupd.service: Failed with result 'timeout'.
Jan 20 14:56:54 t14 systemd[1]: Failed to start Firmware update daemon.
Jan 20 14:56:54 t14 audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=fwupd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Jan 20 14:56:54 t14 kernel: audit: type=1130 audit(1611154614.586:165): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=fwupd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Jan 20 14:57:02 t14 dbus-daemon[805]: [system] Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
Jan 20 14:57:02 t14 polkitd[1469]: Unregistered Authentication Agent for unix-process:330108:1697879 (system bus name :1.13279, object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8) (disconnected from bus)
```

hughsie, 2021-01-20:
Nah, a backtrace, like:

    sudo gdb --args /usr/libexec/fwupd/fwupd --verbose
    (gdb) r

Then wait for it to crash (e.g. try doing fwupdmgr get-devices in another terminal) or if it "hangs" in a specific point you can do ctrl+c and then this for a backtrace:

    (gdb) t a a bt

syscll, 2021-01-20:
Sorry, hadn't seen your previous comment.

```
$ sudo gdb --args /usr/lib/fwupd/fwupd --verbose
GNU gdb (GDB) 10.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/lib/fwupd/fwupd...
(No debugging symbols found in /usr/lib/fwupd/fwupd)
(gdb) r
Starting program: /usr/lib/fwupd/fwupd --verbose
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
15:07:19:0797 FuDebug              Verbose debugging enabled (on console 1)
15:07:19:0797 FuConfig             loading config values from /etc/fwupd/daemon.conf
[New Thread 0x7ffff6086640 (LWP 342540)]
15:07:19:0798 FuRemoteList         loading /usr/share/fwupd/metainfo
15:07:19:0798 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:07:19:0798 XbSilo               file: e109e377-b317-d09e-5e8d-8945b4d2c2dd, current:e109e377-b317-d09e-5e8d-8945b4d2c2dd, cached: (null)
15:07:19:0798 XbSilo               loading silo with file contents
15:07:19:0799 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
15:07:19:0799 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
15:07:19:0799 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
15:07:19:0799 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
15:07:19:0799 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
15:07:19:0799 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
15:07:19:0799 FuRemoteList         ignoring unfound remote fwupd
15:07:19:0799 FuRemoteList         ordering lvfs-testing=lvfs+1
15:07:19:0799 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:07:19:0799 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
15:07:19:0799 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
15:07:19:0799 Jcat                 loaded 1 certificates
15:07:19:0799 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
15:07:19:0799 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
15:07:19:0799 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:07:19:0799 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
15:07:19:0800 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
15:07:19:0800 Jcat                 loaded 1 certificates
15:07:19:0800 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
15:07:19:0800 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
15:07:19:0800 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
15:07:19:0803 FuEngine             client certificate exists and working
15:07:19:0803 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
15:07:19:0804 FuHistory            got schema version of 6
15:07:19:0804 FuIdle               setting timeout to 7200s
15:07:19:0804 FuHwids              smbios property Manufacturer=LENOVO
15:07:19:0804 FuHwids              smbios property EnclosureKind=a
15:07:19:0804 FuHwids              smbios property Family=ThinkPad T14 Gen 1
15:07:19:0804 FuHwids              smbios property ProductName=20UDCTO1WW
15:07:19:0804 FuHwids              smbios property ProductSku=LENOVO_MT_20UD_BU_Think_FM_ThinkPad T14 Gen 1
15:07:19:0804 FuHwids              smbios property BiosVendor=LENOVO
15:07:19:0804 FuHwids              smbios property BiosVersion=R1BET60W(1.29 )
15:07:19:0804 FuHwids              smbios property BiosMajorRelease=01
15:07:19:0804 FuHwids              smbios property BiosMinorRelease=1d
15:07:19:0804 FuHwids              smbios property BaseboardManufacturer=LENOVO
15:07:19:0804 FuHwids              smbios property BaseboardProduct=20UDCTO1WW
15:07:19:0808 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
15:07:19:0808 XbSilo               file: e1bb06a9-7ef8-85aa-2e0a-56eebf5ed5bd, current:e1bb06a9-7ef8-85aa-2e0a-56eebf5ed5bd, cached: (null)
[... 538 lines cut ...]
15:07:33:0006 FuMain               Called GetPlugins()
15:07:33:0008 FuMain               Called SetFeatureFlags(15)
15:07:33:0009 FuMain               Called GetDevices()
15:07:33:0012 FuMain               Called GetHistory()
15:07:33:0018 FuMain               Called GetRemotes()
[Thread 0x7fffef7fe640 (LWP 342587) exited]
15:07:48:0927 FuMain               Called GetPlugins()
15:07:48:0929 FuMain               Called SetFeatureFlags(15)
15:07:48:0930 FuMain               Called GetDevices()
15:07:48:0932 FuMain               Called GetUpgrades(50efef912094e784a03628350441f6927c4920b0)
[... cut ...]
15:07:48:0936 XbSilo               ignoring for OR statement: indexed string '230c8b18-8d9b-53ec-838b-6cfc0383493a' was unfound
15:07:48:0936 XbSilo               ignoring for OR statement: indexed string '38ad8832-d322-5d73-aff7-77b8ae0f1a6f' was unfound
15:07:48:0936 FuEngine             requirement 1.0.8 ge 1.5.5 -> org.freedesktop.fwupd passed
15:07:48:0936 FuEngine             HWID provided 9625ea0b-77b1-5911-8d90-f646bffbe857
15:07:48:0936 FuEngine             ignoring 0.1.29 == 0.1.29
15:07:48:0936 FuEngine             ignoring 0.1.25 < 0.1.29
```

hughsie, 2021-01-20:
So it worked fine? The only weird thing I saw was:

15:07:19:0908 FuEngine             disabling plugin because: failed to startup using upower: no owner for org.freedesktop.UPower

Is upower not installed?

Also, does the daemon start up with systemd with selinux turned off perhaps?

superm1, 2021-01-20:
I would suggest to add --verbose into the systemd unit, reload systemd units and then try to run it again from there

syscll, 2021-01-20:
Weirdly, it decided to work this time even though it was docked. Will reboot and try again.

I did not have upower installed, but it is now.

I also do not have selinux enabled, or at least, I've not enabled in manually. I don't have any selinux related entries in my kernel params.

edwloef, 2023-11-19:
> After doing a bit of digging, I think this is related to the dock I'm using as `fwupd` works as expected when undocked- no timeouts.

This fixed it for me as well, after the issue started happening to me with fwupd 1.9.8-3 and kernel 6.6.1.arch1-1