fwupd/fwupd #2624: invalid cast from 'GParamGType' to 'FwupdDevice'

artgreen, 2020-11-28:
**Describe the bug**
During boot or when adding/removing thunderbolt 3 dock, this message is seen.  fwupd is reported as failed with systemctl.

**Steps to Reproduce**
Boot or add/remove thunderbolt 3 dock.

**Expected behavior**
No SEGV and daemon remains running.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installation of Pop!_OS and all updates.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```
20TK001FUS
│
├─ThinkPad Thunderbolt 3 Dock:
│ │   Device ID:           ffc1970ee618a65b745eaad8461c0aaa2772d1fe
│ │   Current version:     50.00
│ │   Vendor:              Lenovo (TBT:0x0108)
│ │   GUID:                75039175-c354-5df9-a798-5da41461c007
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Device stages updates
│ │ 
│ ├─USB2.0 Hub:
│ │ │   Device ID:         7622d5fdbf1d1e08138156da7d83bf693986ad16
│ │ │   Summary:           USB 2.x Hub
│ │ │   Current version:   51.13
│ │ │   Vendor:            VIA Labs, Inc. (USB:0x17EF)
│ │ │   Install Duration:  15 seconds
│ │ │   GUIDs:             d5b585c2-e278-56ff-bd47-9bf0ad2bf9a9
│ │ │                      8ee94f0e-9b44-596a-bdd9-6f90401664cc
│ │ │                      0987e3c9-b1ee-5763-ac6e-51329b034e4b
│ │ │                      0488c1ec-e9d6-589f-8620-f0a7426a3c4e
│ │ │   Device Flags:      • Updatable
│ │ │                      • Cryptographic hash verification is available
│ │ │                      • Device stages updates
│ │ │                      • Device can recover flash failures
│ │ │ 
│ │ └─MSP430:
│ │       Device ID:       36cd45c89e19806b1cd99ece89baf4eed34ab921
│ │       Summary:         I²C Dock Management Device
│ │       Current version: 28.1
│ │       Vendor:          VIA Labs, Inc. (USB:0x17EF)
│ │       GUID:            3b09a564-e8a7-5e39-b93b-5b8468e5e3b8
│ │       Device Flags:    • Updatable
│ │     
│ ├─USB3.1 Hub:
│ │ │   Device ID:         48b0e21e27b6bab9c8208a603e5b65d8a42344e5
│ │ │   Summary:           USB 3.x Hub
│ │ │   Current version:   51.13
│ │ │   Vendor:            VIA Labs, Inc. (USB:0x17EF)
│ │ │   Install Duration:  15 seconds
│ │ │   GUIDs:             4ad676ff-8529-5cb0-930d-e3a1da2c60cb
[... 178 lines cut ...]
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           aaa9b84f8cf13c5db4bbc894a23240288b663514
      Current version:     0.1.258
      Minimum Version:     0.1.258
      Vendor:              DMI:LENOVO
      GUID:                3fb9a55d-d7f1-4d1b-b216-74e328e28f51
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update

**Additional questions**
- Operating system and version:
Linux rayon 5.8.0-7630-generic #32~1605108853~20.04~8bcf10e-Ubuntu SMP Wed Nov 11 22:42:16 UTC  x86_64 x86_64 x86_64 GNU/Linux
- Have you tried rebooting?
A thousand times.
- Is this a regression?
Unknown - first time dealing with thunderbolt 3 and this issue.

hughsie, 2020-11-28:
Does the problem disappear with 1.4.x or 1.5.x?

superm1, 2020-11-28:
This feels like a hand install and packaged install both on the same system.  Do you have anything lingering in `/usr/local`?

closed 2021-01-08 by stale[bot]