fwupd/fwupd #2627: bcm57xx: timed out trying to acquire lock #1

zeldin, 2020-11-29:
**Describe the bug**
A clear and concise description of what the bug is.

When installing the custom firmware for NetXtreme BCM5719 Gigabit Ethernet PCIe, I get the error message `timed out trying to acquire lock #1`.  However, if I retry the operation, it goes through with `Successfully installed firmware`.  Also, the firmware was actually installed even when getting the error message.

It seems that I get `Successfully installed firmware` every time that `/usr/libexec/fwupd/fwupd` is already running, and `timed out trying to acquire lock #1` every time it is not.

**Steps to Reproduce**
Steps to reproduce the behavior.

1. Make sure `/usr/libexec/fwupd/fwupd` is not running
2. Run `fwupdmgr reinstall 30fe13b6-aa73-5d8c-a19f-c7b600f0117a`
3. Observe the reinstall complete with the message `timed out trying to acquire lock #1`

**Expected behavior**
A clear and concise description of what you expected to happen.

The reinstall completes with the message `Successfully installed firmware`

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.2
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

```shell
# emerge fwupd
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
WARNING: This package has not been validated, it may not work properly.
T2P9D01 REV 1.00
│
├─NetXtreme BCM5719 Gigabit Ethernet PCIe:
│     Device ID:          7de5ffdca08fa52d95fd4bb42aa5d07a4b35d2dd
│     Current version:    0.4.62
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     Update State:       success
│     Update Error:       timed out trying to acquire lock #1
│     Last modified:      2020-11-29 09:35
│     GUIDs:              ff247529-8b0a-5db7-98ef-8746aa018be6 ← PCI\VEN_14E4&DEV_1657&SUBSYS_14E41981&REV_01
│                         5d8b12bf-1973-58fc-8b31-3e50fe31954d ← PCI\VEN_14E4&DEV_1657&SUBSYS_14E41981
│                         3800738e-1413-5a18-846a-8782f5cc8c44 ← PCI\VEN_14E4&DEV_1657&REV_01
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a ← PCI\VEN_14E4&DEV_1657
│     Device Flags:       • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device will backup firmware before installing
│   
└─Caicos PRO [Radeon HD 7450]:
      Device ID:          4b38a5e5245b7a0ad5adfab23f47009b7be9db3a
      Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
      GUIDs:              286c1b1f-198f-5243-84c1-be41dbe5b78f ← PCI\VEN_1002&DEV_677B&SUBSYS_14628099&REV_00
                          73c7bbd0-a0ff-5c75-b331-470148aef723 ← PCI\VEN_1002&DEV_677B&SUBSYS_14628099
                          d5a6986b-ded8-5b0b-857e-85edaf03bd6d ← PCI\VEN_1002&DEV_677B&REV_00
                          74026e84-990c-5207-8f97-1fd22470be0c ← PCI\VEN_1002&DEV_677B
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
    
```

**Additional questions**
- Operating system and version:
```shell
$ lsb_release -a
LSB Version:	n/a
Distributor ID:	Gentoo
Description:	Gentoo Base System release 2.7
Release:	2.7
Codename:	n/a
$ eselect profile show
Current /etc/portage/make.profile symlink:
  default/linux/powerpc/ppc64/17.0/64bit-userland/desktop
$ uname -a
Linux hakua 5.4.65 #1 SMP Sun Sep 13 17:42:06 CEST 2020 ppc64 POWER9, altivec supported PowerNV T2P9D01 REV 1.00 GNU/Linux
```
- Have you tried rebooting?
No.
- Is this a regression?
Don't know, but probably not.

hughsie, 2020-11-29:
> timed out trying to acquire lock

This means we loaded the device in recovery mode, probably during ->setup. I'm guessing this is wrong and the device is actually in runtime mode but misdetected. I wonder if you could get a --verbose log of the fwupd service when it's autostarted by DBus; to get such a thing would probably involve editing the service file to add `--verbose` and then getting the logs from the journal.

zeldin, 2020-11-29:
When DBus autostarts `fwupd`, it sets stdout and stderr to `/dev/null`, so `--verbose` in the service file doesn't really help.

However, I can reproduce the issue when starting `/usr/libexec/fwupd/fwupd` manually.

* Log for the [first reinstall](https://github.com/fwupd/fwupd/files/5612107/log1.txt), which gives `timed out trying to acquire lock #1`
* Log for the [second reinstall](https://github.com/fwupd/fwupd/files/5612108/log2.txt), which gives `Successfully installed firmware`

You are correct that the first log mentions "recovery", while the second one doesn't.

zeldin, 2020-11-29:
The first difference between the two logs seems to be that after  `prepare -> FuBcm57xxDevice`, log2 has

>  Status:               idle

several times where log1 does not show a status at all.

Then, log1 has

>11:55:54:0257 FuMain               Emitting PropertyChanged('Percentage'='100%')
>11:55:54:0288 FuDevice             using 781a4423b872c17f5e5121dc898238f26bfa96c5 for PCI_SLOT_NAME=0004:01:00.0:recovery
>11:55:54:0506 FuPlugin             running superclassed update_attach(bcm57xx)
>11:55:54:0507 FuEngine             cleanup -> FuBcm57xxDevice:
>NetXtreme BCM5719 Gigabit Ethernet PCIe
>  DeviceId:             7de5ffdca08fa52d95fd4bb42aa5d07a4b35d2dd
>  Status:               device-read

where log2 has

>11:58:27:0180 FuMain               Emitting PropertyChanged('Percentage'='100%')
>11:58:27:0242 FuEngine             Emitting PropertyChanged('Status'='device-busy')
>11:58:27:0242 FuMain               Emitting PropertyChanged('Status'='device-busy')
>11:58:27:0242 FuMain               Emitting PropertyChanged('Percentage'='0%')
> [ ... lots of stuff ... ]
>11:58:32:0335 FuPluginBcm57xx      FW version stage1-0.4.62 NCSI v1.3.12.0
>11:58:32:0336 FuEngine             cleanup -> FuBcm57xxDevice:
>NetXtreme BCM5719 Gigabit Ethernet PCIe
>  DeviceId:             7de5ffdca08fa52d95fd4bb42aa5d07a4b35d2dd
>  Status:               device-busy

Then they are the same again, except for

>11:58:32:0336 FuEngine             Updating NetXtreme BCM5719 Gigabit Ethernet PCIe took 26,614799 seconds

and

>11:58:32:0344 FuEngine             Emitting PropertyChanged('Status'='unknown')
>11:58:32:0344 FuMain               Emitting PropertyChanged('Status'='unknown')

which only appear in log2.

hughsie, 2020-12-04:
> Log for the second reinstall

Woah; I thought fwupd was asking you to reboot after each one?

zeldin, 2020-12-04:
> Woah; I thought fwupd was asking you to reboot after each one?

fwupd is not the boss of me.  :smile:

Besides, rebooting after a botched fw install is not really something you want to do.

hughsie, 2020-12-04:
> fwupd is not the boss of me.

Right, but to reset the APE we need to power down the system (rather than just restart it) -- which means I've done zero testing on duplicate writes I'm afraid.

> Besides, rebooting after a botched fw install is not really something you want to do.

The card is recoverable, even without the tg3 kernel driver recognising it.

zeldin, 2020-12-04:
> which means I've done zero testing on duplicate writes I'm afraid.

Well, the duplicate write is the one that _works_.  Had I rebooted after every attempt I would never have discovered how to get it work (run it two times or more without rebooting or otherwise restarting the fwupd backend). :zany_face:

closed 2021-01-14 by stale[bot]