fwupd/fwupd #2628: Invalid Compressed data error

scoronado12, 2020-11-29:
**Describe the bug**

I am unable to update the firmware metadata on my system.

    scoronado@piensalo-bien  ~  fwupdmgr get-updates
    Firmware metadata has not been updated for 30 days and may not be up to date.

    Update now? (Requires internet connection) [y|N]: y
    Updating lvfs
    Downloading…             [***************************************]
    Invalid compressed data
    ✘ scoronado@piensalo-bien  ~  echo $?
    1

**Steps to Reproduce**
Steps to reproduce the behavior.
Run `fwupdmgr get-updates` from the command line and accept the firmware update metadata

**Expected behavior**
I expected the latest metadata with a clean exit. 

**fwupd version information**
Please provide the version of the daemon and client.

    ✘ scoronado@piensalo-bien  ~  fwupdmgr --version
    client version: 1.5.2
    compile-time dependency versions
        gusb:   0.3.5

    daemon version: 1.5.2

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Installed via Arch Linux `community

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

     ✘ scoronado@piensalo-bien  ~  fwupdmgr get-devices --show-all-devices
    20FL000NUS
    │
    ├─Intel Management Engine:
    │     Device ID:          570d24e21cec6aadc4c7019ea2d7cfc16c22191b
    │     Current version:    184.77.3664
    │     Minimum Version:    184.77.3664
    │     Vendor:             DMI:LENOVO
    │     GUIDs:              a7c68c44-60f0-431c-9242-926912892f28
    │                         aaa760ee-7215-5d8e-8dd3-5c7160bc7bff
    │     Device Flags:       • Internal device
    │                         • Updatable
    │                         • System requires external power source
    │                         • Needs a reboot after installation
    │                         • Device is usable for the duration of the update
    │   
    ├─Intel(R) Core™ i7-6600U CPU @ 2.60GHz:
    │     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
    │     Current version:    0x000000d6
    │     Vendor:             Intel
    │     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
    │                         30249f37-d140-5d3e-9319-186b1bd5cac3
    │                         28d7f31c-1663-5b15-a900-94a71cbcd53f
    │                         e4bc7423-9fec-55a6-8f6c-827c40992491
    │     Device Flags:       • Internal device
    │   
    ├─SAMSUNG MZ7LN512HMJP-000L7:
    │     Device ID:          e421b2fc248391f6fe3e55ddbb3c9043be068bd0
    │     Summary:            ATA Drive
    │     Current version:    MAV02L6Q
    │     Vendor:             Samsung (ATA:0x144D)
    │     GUIDs:              b987c66f-bb70-54a1-b880-e1c41ba2c222
    │                         c9706c50-0f93-5b57-af6e-4e3fd77ea10e
    │                         9aa5c587-9906-564a-b5d6-baffbf019028
    │     Device Flags:       • Internal device
    │                         • Updatable
    │                         • System requires external power source
    │                         • Needs a reboot after installation
    │                         • Device is usable for the duration of the update
    │   
    ├─Skylake GT2 [HD Graphics 520]:
    │     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
    │     Current version:    07
    │     Vendor:             Intel Corporation (PCI:0x8086)
    │     GUIDs:              0efd51bf-82d9-5b36-a627-5f4cb8791aae
    │                         9ab6f3db-aa83-5f84-8301-06ba4459738e
    │                         2f7ef0d3-15f0-571c-b852-de72f1c7c730
    │                         bdcb90a0-e7cd-5d77-995d-d79b7183aa29
    │     Device Flags:       • Internal device
    │                         • Cryptographic hash verification is available
    │   
    ├─System Firmware:
    │ │   Device ID:          3762edb590127a8d7498cf374bd920ce65fc54c2
    │ │   Current version:    0.1.34
    │ │   Minimum Version:    0.1.22
    │ │   Vendor:             LENOVO (DMI:LENOVO)
    │ │   GUIDs:              0b145790-0d11-4dee-9c2a-19ddbcbb200c
    │ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
    │ │                       f8a13b1e-34af-5e8d-8773-4ff345768048
    │ │   Device Flags:       • Internal device
    │ │                       • Updatable
    │ │                       • System requires external power source
    │ │                       • Needs a reboot after installation
    │ │                       • Cryptographic hash verification is available
    │ │                       • Device is usable for the duration of the update
    │ │ 
    │ └─Intel AMT [unprovisioned]:
    │       Device ID:        e2623122c99d58220498aacbfcfdb1baebbae3c5
    │       Summary:          Hardware and firmware technology for remote out-of-band management
    │       Current version:  11.8.77.3664
    │       Bootloader Version:11.8.77.3664
    │       Vendor:           Intel Corporation
    │       GUID:             2800f812-b7b4-2d4b-aca8-46e0ff65814c
    │       Device Flags:     • Internal device
    │     
    └─UEFI Device Firmware:
          Device ID:          3b6d9c846a832bbcf3af788610c2baaea129f9bb
          Current version:    65544
          Minimum Version:    1
          Vendor:             DMI:LENOVO
          GUIDs:              2433c86b-6d72-4330-84ef-1972b1fe6e27
                              dbaa8618-6e64-5ffd-8688-ae3cccb5c93b
          Device Flags:       • Internal device
                                 • Updatable
                                • System requires external power source
                                • Needs a reboot after installation
                                • Device is usable for the duration of the update

**Additional questions**
- Operating system and version: Arch Linux x86_64
- Have you tried rebooting? Yes
- Is this a regression? Yes

scoronado12, 2020-11-29:
I have tried renaming the directory, `/var/lib/fwupd/remotes.d/lvfs` to see if `fwupd` would download a fresh copy in its place, and it did not work.

superm1, 2020-11-29:
Can you provide the output of the remote itself in `/etc/fwupd/remotes.d`?

scoronado12, 2020-11-29:
Thanks for replying.

    [root@piensalo-bien remotes.d]# ls -l
    total 32
    -rw-r--r-- 1 root root 247 Nov 25 16:58 dell-esrt.conf
    -rw-r--r-- 1 root root 283 Nov 25 16:58 fwupd-tests.conf
    -rw-r--r-- 1 root root 327 Nov 20  2019 lvfs.conf
    -rw-r--r-- 1 root root 406 Nov 25 16:58 lvfs.conf.pacnew
    -rw-r--r-- 1 root root 372 Nov 20  2019 lvfs-testing.conf
    -rw-r--r-- 1 root root 361 Nov 25 16:58 lvfs-testing.conf.pacnew
    -rw-r--r-- 1 root root 317 Nov 25 16:58 vendor.conf
    -rw-r--r-- 1 root root 296 Nov 25 16:58 vendor-directory.conf

These `pacnew` files look suspicious, should I remove them?

superm1, 2020-11-29:
I suspect the ones that are not pacnew are the problem actually.. I think you should finish the packaging operation (which probably means renaming those files)

scoronado12, 2020-11-29:
Looks like this was an effect of an upgrade that occurred as we had a brief power surge, not an issue with `fwupd` itself.

Thank you @superm1

closed 2020-11-29 by scoronado12

alxchk, 2020-11-30:
I have same situation (Invalid compressed data), but it is not looks like update issue. 
If I disable Keyring=gpg, then I'm able to update. So It looks like something gpg related. Are there any way to trace this out?

superm1, 2020-11-30:
It's related to some changes recently to allow newer library to work with older daemon.
@alxchk are you on 1.5.2?  Hand install?  Packaged install?  Why didn't your .conf files get updated on upgrades?

alxchk, 2020-11-30:
Yes, it's 1.5.2. I'm on gentoo, and looking to the 1.5.2 ebuild I don't see anything related to updating conf files.
Are there any info what should be done?

superm1, 2020-11-30:
Correct it's not needed.  See:  https://github.com/fwupd/fwupd/commit/d5aab65f30bbc7490240b752dc76d7668aa38f7e#diff-49bf8eeb6b1ed2fbc12bc3420d7d449dcead3118f6f414ce46015f7f291beaf5 for details.

Why didn't the ebuild update the conffile?  Shouldn't it have seen conffile changed and triggered a 3 way merge?

hferreiro, 2020-12-02:
I have the same issue on Fedora 33, and fwupdmgr 1.5.2.

elhossary, 2020-12-02:
I have the same issue as well on Arch, , and fwupdmgr 1.5.2.