fwupd/fwupd #2634: Failed to find module 'platform-integrity'

paulmenzel, 2020-12-02:
On Debian sid/unstable with *fwupd* 1.5.1-5 the Linux kernel module `platform-integraty` is not available.

https://github.com/fwupd/fwupd/blob/13bae744be55d5f901133d957a0f3114b7381024/plugins/platform-integrity/fwupd-platform-integrity.conf#L1

    systemd-modules-load[254]: Failed to find module 'platform-integrity'

Searching the Linux source for that module name came up empty. What Linux Kconfig option needs to be selected?

hughsie, 2020-12-02:
>What Linux Kconfig option needs to be selected

It's not quite upstream yet.

superm1, 2020-12-02:
Maybe we should just not install that configuration for now?  I kind of feel like if > 99.9999% of people will see an error in dmesg we probably don't need to be putting that there.  When the module loads, maybe only start installing it on some newer distros?

hughsie, 2020-12-02:
I think maybe add a -Dplugin_foo and turn it off by default is probably best.

closed 2020-12-03 by hughsie

paulmenzel, 2020-12-03:
Yeah, if it’s not available yet, probably it shouldn’t be installed or a configuration option for the packagers be added.

PR #2635 (2020-12-03): trivial: make platform integrity plugin optional and disabled by default
Fixes: #2634
When it's upstream, we may consider to change this for some distros.

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 5198bdcd (2020-12-03): trivial: make platform integrity plugin optional and disabled by default

Fixes: #2634
When it's upstream, we may consider to change this for some distros.