fwupd/fwupd #2637: WARNING: UEFI ESP partition not detected or configured

codingforfun, 2020-12-03:
**Describe the question**

I’m stuck with the error above and don’t get it working with the proposed solutions. 
I’m on Ubuntu 16.04 using fwupd from snap `latest/stable: 1.5.2-38-gab9e6560`
It used to work a long time, but recently it stopped working. 
I get this warning for `fwupd refresh` or `fwupd update`:
```
root@kairos:/snap/fwupd/2632/etc/fwupd# fwupd.fwupdmgr update
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
```
Most of my upgradeable devices say:
```
Update Error:       Not updatable as UEFI ESP partition not detected
```

I tried the suggestions at `https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found` but it didn’t helped. 
- udisk is installed
- EFI is already in `/etc/fstab` and mounted to `/boot/efi` 
- I’m not able to edit `/snap/fwupd/2632/etc/fwupd/uefi.conf`, since it’s write protected even as root and I did manage to get around this. I also tried to add `/etc/fwupd/uefi.conf` but it seams to be ignored by the snap package obviously. 

```
$ fwupd.fwupdtool esp-list --verbose
08:19:56:0282 FuDebug              Verbose debugging enabled (on console 1)
08:19:56:0285 FuMain               failed to call org.freedesktop.UDisks2.GetBlockDevices(): No such method 'GetBlockDevices', falling back to ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
failed to call org.freedesktop.UDisks2.GetBlockDevices(): No such method 'GetBlockDevices': failed to call org.freedesktop.UDisks2.GetBlockDevices(): No such method 'GetBlockDevices'
```

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.
```shell
$ fwupd.fwupdmgr --version
client version:	1.5.2-38-gab9e6560
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.2-38-gab9e6560
```

hughsie, 2020-12-03:
I'd guess and say the udisks in 16.04 is too old to support GetBlockDevices. Unfortunately snap isn't the miracle cure to run new things on old distros...

superm1, 2020-12-03:
Yeah it looks like it landed in 2.7.2 and udisks2 2.1.7 in Ubuntu 16.04
http://storaged.org/doc/udisks2-api/latest/gdbus-org.freedesktop.UDisks2.Manager.html#gdbus-method-org-freedesktop-UDisks2-Manager.GetBlockDevices

What I would suggest you do is modify `/root/snap/fwupd/current/etc/fwupd/uefi.conf` and hardcode your ESP.

codingforfun, 2020-12-03:
@superm1 
Can you give me some more details about your suggestion: 
> What I would suggest you do is modify /root/snap/fwupd/current/etc/fwupd/uefi.conf and hardcode your ESP.

superm1, 2020-12-03:
When the daemon runs it should create a `/root/snap` folder which contains all of the local configuration of the files.  You can change them rather than `/snap/fwupd/2632/etc/fwupd/uefi.conf`

superm1, 2020-12-03:
I've added a note to the bottom of https://github.com/fwupd/fwupd/wiki/fwupd-snap around hardcoding the ESP.

codingforfun, 2020-12-03:
@superm1 Thanks. That already helped me to understand. By _hardcoding the ESP_ you mean setting 
```
OverrideESPMountPoint=/boot/efi
``` 
in `uefi.conf` or something else? 

`/boot/efi` looks like this for me. Is it OK? 
```shell
tree /boot/efi/
/boot/efi/
└── EFI
    └── ubuntu
        ├── fw
        │   ├── fwupd-9e21f98b-fe98-455c-b388-da5450ab6979.cap
        │   └── fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c.cap
        ├── fwupdx64.efi
        ├── fwupx64.efi
        ├── grub.cfg
        ├── grubx64.efi
        ├── mmx64.efi
        └── shimx64.efi
```

I get no errors anymore, but also only 3 supported devices. I don’t know the number anymore but I remember it to be a lot more before this problem.

superm1, 2020-12-03:
Looks right to me.

check `fwupdmgr get-devices --show-all-devices` to look for the rest of your devices.

closed 2020-12-08 by superm1