fwupd/fwupd #2641: Feature request: preventing device bricking when using signed TPM policies for disk encryption

puiterwijk, 2020-12-03:
**Is your feature request related to a problem? Please describe.**
We now have a [clevis pin that supports TPM2 signed policies](https://github.com/fedora-iot/clevis-pin-tpm2), which allows administrators to use signed TPM policies to bind disk encryption to.
This means that a device will only be able to decrypt any encrypted disks if the PCRs are in any of the sets of values supported by the administrators.
It would be great if fwupd could check the policy, to see if a new policy has already been deployed with the expected values for PCRs changed by the firmware update, to avoid getting into a situation where the firmware is updated before a new policy is deployed.

**Describe the solution you'd like**
fwupd should check the policy (as of `clevis-pin-tpm2`, by default in `/boot/clevis_policy.json`).
If this file doesn't exist, assume this feature isn't used. 
If it does exist, search for a policy that has the newly expected values for PCRs in any of the policies in the file.
If there is not one, it should instead error out with an error like `Deployed clevis policy does not contain new firmware expected PCR values.`.
I'd say any admin that has deployed this, should be able to understand this message.

**Additional context**
An example of the signed policy format:
```
[
    {
        "policy_ref": "",
        "signature": "wVTpOxXo4l3ShmQv2CGU/HQ0Pzc+teXxrOZBqJEPvbXaU5qNMg4N0QDl3EM1yaiK0a0oJyX/1LZNRJWlfWn+eHUFeF1suBAJsLqsydCkUORoXl5NmJkAkku2vOfMIm025lrOJIoMHjgZ3ZnKU8StmlWoxS5u4Eyyy9BzuzsdNNQMeh+PHVEAZ6bxTOCNUtAit8nH+rYLqTt+z9/LKj/5mx0PEJ8g4OL207Ry1SVZbxLG7TZ2/ZmsclfDSuPL/7Q9wRzkoNhKKakWwuDf044vU8/t8Q/lLCFs52nGuyarneeAfUzHfY2wDsYDGnOJtK8SXlHXyyrOCGaTAkEZMprQ3g==",
        "steps": [
            {
                "PCRs": {
                    "hash_algorithm": "SHA256",
                    "pcr_ids": [
                        21,
                        22
                    ],
                    "pcr_values": {
                        "21": "//////////////////////////////////////////8=",
                        "22": "//////////////////////////////////////////8="
                    },
                    "value": "hmfnGClOng3x0wYAuj7rIB92Sq0trXJ0hkPkooXh0fc="
                }
            }
        ]
    }
]
```
Basically, it's a list of signed policies, where each policy contains a `ref`, and then lists a set of steps (each of the steps need to pass).
For any steps where the key `PCRs` exists, we want to look at the `pcr_values` value, which is a PCR ID -> expected value dict, with values encoded as base64-url.

superm1, 2020-12-03:
When you propose `/boot`, is that because it's an ESP?  Or just a convenient path?

superm1, 2020-12-03:
> It would be great if fwupd could check the policy, to see if a new policy has already been deployed with the expected values for PCRs changed by the firmware update, to avoid getting into a situation where the firmware is updated before a new policy is deployed.

The problem with this type of decision is there is no relation between _whether_ or **what** firmware will change.  So you can't know if the decision being made is correct for the system without a heuristic.
For example if you install a CPU microcode UEFI capsule that is part of a FV on the SPI, this isn't normally part of the PCR event log (IIRC).  
See https://github.com/fwupd/fwupd/issues/2548 for more context on the way this should be solved though.

hughsie, 2020-12-04:
> The problem with this type of decision is there is no relation between whether or what firmware will change

I think for a first cut we'd say that any UEFI firmware update _could_ change the PCR0.

superm1, 2020-12-04:
I guess that's a reasonably conservative approach that I'm amenable to.

closed 2020-12-08 by superm1