fwupd/fwupd #2642: No fwupd package in CentOS copr

dnoliver, 2020-12-04:
**Describe the bug**
A clear and concise description of what the bug is.

There is no fwupd package in https://copr-be.cloud.fedoraproject.org/results/rhughes/fwupd/epel-8-x86_64/
Latest fwupd cannot be tested in CentOS 8

**Steps to Reproduce**
Steps to reproduce the behavior.

```shell
# In CentOS 8
$ dnf copr enable rhughes/fwupd
$ dnf update fwupd
```

**Expected behavior**
A clear and concise description of what you expected to happen.

fwupd should be updated to version 1.5

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```

```shell
bash-4.4# fwupdmgr --version
client version: 1.1.4
compile-time dependency versions
        appstream-glib: 0.7.14
        gusb:   0.3.0
        efivar: 36
daemon version: 1.1.4
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

```shell
bash-4.4# fwupdmgr get-devices --show-all-devices
SYS-E403-9P-FN2T System Firmware
  DeviceId:             8c7c8f687a2a7a80b4c50e575d0fd84762edf797
  Guid:                 00000000-0000-0000-0000-000000000000
  Guid:                 a6924b3b-5af6-5f8a-83bc-40100e865b90
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.0.3
  VersionLowest:        0.0.3
  Icon:                 computer
  Created:              2020-12-03

INTEL SSDPEKKA256G8
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 79517f86-8df8-5d6e-a18b-33f0b36a78e9
  Guid:                 68db11e5-b0cf-5bc9-a94e-17e28496e505
  Guid:                 3f235826-4614-56f6-8fd4-b67b346882c9
  Serial:               PHHH944500FQ256B
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Intel Corporation
  VendorId:             NVME:0x8086
  Version:              007D
  Icon:                 drive-harddisk
  Created:              2020-12-03
```

**Additional questions**
- Operating system and version: CentOS 8
- Have you tried rebooting? Yes
- Is this a regression? Unknown

hughsie, 2020-12-04:
This is probably my fault:

> No matching package to install: 'libcurl-devel >= 7.62.0'

RHEL 8 seems to ship libcurl-7.61.1-12.el8.x86_64

hughsie, 2020-12-07:
I built a version in the copr this morning with two patches to fix RHEL8 builds.

closed 2020-12-07 by hughsie

dnoliver, 2020-12-07:
Works now!

I just had to disabled SELinux to make the fwupd systemd service work (I think because of https://bugzilla.redhat.com/show_bug.cgi?id=1832231). This seems to be fixed in RHEL 8 only.

hughsie, 2020-12-08:
> I just had to disabled SELinux

Does the build in the COPR from today work with SELinux turned on? Thanks.

dnoliver, 2020-12-09:
Now it works. Tested on a clean CentOS 8 VM 

```shell
[root@centos-8 ~]# getenforce
Enforcing

[root@centos-8 ~]# rpm -qa fwupd
**fwupd-1.5.4-0.673.20201208git.el8.x86_64**

[root@centos-8 ~]# fwupdmgr --version
client version: 1.5.4
compile-time dependency versions
        gusb:   0.3.6

daemon version: 1.5.4

[root@centos-8 ~]# systemctl start fwupd

[root@centos-8 ~]# systemctl status fwupd
● fwupd.service - Firmware update daemon
   Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
   Active: active (running) since Wed 2020-12-09 09:17:22 PST; 1s ago
     Docs: https://fwupd.org/
 Main PID: 27870 (fwupd)
    Tasks: 6 (limit: 8148)
   Memory: 7.5M
   CGroup: /system.slice/fwupd.service
           └─27870 /usr/libexec/fwupd/fwupd

Dec 09 09:17:21 centos-8 systemd[1]: Starting Firmware update daemon...
Dec 09 09:17:22 centos-8 systemd[1]: Started Firmware update daemon.
```

hughsie, 2020-12-09:
> Now it works. Tested on a clean CentOS 8 VM

Great, thanks for reporting back.