fwupd/fwupd #2648: fwupdmgr shows eMMC as “Unknown Product”

heitbaum, 2020-12-07:
**Describe the bug**
A clear and concise description of what the bug is.

fwupdmgr shows eMMC as “Unknown Product”

**Steps to Reproduce**
Steps to reproduce the behavior.

Run “ fwupdmgr --show-all-devices get-devices”

**Expected behavior**
A clear and concise description of what you expected to happen.

I expected that fwupd would not mark the eMMC as “Unknown Product”

**fwupd version information**
Please provide the version of the daemon and client.
```shell
root@groovy-server:/var/log# fwupdmgr --version
client version: 1.4.5
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.5
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
root@groovy-server:/var/log# fwupdmgr --show-all-devices get-devices -v
(fwupdmgr:3978): GLib-DEBUG: 12:48:55.445: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:3978): Fwupd-DEBUG: 12:48:55.451: looking for os-release at /etc/os-release
(fwupdmgr:3978): FuMain-DEBUG: 12:48:55.465: BJTD4R
  DeviceId:             ff107b3d5f6361671dbadaf255f311458d248a09
  Guid:                 50f53828-ad52-5685-94d9-5726f1699f60 <- EMMC\BJTD4R
  Guid:                 25f9d058-f3b2-5d87-bbdb-ac8845003620 <- EMMC\0021&0256
  Guid:                 5636bf24-da30-5393-8cbf-045092373a80 <- EMMC\0021&0256&BJTD4R
  Serial:               0x67b0fa26
  Plugin:               emmc
  Protocol:             org.jedec.mmc
  Flags:                internal|updatable|registered
  Vendor:               Samsung/Sandisk/LG
  VendorId:             EMMC:0x000015
  Version:              0x0300000000000000
  VersionFormat:        number
  Icon:                 media-memory
  Created:              2020-12-07

root@groovy-server:/var/log# fwupdmgr --show-all-devices get-devices
Unknown Product
│
└─BJTD4R:
      Device ID:           ff107b3d5f6361671dbadaf255f311458d248a09
      Current version:     0x0300000000000000
      Vendor:              Samsung/Sandisk/LG (EMMC:0x000015)
      Serial Number:       0x67b0fa26
      GUIDs:               50f53828-ad52-5685-94d9-5726f1699f60 ← EMMC\BJTD4R
                           25f9d058-f3b2-5d87-bbdb-ac8845003620 ← EMMC\0021&0256
                           5636bf24-da30-5393-8cbf-045092373a80 ← EMMC\0021&0256&BJTD4R
      Device Flags:        • Internal device
                           • Updatable
```

**Additional questions**
- Operating system and version:

```shell
root@groovy-server:/var/log# dmesg | grep -e soc0 -e model
[    0.000000] Machine model: Hardkernel ODROID-N2Plus
[    1.349585] soc soc0: Amlogic Meson G12B (S922X) Revision 29:c (40:2) Detected

root@groovy-server:/var/log# uname -a
Linux groovy-server 5.10.0-odroid-panfrost-arm64 #1 SMP PREEMPT Ubuntu 5.10.0-202012021439~groovy (2020-12-02) aarch64 aarch64 aarch64 GNU/Linux
```

- Have you tried rebooting?
- Is this a regression?

hughsie, 2020-12-07:
> root@groovy-server:/var/log# fwupdmgr --show-all-devices get-devices
> Unknown Product

That's from the DMI data; what does `sudo fwupdtool hwids` say?

heitbaum, 2020-12-07:
root@groovy-server:~# fwupdtool hwids
Failed to open file “/sys/firmware/dmi/tables/smbios_entry_point”: No such file or directory
root@groovy-server:~# ls /sys/firmware/
devicetree/ fdt

hughsie, 2020-12-07:
You need fwupd 1.5.x if you want DeviceTree support.

heitbaum, 2020-12-07:
Hi @hughsie  - I can confirm that pulling the upstream hirsute fwupd into groovy has updated the output - as below

```shell 
root@groovy-server:/etc/apt# fwupdmgr --version
client version: 1.5.1
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.1

root@groovy-server:/etc/apt# fwupdmgr get-devices --show-all-devices
WARNING: UEFI capsule updates not available or enabled
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Hardkernel ODROID-N2Plus
│
└─BJTD4R:
      Device ID:          ff107b3d5f6361671dbadaf255f311458d248a09
      Current version:    0x0300000000000000
      Vendor:             Samsung/Sandisk/LG (EMMC:0x000015)
      Serial Number:      0x67b0fa26
      Device Flags:       • Internal device
                          • Updatable

root@groovy-server:/etc/apt# fwupdmgr get-devices --show-all-devices -v
(fwupdmgr:54977): GLib-DEBUG: 16:47:33.525: setenv()/putenv() are not thread-safe and should not be used after threads are created
WARNING: UEFI capsule updates not available or enabled
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
(fwupdmgr:54977): FuMain-DEBUG: 16:47:33.587: BJTD4R
  DeviceId:             ff107b3d5f6361671dbadaf255f311458d248a09
  Serial:               0x67b0fa26
  Plugin:               emmc
  Protocol:             org.jedec.mmc
  Flags:                internal|updatable|registered
  Vendor:               Samsung/Sandisk/LG
  VendorId:             EMMC:0x000015
  Version:              0x0300000000000000
  VersionFormat:        number
  Icon:                 media-memory
  Created:              2020-12-07

root@groovy-server:/etc/apt# fwupdtool hwids
Computer Information
--------------------
ProductName: Hardkernel ODROID-N2Plus

Hardware IDs
------------
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown

root@groovy-server:/etc/apt# fwupdtool hwids -v
16:49:29:0115 FuDebug              Verbose debugging enabled (on console 1)
16:49:29:0116 FuHwids              ignoring Manufacturer: no data available
16:49:29:0116 FuHwids              ignoring EnclosureKind: offset bigger than size 0
16:49:29:0116 FuHwids              ignoring Family: offset bigger than size 6
16:49:29:0116 FuHwids              smbios property ProductName=Hardkernel ODROID-N2Plus
16:49:29:0116 FuHwids              ignoring ProductSku: offset bigger than size 6
[... 26 lines cut ...]
------------
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown

root@groovy-server:/etc/apt# fwupdmgr get-updates
WARNING: UEFI capsule updates not available or enabled
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Devices with no available firmware updates: 
 • BJTD4R
No updatable devices

```

hughsie, 2020-12-07:
> not available as 'Manufacturer' unknown

Is there anything in DT suggesting the vendor? e.g. `vendor` or anything in the DT sysfs directory?

superm1, 2020-12-07:
> WARNING: UEFI capsule updates not available or enabled
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Devices with no available firmware updates: 

Just FYI this message is fixed in 1.5.2. (see https://github.com/fwupd/fwupd/commit/bdaa080be3f7fa18ba6e7a99822cc097170dddca)

superm1, 2020-12-07:
> Is there anything in DT suggesting the vendor? e.g. vendor or anything in the DT sysfs directory?

If you don't have enough SMBIOS data you shouldn't make a CHID.  This seems like expected behavior to me.

hughsie, 2020-12-07:
> If you don't have enough SMBIOS data you shouldn't make a CHID

I wondered if we should fall back, but I guess you're right -- we should require a `vendor` for the CHID.

heitbaum, 2020-12-08:
I have done a 1.5.3 compile on groovy (with hirsute updates)

```shell
meson build -Dplugin_flashrom=false -Dplugin_dell=false -Dplugin_uefi=true -Dplugin_redfish=true -Dplugin_nvme=true -Dplugin_msr=false -Dplugin_dummy=true -Dgtkdoc=true -Dsupported_build=true --prefix /usr
```

Output below.
```shell
root@groovy-server:/data/fwupd-1.5.3# systemctl reload dbus
root@groovy-server:/data/fwupd-1.5.3# fwupdmgr --version
client version: 1.5.3
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.3
root@groovy-server:/data/fwupd-1.5.3# fwupdmgr get-devices --show-all-devices
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: UEFI capsule updates not available or enabled
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Hardkernel ODROID-N2Plus
│
└─BJTD4R:
      Device ID:          ff107b3d5f6361671dbadaf255f311458d248a09
      Current version:    0x0300000000000000
      Vendor:             Samsung/Sandisk/LG (EMMC:0x000015)
      Serial Number:      0x67b0fa26
      GUIDs:              50f53828-ad52-5685-94d9-5726f1699f60 ← EMMC\BJTD4R
                          25f9d058-f3b2-5d87-bbdb-ac8845003620 ← EMMC\0021&0256
                          5636bf24-da30-5393-8cbf-045092373a80 ← EMMC\0021&0256&BJTD4R
      Device Flags:       • Internal device
                          • Updatable
    
root@groovy-server:/data/fwupd-1.5.3# fwupdmgr get-devices --show-all-devices -v
(fwupdmgr:89226): GLib-DEBUG: 12:00:58.001: setenv()/putenv() are not thread-safe and should not be used after threads are created
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: UEFI capsule updates not available or enabled
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
(fwupdmgr:89226): FuMain-DEBUG: 12:00:58.074: BJTD4R
  DeviceId:             ff107b3d5f6361671dbadaf255f311458d248a09
  Guid:                 50f53828-ad52-5685-94d9-5726f1699f60 <- EMMC\BJTD4R
  Guid:                 25f9d058-f3b2-5d87-bbdb-ac8845003620 <- EMMC\0021&0256
  Guid:                 5636bf24-da30-5393-8cbf-045092373a80 <- EMMC\0021&0256&BJTD4R
  Serial:               0x67b0fa26
  Plugin:               emmc
  Protocol:             org.jedec.mmc
  Flags:                internal|updatable|registered
  Vendor:               Samsung/Sandisk/LG
  VendorId:             EMMC:0x000015
  Version:              0x0300000000000000
  VersionFormat:        number
  Icon:                 media-memory
  Created:              2020-12-08

root@groovy-server:/data/fwupd-1.5.3# fwupdtool hwids -v
12:01:26:0657 FuDebug              Verbose debugging enabled (on console 1)
12:01:26:0657 FuHwids              ignoring Manufacturer: no data available
12:01:26:0657 FuHwids              ignoring EnclosureKind: offset bigger than size 0
12:01:26:0657 FuHwids              ignoring Family: offset bigger than size 6
12:01:26:0657 FuHwids              smbios property ProductName=Hardkernel ODROID-N2Plus
12:01:26:0657 FuHwids              ignoring ProductSku: offset bigger than size 6
12:01:26:0657 FuHwids              ignoring BiosVendor: offset bigger than size 0
12:01:26:0657 FuHwids              ignoring BiosVersion: offset bigger than size 0
12:01:26:0657 FuHwids              ignoring BiosMajorRelease: offset bigger than size 0
12:01:26:0657 FuHwids              ignoring BiosMinorRelease: offset bigger than size 0
12:01:26:0657 FuHwids              ignoring BaseboardManufacturer: offset bigger than size 0
12:01:26:0658 FuHwids              ignoring BaseboardProduct: offset bigger than size 0
12:01:26:0658 FuHwids              HardwareID-0 is not available, not available as 'Manufacturer' unknown
12:01:26:0658 FuHwids              HardwareID-1 is not available, not available as 'Manufacturer' unknown
12:01:26:0658 FuHwids              HardwareID-2 is not available, not available as 'Manufacturer' unknown
12:01:26:0658 FuHwids              HardwareID-3 is not available, not available as 'Manufacturer' unknown
[... 37 lines cut ...]
Hardware IDs
------------
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
root@groovy-server:/data/fwupd-1.5.3# fwupdmgr get-updates
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: UEFI capsule updates not available or enabled
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Devices with no available firmware updates: 
 • BJTD4R
No updatable devices
```

superm1, 2020-12-08:
>WARNING: UEFI capsule updates not available or enabled
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.

@hughsie since capsule updates aren't super common on ARM currently, I feel like we should only be showing that warning on x86.  What do you think?

Otherwise everything in 1.5.3 looks expected to me now.

heitbaum, 2020-12-10:
Hi guys - where is the Manufacturer suppose to come from? 
Is this a problem in the Device Tree?

sys# find /sys/devices/ | pastebinit 
https://paste.ubuntu.com/p/KHMx6gxHJz/
Sys# find /proc/device-tree | pastebinit 
https://paste.ubuntu.com/p/m7mQvXBDgs/

Ref: https://elinux.org/Device_Tree_Reference

hughsie, 2020-12-10:
@heitbaum it should be listed in `/sys/firmware/devicetree/base/vendor` according to the DT spec.

heitbaum, 2020-12-13:
Hi @hughsie - the DTS for the DTB file looks right(ish) not sure why / how the manufacturer in the DTS is suppose to be visible in the device tree. The spec from https://elinux.org/Device_Tree_Usage looks to match this DTS file.

root@groovy-server:/# cat 2.dts | pastebinit
https://paste.ubuntu.com/p/z8gyZBZwMj/

heitbaum, 2020-12-14:
As being discussed in the LibreELEC #amlogic slack

Maybe because the /proc/cpuinfo Manufacturer field was dropped on arm64, it's only available on arm - there is no replacement.

What does this Manufacturer means ? the SoC ? the Board ? the CPU core ?
- does HWIDS need to fail because missing manufacturer, even though vendor info in the DTB?
ARM providing the CPU IP, Amlogic providing the SoC and HardKernel the board
- ARM vendor/ AArch64 showing up in /proc/cpuinfo
- Amlogic in the /sys/bus/soc/devices/soc0 
- Hardkernel using the DT top compatible

Should be true for all upstream supported ARM64 boards

I’m expecting my Khadas VIM3 today/tomorrow and will check it out too.

hughsie, 2020-12-14:
Doesn't the spec say to include a `vendor` attribute on each node if applicable? 

> What does this Manufacturer means

In the context of the root node, the "_entire system_" e.g. "Talos" if that's the OEM or integrator. We're reading out the "baseboard" from `vpd/root-node-vpd@a000/enclosure@1e00/backplane@800/vendor` which can specify the base board vendor if that's different.

hughsie, 2021-01-10:
I don't think this is a bug; I think implementing a "vendor" attribute in DT would magically fix this.

closed 2021-01-10 by hughsie