fwupd/fwupd #2650: X1 Extreme Gen 2 fingerprint firmware update

HadiModarres, 2020-12-08:
fwupd stuck on upgrading fingerprint firmware ("Prometheus") on Thinkpad X1 Extreme Gen 2: 

> ^CTerminated
(base) hadi@hadi-ThinkPad-X1-Extreme-2nd:~$ fwupdmgr update -vv
(fwupdmgr:8784): GLib-DEBUG: 01:33:16.619: setenv()/putenv() are not thread-safe and should not be used after threads are created
• Thunderbolt Controller has the latest available firmware version
(fwupdmgr:8784): FuMain-DEBUG: 01:33:16.628: current version is 59.00: 59.00=same
Upgrade available for Prometheus from 10.01.3121519 to 10.01.3273255
Prometheus and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading 10.01.3273255 for Prometheus...
(fwupdmgr:8784): FuCommon-DEBUG: 01:33:21.221: creating path /home/hadi/.cache/fwupd
(fwupdmgr:8784): FuMain-DEBUG: 01:33:21.225: skpping download as file already exists
(fwupdmgr:8784): Fwupd-DEBUG: 01:33:21.226: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:8784): Fwupd-DEBUG: 01:33:21.228: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:8784): Fwupd-DEBUG: 01:33:21.251: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:8784): Fwupd-DEBUG: 01:33:21.253: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]

The upgrade is stuck on Authenticating with no further info or messages as to why it's failing to go forward. Any help is appreciated

hughsie, 2020-12-08:
What distro? You ignored most of the details we need...

HadiModarres, 2020-12-08:
@hughsie oh sorry about that, it's Ubuntu 20.04

fwupdmgr version: 
`client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
`

devices: 

> 20QVCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           607d06b8b13f22018453d6157e624da99ebb1340
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     59.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               afa98a84-bdd7-5c4d-8df0-a935055e0df9
│                          3fb91848-3767-5700-880c-d2a82d7bdb89
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Prometheus:
│ │   Device ID:           d432baa2162a32c1554ef24bd8281953b9d07c11
│ │   Summary:             Fingerprint reader
│ │   Current version:     10.01.3121519
│ │   Vendor:              Synaptics (USB:0x06CB)
│ │   Install Duration:    2 seconds
│ │   Update State:        failed
│ │   Last modified:       2020-12-08 14:33
│ │   GUIDs:               09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                        8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:        • Updatable
│ │                        • Supported on remote server
│ │                        • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:         8dcffb5d059857368caa2b7b22371ebd831b0c0c
│       Current version:   0022
│       Minimum Version:   0022
│       Vendor:            Synaptics (USB:0x06CB)
│       GUID:              7c5a1e62-38fa-5859-9337-09dbac6377e4
│       Device Flags:      • Updatable
│                          • Supported on remote server
│     
├─SAMSUNG MZVLB256HBHQ-000L7:
│     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:             NVM Express Solid State Drive
│     Current version:     4M2QEXH7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          1b3b43f9-e0b0-5978-a89b-6f0866124233
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Samsung SSD 970 EVO 500GB:
│     Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:             NVM Express Solid State Drive
│     Current version:     2B2QEXE7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
[... 82 lines cut ...]
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           0844f90d5c648645f4d6c31672731d9d83ede5f2
│     Current version:     0.0.26889
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                88b8e549-d3a1-4371-9810-f30d43abde6e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UHD Graphics 630 (Mobile):
      Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
      Vendor:              Intel Corporation (PCI:0x8086)
      GUIDs:               12c60b98-52cd-500a-b99e-087aa6b47f0b
                           db677673-1102-5481-a11e-6f397e123ced
      Device Flags:        • Internal device
                           • Cryptographic hash verification is available
    
________________________________________________

HadiModarres, 2020-12-08:
@hughsie I removed fprintd and tried again and it went through, so I guess the issue was probably that the device was being used already. Maybe can add a check to the code to cancel installation if device is being used? 

cheers

superm1, 2020-12-08:
Mmm that doesn't sound right.  There needs to be some handshaking to inhibit fprintd that means.

hughsie, 2020-12-10:
Ohh I see why we hit this. fprint only keeps the device open for 5 seconds and then releases it, which seems like a small window to hit. **But**, we're asking the user to authenticate with the same device we're about to upgrade so a different part of the stack woke up the hardware just before we're about to deploy an update onto it.

According to the fprint team just waiting 2 seconds before opening the device should be plenty enough to make this work, and is orders of magnitide simpler than an Inhibit() and Release() or fd API.

benzea, 2020-12-10:
> According to the fprint team just waiting 2 seconds before opening the device should be plenty enough to make this work

Yup. You are "only" racing with fprintd closing the device after having authorized the operation. In theory fprintd might wait for the finger to be removed, but I think even that is cancelled. So the window should be much shorter than 2 seconds.

In the long run we may need the API in fprintd. I have added a not to the ticket discussing DBus API changes.

hughsie, 2020-12-10:
@HadiModarres -- would it possible to get a fwupd daemon log with `--verbose` when the bug happens? I don't know if it's failing to open the device or failing to claim the interface or failing to do the bulk transfer. I suspect the first one.

closed 2021-01-04 by hughsie