fwupd/fwupd #2671: BIOS update for XPS 15 9550 is not detected

jmarino, 2020-12-10:
BIOS update for XPS 15 9550 laptop is not detected. The current BIOS version installed is 1.13.1:
```shell
$ fwupdmgr get-devices | grep -A 7 'System Firmware'
├─System Firmware:
│ │   Device ID:          2da31270d317b076424992de14a0f08ae373c137
│ │   Current version:    1.13.1
│ │   Minimum Version:    1.13.1
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              124c207d-5db8-4d95-bd31-34fd971b34f9
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       0d25ab20-f5f1-591f-8baf-c59f009f9548 ← UEFI\RES_{124C207D-5DB8-4D95-BD31-34FD971B34F9}
```
Latest BIOS version available is 1.14.0 as listed in [Dell's website](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=90khw&oscode=wt64a&productcode=xps-15-9550-laptop) and [LVFS database](https://fwupd.org/lvfs/devices/com.dell.uefi124c207d.firmware).

**Steps to Reproduce**
I run `fwupdmgr refresh` and then:
```shell
$ fwupdmgr get-updates 124c207d-5db8-4d95-bd31-34fd971b34f9
Selected device: System Firmware
Devices with no available firmware updates: 
 • System Firmware
No updatable devices
```

**Expected behavior**
It should update the BIOS to the latest version 1.14.0.

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.5.3
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.3
```
Installed with pacman, package `fwupd 1.5.3-1`

**fwupd device information**
```shell
$ fwupdmgr get-devices --show-all-devices
```
Output from previous command: [fwupdmgr-get-devices.txt](https://github.com/fwupd/fwupd/files/5675742/fwupdmgr-get-devices.txt)

**System UEFI configuration**
```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 2 seconds
BootOrder: 0000,0001,0002,0004,0007,0003
Boot0000* Linux Boot Manager	HD(1,GPT,21f62040-b533-49c6-81c0-f7bf0fa6e0da,0x800,0x64000)/File(\EFI\systemd\systemd-bootx64.efi)
Boot0001* Diskette Drive	BBS(Floppy,Diskette Drive,0x0)..BO
Boot0002* M.2 PCIe SSD	BBS(HD,PM951 NVMe SAMSUNG 512GB ,0x0)..BO
Boot0003* Linux Firmware Updater	HD(1,GPT,21f62040-b533-49c6-81c0-f7bf0fa6e0da,0x800,0x64000)/File(\EFI\arch\fwupdx64.efi)
Boot0004* CD/DVD/CD-RW Drive	BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0005* Linux Firmware Updater	HD(1,GPT,21f62040-b533-49c6-81c0-f7bf0fa6e0da,0x800,0x64000)/File(\EFI\arch\fwupdx64.efi)
Boot0007* USB Storage Device	BBS(USB,USB Storage Device,0x0)..BO
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-124c207d-5db8-4d95-bd31-34fd971b34f9-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-124c207d-5db8-4d95-bd31-34fd971b34f9-0
```

```shell
$ tree /boot
/boot
├── EFI
│   ├── arch
│   │   ├── fw
│   │   │   ├── fwupd-124c207d-5db8-4d95-bd31-34fd971b34f9.cap
│   │   │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │   │   └── fwupdate-Ni6RxZ.cap
│   │   ├── fwupdx64.efi
│   │   └── fwupx64.efi
│   ├── Boot
│   │   └── BOOTX64.EFI
│   ├── Dell
│   │   └── Bios
│   │       └── Recovery
│   │           ├── BIOS_CUR.RCV
│   │           └── BIOS_PRE.rcv
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   ├── linux-jose.conf.bak
│   │   └── linux-lts.conf
│   └── loader.conf
├── vmlinuz-linux
└── vmlinuz-linux-lts
```

**Additional questions**
Running Arch Linux, with kernel 5.9.13-arch1-1.
This is a regression, I have updated the BIOS with fwupd before.
The hard drive in the laptop is NVMe.

hughsie, 2020-12-10:
Is the lvfs enabled? If you do `fwupdmgr refresh` what gets printed?

superm1, 2020-12-10:
This sounds like it might be the same bug as https://github.com/fwupd/firmware-dell/issues/38 which I didn't get a chance to dig into details yet.

jmarino, 2020-12-10:
```shell
$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 1 local device supported
```

superm1, 2020-12-10:
Can you please run

`fwupdmgr get-releases`

And pick System firmware when it comes up?

jmarino, 2020-12-10:
I'm not seeing System firmware in the output of `fwupdmgr get-releases`:
```
$ fwupdmgr get-releases
Selected device: Unifying Receiver
XPS 15 9550
│
├─Unifying Receiver (RQR12) Device Update:
│     New version:        RQR12.10_B0032
│     Remote ID:          lvfs
│     Summary:            Firmware for the Logitech Unifying Receiver (RQR12.xx)
│     License:            Proprietary
│     Size:               56.8 kB
│     Created:            2019-07-18
│     Urgency:            High
│     Vendor:             Logitech
│     Duration:           30 seconds
│     Description:        
│     This release addresses an encrypted keystroke injection vulnerability sent by pointing devices. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.
│     
│     A few of Logitech's devices used to send select buttons in an unencrypted way, and in an effort to protect against this vulnerability, Logitech removed the feature. Affected hardware is:
│     
│     • Wireless Mouse M335
│     • Zone Touch Mouse T400
│     • Wireless Mouse M545
│     • Wireless Mouse M560
│     • Touch Mouse M600
│     • Touch Mouse T620
│     • Wireless Rechargeable Touchpad T650
│     
│     Although Logitech does not recommend it, these features may be re-activated by keeping/downgrading the receiver to an older firmware.
│   
├─Unifying Receiver (RQR12) Device Update:
│     New version:        RQR12.08_B0030
│     Remote ID:          lvfs
│     Summary:            Firmware for the Logitech Unifying Receiver (RQR12.xx)
│     License:            Proprietary
│     Size:               72.7 kB
│     Created:            2017-06-26
│     Urgency:            High
│     Vendor:             Logitech
│     Duration:           30 seconds
│     Flags:              is-downgrade
│     Description:        
│     This release addresses an encrypted keystroke injection issue known as Bastille security issue #13. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.
│     
│     A few of Logitech's devices used to send select buttons in an unencrypted way, and in an effort to protect against this vulnerability, Logitech removed the feature. Affected hardware is:
│     
│     • Wireless Mouse M335
│     • Zone Touch Mouse T400
│     • Wireless Mouse M545
│     • Wireless Mouse M560
│     • Touch Mouse M600
│     • Touch Mouse T620
│     • Wireless Rechargeable Touchpad T650
│     
│     Although Logitech does not recommend it, these features may be re-activated by keeping/downgrading the receiver to an older firmware.
│   
└─Unifying Receiver (RQR12) Device Update:
      New version:        RQR12.07_B0029
      Remote ID:          lvfs
      Summary:            Firmware for the Logitech Unifying Receiver (RQR12.xx)
      License:            Proprietary
      Size:               69.6 kB
      Created:            2017-05-03
      Urgency:            High
      Vendor:             Logitech
      Duration:           30 seconds
      Flags:              is-downgrade
      Description:        
      This release addresses an unencrypted keystroke injection issue known as Bastille security issue #11. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.
```

superm1, 2020-12-10:
Do you have udisks2 installed?  If not can you please provide this output:

`# fwupdtool get-devices --plugins=uefi --verbose`

jmarino, 2020-12-10:
I have udisks2 installed:
```shell
$ pacman -Q udisks2
udisks2 2.9.1-1
```
I ran the command anyway, here's the output:
[fwupdtool-get-devices.txt](https://github.com/fwupd/fwupd/files/5676005/fwupdtool-get-devices.txt)

jmarino, 2020-12-10:
Before you ask:
```shell
$ cat /etc/fwupd/uefi.conf
[uefi]

# the shim loader is required to chainload the fwupd EFI binary unless
# the fwupd.efi file has been self-signed manually
#DisableShimForSecureBoot=true

# the EFI system partition path used
# if this is is not /boot/efi, /boot, or /efi
OverrideESPMountPoint=/boot

# amount of free space required on the ESP, for example using 0x2000000 for 32Mb
#RequireESPFreeSpace=

# with the UEFI removable path enabled, the default esp path is set to /EFI/boot
# the shim EFI binary and presumably this is $ESP/EFI/boot/bootx64.efi
#FallbacktoRemovablePath=false
```

superm1, 2020-12-10:
OK so in there this is showing up:
```
22:29:25:0961 FuEngine             failed to get releases for System Firmware: No releases found: no HWIDs matched d98dc575-9143-5a8f-a620-9440ef640c8d
```

superm1, 2020-12-10:
I'm not sure where that's actually coming from.  the FW on LVFS doesn't have it.
https://fwupd.org/lvfs/components/6572/requires

superm1, 2020-12-10:
Can you please manually download it (https://fwupd.org/downloads/b656e631b3724e2dd129333f27e60472e4ef2ff9918669e5e52f9cc8f5323d59-XPS_9550_1.14.0.cab) and try to install with 
`fwupdtool --verbose install foo.cab`

superm1, 2020-12-10:
I expect that will work.  
@hughsie looking over the stable metadata I think what's happening here is that "one" of the releases has a requirement for
```
<requires>
      <hardware>d98dc575-9143-5a8f-a620-9440ef640c8d</hardware>
    </requires>
```

and the metadata is treating those requirements as all releases need them.

jmarino, 2020-12-10:
That seems to have worked. Here's the output: 
[fwupdtool-install.txt.gz](https://github.com/fwupd/fwupd/files/5676090/fwupdtool-install.txt.gz)

hughsie, 2020-12-11:
@superm1 the requires are coming from https://fwupd.org/lvfs/components/610/requires

> the metadata is treating those requirements as all releases need them

This is by design; you need to use a different <id> if the requirements are different for a specific firmware.

hughsie, 2020-12-11:
No, I'm wrong -- most of the components have no requirements, but :

 * https://fwupd.org/lvfs/components/610/requires has `e7eb42e2-06c6-5185-9c5f-33ff1dbb1c0b` and `d98dc575-9143-5a8f-a620-9440ef640c8d`
 * https://fwupd.org/lvfs/components/5496/requires has just `d98dc575-9143-5a8f-a620-9440ef640c8d`
 * https://fwupd.org/lvfs/components/6405/requires has just `d98dc575-9143-5a8f-a620-9440ef640c8d`

...the 610 isn't being used as we only include the last 5 releases and that fails that requirement. which explains why we only see:

    <requires>
      <hardware>d98dc575-9143-5a8f-a620-9440ef640c8d</hardware>
    </requires>

...in the metadata. Note, I've just merged a patch to the LVFS so you can search for requirements using the internal logged-in search function -- which is how I found the other two.

I think the LVFS is working as designed here. @superm1 over to you?

superm1, 2020-12-11:
>https://fwupd.org/lvfs/components/5496/requires has just d98dc575-9143-5a8f-a620-9440ef640c8d https://fwupd.org/lvfs/components/6405/requires has just d98dc575-9143-5a8f-a620-9440ef640c8d

OK fine on those (1.15.0 and 1.14.2) but how about 1.14.0?  That should be less than 5 versions ago and it doesn't have any requirements set - https://fwupd.org/lvfs/components/6572

hughsie, 2020-12-11:
> and it doesn't have any requirements set

Remember the <release> get combined into one component in the AppStream metadata. So if any of the components of the same <id> have a requirement, all of them do.

superm1, 2020-12-11:
> Remember the get combined into one component in the AppStream metadata. So if any of the components of the same have a requirement, all of them do.

Well that's the problem then :(
I don't think this comes across clearly on LVFS at all.  I understand it that that there is a split now that was reflected via CHID, but CHID doesn't **ACTUALLY** reflect release on LVFS.

The precarious position is that;
* <= 1.14.0 is intended to go on both
* 1.14.2  and 1.15.0 is intended to go only on the Precision 5510

You can see on support.dell.com the most recent one for the XPS is 1.14.0: https://www.dell.com/support/home/en-us/product-support/product/xps-15-9550-laptop/drivers and for the Precision is 1.15.0: https://www.dell.com/support/home/en-us/product-support/product/precision-m5510-workstation/drivers

hughsie, 2020-12-11:
>   <= 1.14.0 is intended to go on both
>  1.14.2 and 1.15.0 is intended to go only on the Precision 5510

The Precision 5510 specific ones need a different ID then. I think changing the <id>s on the LVFS will magically fix this.

superm1, 2020-12-11:
I don't appear to be able to do it myself, here it is: https://fwupd.org/lvfs/components/6572

hughsie, 2020-12-11:
It depends; we can change all the <= 1.14.0 ones -- or just  1.14.2 and 1.15.0 -- the latter makes more sense I think. What would you like the new ID to be?

superm1, 2020-12-11:
Yeah you're probably right changing just the later ones make sense.  I think just adding a suffix of `precision5510` would be fine.

hughsie, 2020-12-11:
Okay, done those two, but what about https://fwupd.org/lvfs/components/610/overview ?

superm1, 2020-12-11:
I wouldn't worry too much about older.

superm1, 2020-12-11:
@jmarino can you please run `fwupdmgr refresh` and then `fwupdmgr get-releases` to confirm you see the firmware now?

jmarino, 2020-12-11:
I don't think the changes have propagated to me yet. I did `fwupdmgr refresh --force` and `fwupdmgr get-releases` and see no mention of the firmware.
The reason I think the changes have not propagated to me is because this produces no output:
```shell
$ zcat /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz | grep precision5510
```

hughsie, 2020-12-11:
It's up to 4 hours for the sync.

hughsie, 2020-12-11:
> It's up to 4 hours for the sync.

Try now with `fwupdmgr refresh --force`

jmarino, 2020-12-11:
Ok now!
I confirm that I see a firmware option when I run `fwupdmgr get-releases`, with version 1.14.0 listed as the latest.
Thanks!

closed 2021-01-10 by hughsie

Gelmo, 2021-05-26:
Just wanted to report that I had the same issue today with the same laptop on 1.13.1 (Device ID 2da31270d317b076424992de14a0f08ae373c137 and GUID 124c207d-5db8-4d95-bd31-34fd971b34f9). There still seems to be something interfering that you may want to review for other device owners.

fwupdmgr did not find this as a supported device, even after a forced refresh. I manually installed 1.14.0 via fwupdtool.

After the update to 1.14.0, fwupdmgr still does not see this as a supported device.

The Dell site suggests that the latest bios for this is 1.14.0. When searching the GUID on fwupd.org, the results show that the latest version is 1.16.1. Should I attempt to manually install 1.16.1? Is that version not intended for my device? If so, perhaps the listings at fwupd.org need more adjustments. Let me know if there's anything I should provide to assist with this.