fwupd/fwupd #2676: fwupd 1.5.2 failed to detect UEFI ESP partition in some machines

lcp, 2020-12-14:
I got a report for openSUSE Tumbleweed that the user got the following warning after upgrading fwupd to 1.5.2:

WARNING: UEFI ESP partition not detected or configured

It turned out that fwupd thought the system disk is a removable disk. Here is the output of "udisksctl info --block-device /dev/sdb"

/org/freedesktop/UDisks2/block_devices/sdb:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/sdb
    DeviceNumber:               2064
    Drive:                      '/org/freedesktop/UDisks2/drives/TEAM_L5_3D_240GB_AB20170727A0104534'
    HintAuto:                   true
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 false
    Id:                         
    IdLabel:                    
    IdType:                     
    IdUUID:                     
    IdUsage:                    
    IdVersion:                  
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/sdb
    ReadOnly:                   false
    Size:                       240057409536
    Symlinks:                   /dev/disk/by-id/ata-TEAM_L5_3D_240GB_AB20170727A0104534
                                /dev/disk/by-id/scsi-0ATA_TEAM_L5_3D_240GB_AB20170727A0104534
                                /dev/disk/by-id/scsi-1ATA_TEAM_L5_3D_240GB_AB20170727A0104534
                                /dev/disk/by-id/scsi-SATA_TEAM_L5_3D_240GBAB20170727A0104534
                                /dev/disk/by-id/scsi-SATA_TEAM_L5_3D_240GB_AB20170727A0104534
                                /dev/disk/by-path/pci-0000:06:00.0-ata-2
                                /dev/disk/by-path/pci-0000:06:00.0-ata-2.0
                                /dev/disk/by-path/pci-0000:06:00.0-scsi-5:0:0:0
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.PartitionTable:
    Partitions:         /org/freedesktop/UDisks2/block_devices/sdb1
                        /org/freedesktop/UDisks2/block_devices/sdb3
                        /org/freedesktop/UDisks2/block_devices/sdb2
    Type:               gpt

Although the disk is clearly a SATA disk. However, the UEFI firmware supports SATA hot plug (probably for the RAID mode), and udisk2 reported "HintSystem: false". Maybe "HintSystem" is not a reliable flag for the ESP detection.

hughsie, 2020-12-14:
That's already in the fallback path; the bigger question was why the partition was not marked with the correct GUID of `c12a7328-f81f-11d2-ba4b-00a0c93ec93b` -- perhaps you can ask how was it created? Thanks.

@superm1 Do you think we should remove the check for internal devices for the fallback case?

lcp, 2020-12-14:
$ sudo fwupdtool esp-list -v
    [sudo] password for root: 
    10:50:56:0041 FuDebug              Verbose debugging enabled (on console 1)
    10:50:56:0064 FuCommon             device /org/freedesktop/UDisks2/block_devices/sdb1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 0, fs: vfat
    10:50:56:0068 FuCommon             device /org/freedesktop/UDisks2/block_devices/sdc1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 0, fs: xfs
    10:50:56:0071 FuCommon             device /org/freedesktop/UDisks2/block_devices/sdb3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 0, fs: btrfs
    10:50:56:0076 FuCommon             device /org/freedesktop/UDisks2/block_devices/sdd1, type: 0x83, internal: 0, fs: ext4
    10:50:56:0079 FuCommon             device /org/freedesktop/UDisks2/block_devices/sdb2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 0, fs: swap
    10:50:56:0083 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 0, fs: xfs
    Choose a volume:
    0.      Cancel

The partition type looks correct to me.

hughsie, 2020-12-14:
Ahh, we're doing the filter in both cases. I think it might be safer just to not trust the hint. PR incoming.

lcp, 2020-12-14:
Thanks for the patch. Will package it for the original bug reporter for the further testing.

lcp, 2020-12-15:
Confirmed that the patch works. Close this bug.

closed 2020-12-15 by lcp