fwupd/fwupd #2689: Dell WD19 upgrade issue

joedborg, 2020-12-15:
**Describe the bug**
A clear and concise description of what the bug is.

```
➜ sudo fwupdmgr update --verbose
(fwupdmgr:11439): GLib-DEBUG: 09:19:53.367: setenv()/putenv() are not thread-safe and should not be used after threads are created
Upgrade available for Thunderbolt controller in Dell dock from 40.00 to 43.00
Thunderbolt controller in Dell dock and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading 43.00 for Thunderbolt controller in Dell dock...
(fwupdmgr:11439): FuCommon-DEBUG: 09:19:56.416: creating path /root/.cache/fwupd
(fwupdmgr:11439): FuMain-DEBUG: 09:19:56.433: skpping download as file already exists
(fwupdmgr:11439): Fwupd-DEBUG: 09:19:56.438: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:11439): Fwupd-DEBUG: 09:19:56.449: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:11439): Fwupd-DEBUG: 09:19:56.483: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:11439): Fwupd-DEBUG: 09:19:56.484: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
could not validate firmware: Error reading from file: Input/output error. See https://github.com/fwupd/fwupd/wiki/Thunderbolt:-Validation-failed-or-unknown-device for more information.
```

**Steps to Reproduce**
Run `sudo fwupdmgr update --verbose` with the Dell dock attached.

**Expected behavior**
A clear and concise description of what you expected to happen.

FW update to succeed.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version

client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the external fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --filter=~internal

XPS 13 9370
│
└─WD19TB:
  │   Device ID:           275f399553a6ac1bbbe1e463791dfd12652efb6f
  │   Summary:             High performance dock
  │   Current version:     01.00.00.00
  │   Minimum Version:     01.00.00.00
  │   Vendor:              Dell Inc. (USB:0x413C)
  │   Install Duration:    1 minute
  │   Serial Number:       6QFHQY2/3137095740343049
  │   Update Error:        A pending update will be completed next time the dock is unplugged from your computer
  │   GUID:                cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
  │   Device Flags:        • Requires AC power
  │                        • Supported on remote server
  │                        • Hardware is waiting to be replugged
  │                        • Device update needs activation
  │                        • Device stages updates
  │                        • Device can recover flash failures
  │                        • Device is usable for the duration of the update
  │ 
  ├─Thunderbolt controller in Dell dock:
  │     Device ID:         d06ed328732c6080e634a0bbb407e2fb0c7cd87d
  │     Summary:           Thunderbolt controller
  │     Current version:   40.00
  │     Vendor:            Dell Inc. (TBT:0x00D4)
  │     Install Duration:  22 seconds
  │     Update State:      failed
  │     Update Error:      could not validate firmware: Error reading from file: Input/output error. See https://github.com/fwupd/fwupd/wiki/Thunderbolt:-Validation-failed-or-unknown-device for more information.
  │     Last modified:     2020-12-15 14:19
  │     GUID:              c94770ca-1773-592c-b20a-e87243bc7cd0 ← TBT-00d4b070
  │     Device Flags:      • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Device stages updates
  │   
  ├─Package level of Dell dock:
  │     Device ID:         4e3369321c28610148028a97195dede6dab40986
  │     Summary:           A representation of dock update status
  │     Current version:   01.00.14.00
  │     Vendor:            Dell Inc. (USB:0x413C)
  │     Install Duration:  5 seconds
  │     Update Error:      A pending update will be completed next time the dock is unplugged from your computer
  │     GUID:              8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
  │     Device Flags:      • Supported on remote server
  │                        • Device can recover flash failures
  │                        • Device is usable for the duration of the update
  │   
  ├─RTS5413 in Dell dock:
  │     Device ID:         cadd482d8f2e6709ba268a8076df2c80f32088eb
  │     Summary:           USB 3.1 Generation 1 Hub
  │     Current version:   01.21
  │     Vendor:            Dell Inc. (USB:0x413C)
  │     Install Duration:  14 seconds
  │     Update Error:      A pending update will be completed next time the dock is unplugged from your computer
  │     GUIDs:             86fb40c0-8bf5-5a8b-a4ad-3156cf6bfaf4 ← USB\VID_413C&PID_B06F&REV_0101
  │                        b27d25f1-019d-5718-b41a-02ddaefe5577 ← USB\VID_413C&PID_B06F
  │                        ac5b774c-b49d-566b-9255-85f0f7f8a4ed ← USB\VID_413C&PID_B06F&hub
  │     Device Flags:      • Requires AC power
  │                        • Supported on remote server
  │                        • Device stages updates
  │                        • Device is usable for the duration of the update
  │   
  ├─RTS5487 in Dell dock:
  │     Device ID:         c71b6a87faba01b30b2a765c65131f264b11cd74
  │     Summary:           USB 3.1 Generation 2 Hub
  │     Current version:   01.47
  │     Vendor:            Dell Inc. (USB:0x413C)
  │     Install Duration:  3 seconds
  │     Update Error:      A pending update will be completed next time the dock is unplugged from your computer
  │     GUIDs:             707c63d2-e597-5c40-84db-9b1bb4c48d96 ← USB\VID_413C&PID_B06E&REV_0101
  │                        acfcd89b-105d-55b9-b85b-08bf8508f38c ← USB\VID_413C&PID_B06E
  │                        568ffa1e-a0db-5287-9ea3-872b60f7730b ← USB\VID_413C&PID_B06E&hub
  │     Device Flags:      • Requires AC power
  │                        • Supported on remote server
  │                        • Device stages updates
  │                        • Device is usable for the duration of the update
  │   
  └─VMM5331 in Dell dock:
        Device ID:         68e822bd516207129427e07f5bb058ed2f801bc5
        Summary:           Multi Stream Transport controller
        Current version:   05.04.03
        Vendor:            Dell Inc. (USB:0x413C)
        Install Duration:  6 minutes
        Update Error:      A pending update will be completed next time the dock is unplugged from your computer
        GUID:              89fec0b6-6b76-5008-b82c-5e5c6c164007 ← MST-panamera-vmm5331-259
        Device Flags:      • Requires AC power
                           • Supported on remote server
                           • Device stages updates
                           • Device is usable for the duration of the update
      
________________________________________________

Devices that were not updated correctly:

 • VMM5331 in Dell dock (5.03.10 → 05.04.03)
 • Thunderbolt controller in Dell dock (40.00 → 43.00)

Devices that have been updated successfully:

 • Package level of Dell dock (01.00.04.01 → 01.00.14.01)
 • VMM5331 in Dell dock (05.03.10 → 05.04.03)
```

**Dock SKU**
Please mention which module is installed in your WD19.

- [ ] WD19 (Single-C)
- [x ] WD19TB (Thunderbolt)
- [ ] WD19DC (Dual-C)

**Peripherals connected to the dock**
Please describe all devices connected to the dock.  Be as specific as possible,
including USB devices, hubs, monitors, and downstream type-C devices.

* HDMI Monitor
* GigE
* USB2 soundcard, webcam, mic, keyboard, mouse

**Verbose daemon logs**
First enable daemon verbose logs collection.
```shell
fwupdmgr modify-config "VerboseDomains" "*"
```

Then try to reproduce the issue.  Even if it doesn't reproduce, please attach the
daemon verbose logs collected from the system journal.
```shell
journalctl -b -u fwupd.service
```
[fwupd.log](https://github.com/fwupd/fwupd/files/5696472/fwupd.log)

**Additional questions**
- Operating system and version: Ubuntu 20.04
- Have you tried unplugging the dock or any peripherals from your machine? No
- Have you tried to power cycle the dock from the AC adapter? Yes
- Is this a regression? Unsure

joedborg, 2020-12-15:
Still an issue when running with `--force`

```
➜ sudo fwupdmgr update --force --verbose
(fwupdmgr:25784): GLib-DEBUG: 09:31:20.156: setenv()/putenv() are not thread-safe and should not be used after threads are created
Upgrade available for Thunderbolt controller in Dell dock from 40.00 to 43.00
Thunderbolt controller in Dell dock and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading 43.00 for Thunderbolt controller in Dell dock...
(fwupdmgr:25784): FuCommon-DEBUG: 09:31:22.345: creating path /root/.cache/fwupd
(fwupdmgr:25784): FuMain-DEBUG: 09:31:22.349: skpping download as file already exists
(fwupdmgr:25784): Fwupd-DEBUG: 09:31:22.351: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:22.363: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:25784): Fwupd-DEBUG: 09:31:22.417: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:22.419: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
(fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.410: Emitting ::status-changed() [device-write]
Updating Thunderbolt controller in Dell dock…                    ]
Writing…                 [  -                                    ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.412: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
(fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.413: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [                                       ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.415: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [                                       ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.417: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [*                                      ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.419: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [*                                      ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.422: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [*                                      ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.430: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [**                                     ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.430: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [**                                     ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.431: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [***                                    ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.432: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [***                                    ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.436: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [***                                    ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.437: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [****                                   ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.438: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [****                                   ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.445: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [*****                                  ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.446: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [*****                                  ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.447: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [*****                                  ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.447: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [******                                 ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.448: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [******                                 ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.448: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [*******                                ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.449: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [*******                                ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.449: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [*******                                ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.450: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
Writing…                 [********                               ](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.451: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
[... cut ...]
Writing…                 [***************************************](fwupdmgr:25784): Fwupd-DEBUG: 09:31:43.490: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
(fwupdmgr:25784): Fwupd-DEBUG: 09:32:24.375: Emitting ::status-changed() [idle]
Writing…                 [***************************************]
(fwupdmgr:25784): Fwupd-DEBUG: 09:32:24.375: Emitting ::device-changed(d06ed328732c6080e634a0bbb407e2fb0c7cd87d)
could not start thunderbolt device upgrade: could not write to 'nvm_authenticate': Input/output error
```

superm1, 2020-12-15:
Hmm.  Can you please share your `dmesg` output?

superm1, 2020-12-15:
Looking through the logs it's notable that there was a failure reading native mode too for you too. 
```
Dec 15 09:19:04 mia fwupd[8110]: 14:19:04:0112 FuPluginThunderbolt  failed to get native mode status: Error reading from file: Input/output error
```

joedborg, 2020-12-15:
> Looking through the logs it's notable that there was a failure reading native mode too for you too.
> 
> ```
> Dec 15 09:19:04 mia fwupd[8110]: 14:19:04:0112 FuPluginThunderbolt  failed to get native mode status: Error reading from file: Input/output error
> ```

This is all I see 

```
[ 2016.092453] hid-generic 0003:413C:B06E.0017: hiddev0,hidraw2: USB HID v1.11 Device [Dell dock] on usb-0000:3c:00.0-2.5/input0
```

joedborg, 2020-12-15:
> Looking through the logs it's notable that there was a failure reading native mode too for you too.
> 
> ```
> Dec 15 09:19:04 mia fwupd[8110]: 14:19:04:0112 FuPluginThunderbolt  failed to get native mode status: Error reading from file: Input/output error
> ```

What does this mean?

superm1, 2020-12-15:
> What does this mean?

That is a check on the host controller to read out the NVM and check an offset.  If you can still provide a full dmesg it would be preferable so we can see if anything else stands out that is non-obvious.

@YehezkelShB @westeri  any ideas?  This first error is from fwupd 1_3_X w/ a Ubuntu 5.4.x (before plugin overhaul or update on disconnect can be supported), which tried to fetch the NVM and compare it to new image and validation didn't pass.  The `--force` ignored validation and just tries writes image and then tries to trigger with `nvm_authenticate`.

westeri, 2020-12-15:
Are you able to read the NVM contents directly? Something like this:
```
# hexdump -C -n 1024 /sys/bus/bus/thunderbolt/devices/DEVICE/nvm_activeX/nvmem
```
Where `DEVICE` is `0-1` or `0-3` depending on which port you connected the dock and X is probably `0` or `1`.

joedborg, 2020-12-15:
```
sudo cat /sys/bus/thunderbolt/devices/0-3/device_name
WD19TB Thunderbolt Dock
```

hexdump just hangs until I try to do another `fwupdmgr update`, then I see
```
➜ sudo hexdump -C -n 1024  /sys/bus/thunderbolt/devices/0-3/nvm_active1/nvmem

hexdump: /sys/bus/thunderbolt/devices/0-3/nvm_active1/nvmem: Input/output error
```

superm1, 2020-12-15:
Based on the hang - is runtime PM getting stuck perhaps?

joedborg, 2020-12-15:
> Based on the hang - is runtime PM getting stuck perhaps?

Any advice on how I can debug / stop this?  Disable TLP for example?

superm1, 2020-12-15:
To see if that's it, you can toggle the sysfs knobs for the Thunderbolt NHI and bridges.  Powertop has a nice UI for it with a tunables tab you can find all of them and press _enter_  to toggle.

westeri, 2020-12-15:
You can do this
```
# echo on > /sys/bus/pci/devices/DEVICE/power/control
```
Where `DEVICE` is the Thunderbolt NHI (you can find it in lspci output).

Also full dmesg might help to debug this further (as Mario commented above).

joedborg, 2020-12-15:
Thanks @westeri 

Which of these would be the right choice?

```
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 08)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
00:1c.2 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #3 (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point LPC Controller/eSPI Controller (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
01:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS525A PCI Express Card Reader (rev 01)
02:00.0 Network controller: Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)
03:00.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
04:00.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
04:01.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
04:02.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
04:04.0 PCI bridge: Intel Corporation JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016] (rev 02)
05:00.0 System peripheral: Intel Corporation JHL6540 Thunderbolt 3 NHI (C step) [Alpine Ridge 4C 2016] (rev 02)
3a:00.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge DD 2018] (rev 06)
3b:02.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge DD 2018] (rev 06)
3b:04.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge DD 2018] (rev 06)
3c:00.0 USB controller: Intel Corporation JHL7540 Thunderbolt 3 USB Controller [Titan Ridge DD 2018] (rev 06)
6e:00.0 Non-Volatile memory controller: Toshiba Corporation Device 0116
```

What can I add to dmesg to give the level of verbosity required?

Thanks

superm1, 2020-12-15:
05:00.0 is the NHI

joedborg, 2020-12-15:
Thanks, `cat /sys/bus/pci/devices/0000:05:00.0/power/control` is "on" but still getting the error.

joedborg, 2020-12-15:
```
sudo dmesg -w -l debug 
[    0.000000] e820: update [mem 0x1a05f018-0x1a06f057] usable ==> usable
[    0.000000] e820: update [mem 0x1a05f018-0x1a06f057] usable ==> usable
[    0.000751] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000752] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000762] MTRR default type: write-back
[    0.000762] MTRR fixed ranges enabled:
[    0.000763]   00000-9FFFF write-back
[    0.000764]   A0000-BFFFF uncachable
[    0.000764]   C0000-FFFFF write-protect
[    0.000765] MTRR variable ranges enabled:
[    0.000766]   0 base 0080000000 mask 7F80000000 uncachable
[    0.000766]   1 base 0060000000 mask 7FE0000000 uncachable
[    0.000767]   2 base 0050000000 mask 7FF0000000 uncachable
[    0.000767]   3 base 004C000000 mask 7FFC000000 uncachable
[    0.000768]   4 base 004B000000 mask 7FFF000000 uncachable
[    0.000768]   5 disabled
[    0.000768]   6 disabled
[    0.000768]   7 disabled
[    0.000769]   8 disabled
[    0.000769]   9 disabled
[    0.007408] ACPI: Local APIC address 0xfee00000
[    0.008029] On node 0 totalpages: 4116177
[    0.008030]   DMA zone: 64 pages used for memmap
[    0.008031]   DMA zone: 21 pages reserved
[    0.008031]   DMA zone: 3996 pages, LIFO batch:0
[    0.008062]   DMA32 zone: 3933 pages used for memmap
[    0.008062]   DMA32 zone: 251701 pages, LIFO batch:63
[    0.013217]   Normal zone: 60320 pages used for memmap
[    0.013217]   Normal zone: 3860480 pages, LIFO batch:63
[    0.055676] ACPI: Local APIC address 0xfee00000
[    0.055719] ACPI: IRQ0 used by override.
[    0.055720] ACPI: IRQ9 used by override.
[    0.055726] e820: update [mem 0x39f79000-0x39fedfff] usable ==> reserved
[    0.056006] pcpu-alloc: s184320 r8192 d28672 u262144 alloc=1*2097152
[    0.056006] pcpu-alloc: [0] 0 1 2 3 4 5 6 7 
[    0.063213] Calgary: detecting Calgary via BIOS EBDA area
[    0.063215] Calgary: Unable to locate Rio Grande table in EBDA - bailing!
[    0.129450] spurious 8259A interrupt: IRQ7.
[    0.392876] libata version 3.00 loaded.
[    0.413108] PCI: pci_cache_line_size set to 64 bytes
[    0.417859] e820: reserve RAM buffer [mem 0x00058000-0x0005ffff]
[    0.417859] e820: reserve RAM buffer [mem 0x0009e000-0x0009ffff]
[    0.417860] e820: reserve RAM buffer [mem 0x1a05f018-0x1bffffff]
[    0.417860] e820: reserve RAM buffer [mem 0x2c337000-0x2fffffff]
[    0.417861] e820: reserve RAM buffer [mem 0x39f79000-0x3bffffff]
[    0.417861] e820: reserve RAM buffer [mem 0x3d936000-0x3fffffff]
[    0.417862] e820: reserve RAM buffer [mem 0x48e00000-0x4bffffff]
[    0.417863] e820: reserve RAM buffer [mem 0x4ae800000-0x4afffffff]
[    0.427266] system 00:00: Plug and Play ACPI device, IDs PNP0c02 (active)
[    0.427350] pnp 00:01: Plug and Play ACPI device, IDs PNP0b00 (active)
[    0.427384] system 00:02: Plug and Play ACPI device, IDs INT3f0d PNP0c02 (active)
[    0.427497] pnp 00:03: Plug and Play ACPI device, IDs PNP0303 (active)
[    0.427509] pnp 00:04: Plug and Play ACPI device, IDs DLL07e6 PNP0f13 (active)
[    0.427754] system 00:05: Plug and Play ACPI device, IDs PNP0c02 (active)
[    0.427797] system 00:06: Plug and Play ACPI device, IDs PNP0c02 (active)
[    0.428075] system 00:07: Plug and Play ACPI device, IDs PNP0c02 (active)
[    0.429093] system 00:08: Plug and Play ACPI device, IDs PNP0c02 (active)
[    0.652878] intel_idle: MWAIT substates: 0x11142120
[    0.652879] intel_idle: v0.4.1 model 0x8E
[    0.653359] intel_idle: lapic_timer_reliable_states 0xffffffff
[    0.988257] checking generic (50000000 1fa4000) vs hw (50000000 10000000)
[    4.326487] pci_bus 0000:04: Allocating resources
[   47.024416] ath: EEPROM regdomain: 0x6c
[   47.024417] ath: EEPROM indicates we should expect a direct regpair map
[   47.024418] ath: Country alpha2 being used: 00
[   47.024418] ath: Regpair used: 0x6c
[   61.895617] rfkill: input handler disabled
[   77.811619] rfkill: input handler enabled
[   84.327953] rfkill: input handler disabled
```

superm1, 2020-12-15:
Could you see if a mainline kernel reproduces?

joedborg, 2020-12-15:
> Could you see if a mainline kernel reproduces?

This is the standard one on Ubuntu 20.04.

superm1, 2020-12-15:
Yeah, I'm wondering if we're looking at something fixed between 5.4 and 5.10. if you can read NVM with 5.10.

joedborg, 2020-12-15:
I briefly tried to update to 5.10 using Mainline bit my system wouldn't boot after (luckily I have zfs snapshots enabled).  As such, I don't think I'll be able to test on 5.10 but fwupd updates have always worked in the past.

superm1, 2020-12-15:
Could you try with a newer Ubuntu kernel? Like grab from 21.04 repos? 5.4 is quite old.

superm1, 2020-12-16:
Here, grab this one: https://launchpad.net/ubuntu/+source/linux-oem-5.10

joedborg, 2020-12-16:
The most recent I could get to boot is 5.8.0 and the issue still persists.

superm1, 2020-12-16:
Hmm.  OK... Can you try to read the NVM of the host controller instead of the dock controller?  We might be able to at least take the dock out of the equation if that fails too.

superm1, 2020-12-16:
And btw - since we've got it narrowed down that it's a failure with accessing the NVM, I'll suggest another way you can get it fully upgraded in the interim.

If you connect it to the USB-C port on the other side of the machine it will allow you to upgrade using another non-TBT interface.  The upgrade is [much slower](https://github.com/fwupd/fwupd/blob/master/plugins/dell-dock/dell-dock.quirk#L113), but it should avoid this problem.

joedborg, 2020-12-16:
@superm1 thanks for the continuing support.

I seem to have the same issue:

```
➜ sudo cat /sys/bus/thunderbolt/devices/domain0/0-0/device_name 
XPS 13 9370

➜ sudo cat /sys/bus/thunderbolt/devices/domain0/0-0/nvm_active0/nvmem 
cat: /sys/bus/thunderbolt/devices/domain0/0-0/nvm_active0/nvmem: Input/output error
```

superm1, 2020-12-16:
OK so that confirms it's not actually a fwupd issue or a WD19TB dock issue.  I think this should be closed here and shifted over to a kernel bugzilla to conclude if it's a kernel bug or a problem with the TBT controller.

superm1, 2020-12-17:
As it's not a fwupd bug, but rather something wrong with host or kernel I'm closing it here.  If you open a kernel bug - please share it here.

closed 2020-12-17 by superm1

joedborg, 2020-12-17:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1908322

superm1, 2020-12-17:
I meant a kernel bugzilla: https://bugzilla.kernel.org/

joedborg, 2020-12-17:
https://bugzilla.kernel.org/show_bug.cgi?id=210761