fwupd/fwupd #2690: libfwupdplugin for 1.5.x hardcodes /lib64 path

Aerdan, 2020-12-16:
**Describe the bug**
`libfwupdplugin` expects `/lib64` to be unconditionally present, which in turn causes an assertion failure in its testsuite on 64-bit systems where `/lib64` does not exist.

**Steps to Reproduce**
1. Install Adélie Linux 1.0rc2.
2. Add [APK Fission](https://apkfission.net/install.html) with `adelie/current` distribution.
3. Install dependencies for `fwupd` package
4. Build fwupd
5. Run testsuite

**Expected behavior**
libfwupdplugin testsuite should pass

**fwupd version information**
1.5.3, building preparatory to packaging.

**Additional questions**
- [Adélie Linux 1.0rc](https://www.adelielinux.org)
- Problem did not exist in 1.4.2.

superm1, 2020-12-16:
Would you mind including a log to show exactly which test is failing?

Aerdan, 2020-12-16:
```
ERROR:../libfwupdplugin/fu-self-test.c:563:fu_common_firmware_builder_func: assertion failed (error == NULL): failed to build firmware: bwrap: Can't find source path /lib64: No such file or directory
```
[Full output](https://tpaste.us/BEWR)

superm1, 2020-12-16:
It looks like the error stems from this commit: b821557b35495e87b35d92e9f7948f399cf380d4

sroracle, 2020-12-16:
It should use `--ro-bind-try` instead of `--ro-bind`.

commit 3f9bf6c1 (2020-12-16): trivial: libfwupdplugin: don't fail to set up bwrap if /lib64 doesn't exist
Fixes: #2690
PR #2691 (2020-12-16): trivial: libfwupdplugin: don't fail to set up bwrap if /lib64 doesn't…
… exist

Fixes: #2690

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit ed4e9123 (2020-12-16): trivial: libfwupdplugin: don't fail to set up bwrap if /lib64 doesn't exist
Fixes: #2690
closed 2020-12-16 by hughsie