fwupd/fwupd #2695: Add test for detecting the "AMI Test PK" in the HSI

dnoliver, 2020-12-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Several device manufacturers decide to ship the default "AMI Test PK" platform key instead of a Device Manufacturer specific one. In this cases, the owner of the device (user or organization) that wants to enable Secure Boot can do 2 things:
1. Ask the device manufacturer to ship a system with a ODM specific key instead of the "AMI Test PK"
2. Perform a full Secure Boot Customization as described in https://media.defense.gov/2020/Sep/15/2002497594/-1/-1/0/CTR-UEFI-Secure-Boot-Customization-UOO168873-20.PDF

I think the HSI could help detect devices that are shipping this key by default and recommend a remediation.

*NOTE to self*: I have no idea why my system keeps booting in Secure Boot mode with an already expired Platform Key

```shell
[~]$ mokutil --pk
[key 1]
SHA1 Fingerprint: a7:73:11:3b:af:af:51:29:aa:83:fd:09:12:e9:5d:a4:fa:55:5f:91
Certificate:
Data:
Version: 3 (0x2)
Serial Number:
(Negative)1b:ed:93:e2:59:4e:2b:60:be:6b:1f:01:c9:af:a6:37
Signature Algorithm: sha256WithRSAEncryption
Issuer: CN=DO NOT TRUST - AMI Test PK
Validity
Not Before: Apr 30 22:50:15 2013 GMT
Not After : Apr 30 22:50:14 2017 GMT
Subject: CN=DO NOT TRUST - AMI Test PK
Subject Public Key Info:
Public Key Algorithm: rsaEncryption
RSA Public-Key: (2048 bit)
Modulus:
00:9c:18:19:79:cd:2e:05:c9:f5:29:24:5b:f8:8e:
7f:40:9a:91:70:be:14:d9:fe:a2:d8:fd:33:60:e0:
6f:57:78:f4:0a:44:2a:d8:f5:cb:23:04:67:7f:25:
0f:cc:27:77:6b:2f:8e:e3:1c:f8:ad:9a:33:b8:0c:
26:a4:d4:41:a2:38:56:dc:eb:74:33:e0:61:09:bf:
79:a9:bb:7c:72:f8:0d:22:fa:4e:b8:2e:ed:a7:da:
e3:2f:13:9b:14:96:e8:4a:8a:80:cd:85:63:88:4f:
d8:c8:cc:1d:9a:32:4f:db:a9:56:30:fe:79:89:e7:
fd:66:b0:3f:89:8f:4d:d7:23:8c:af:3b:83:23:8c:
b3:0b:2c:fc:a7:cb:9b:c2:2f:aa:12:1e:be:83:8c:
b3:2f:77:b9:0d:37:c3:59:12:2c:9f:4d:0d:37:47:
ad:a9:92:7a:14:e5:62:d1:22:7d:af:03:25:6d:b0:
64:e6:cf:f6:5e:88:f6:ae:f5:07:84:82:9f:4c:0d:
57:2f:55:f7:79:30:54:61:69:ef:8c:93:8f:ee:e4:
74:e6:29:c2:2c:1a:ef:19:78:42:d7:f0:dd:13:35:
3a:f1:65:bb:fc:11:a4:7e:74:3f:12:ff:dc:bf:bd:
e8:1f:1a:b4:a3:90:74:cf:20:92:eb:39:63:b5:b0:
a3:e9
Exponent: 65537 (0x10001)
X509v3 extensions:
2.5.29.1:
0M..d.i........"W0.n.'0%1#0!..U....DO NOT TRUST - AMI Test
PK....l.....A...6PY.
```

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

`fwupdmgr security --force` should fail HSI 1 if the "AMI Test PK" is present

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

The alternative is to run `mokutil --pk` and visually inspect that the AMI Test PK is not there

**Additional context**
Add any other context or screenshots about the feature request here.

I think this is very clear, AMI is saying that you should not trust that PK.

superm1, 2020-12-18:
I think this is a great idea to call out two things you noted:

1. Expired platform key
2. Development platform key

HSI1 makes sense to me too for these.

> NOTE to self: I have no idea why my system keeps booting in Secure Boot mode with an already expired Platform Key

From the design I don't think that an expired PK should prevent secure boot from running.  But it should prevent you from being able to update anything that is later signed by the PK.

hughsie, 2020-12-20:
I think this is a really good idea. Is the PK key a single X509 certificate? I did expect to do something like:

    gnutls_x509_crt_import (crt, &d, GNUTLS_X509_FMT_DER);

but got:

    crt_import: ASN1 parser: Error in TAG. [-73]

...any ideas?

hughsie, 2020-12-21:
Ahh, it's an EFI_SIG_LIST! Parsing works now. Could yo upload the AMI test PK somewhere please? Thanks!

hughsie, 2020-12-21:
As in, `cp /sys/firmware/efi/efivars/PK-8be4df61-93ca-11d2-aa0d-00e098032b8c /tmp/PK` then upload `PK` -- thanks.

dnoliver, 2020-12-22:
Uploading as zip because github does not allow me to upload the raw file :) 

[PK.zip](https://github.com/fwupd/fwupd/files/5731286/PK.zip)