fwupd/fwupd #2696: DFU Device stalled an unexpected request

amfern, 2020-12-21:
**Describe the bug**
I use the dfu-tool to flash inav firmeware on kakutef7 flightcontroller, up untill today it worked flawlsly, but after adding some code the flash size grew from `460647 B` to `527927 B` and i got an error `Device stalled an unexpected request`.
Note that flashing the same file using dfu-util or inav-configurator works perfectly fine.

can some one point me to a direction to what may cause this error?

**Steps to Reproduce**
```shell
$ dfu-tool write -v bin/KAKUTEF7.dfu

(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:04.971: writing sector at 0x808c800 (0x2048)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:04.975: refreshed status=OK and state=dfuDNBUSY (dnload=104)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:04.975: sleeping for 104ms…
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.079: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.079: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.079: writing sector at 0x808d000 (0x2048)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.084: refreshed status=OK and state=dfuDNBUSY (dnload=104)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.084: sleeping for 104ms…
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.188: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.188: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.188: writing sector at 0x808d800 (0x2048)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.193: refreshed status=OK and state=dfuDNBUSY (dnload=104)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.193: sleeping for 104ms…
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.297: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.297: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.297: setting percentage 97% of device-write
device-write:	97%
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.297: writing sector at 0x808e000 (0x2048)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.302: refreshed status=OK and state=dfuDNBUSY (dnload=104)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.302: sleeping for 104ms…
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.406: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.406: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.406: writing sector at 0x808e800 (0x2048)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.410: refreshed status=OK and state=dfuDNBUSY (dnload=104)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.411: sleeping for 104ms…
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.515: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.515: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.515: setting percentage 98% of device-write
device-write:	98%
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.515: writing sector at 0x808f000 (0x2048)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.519: refreshed status=OK and state=dfuDNBUSY (dnload=104)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.519: sleeping for 104ms…
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.624: refreshed status=errSTALLDPKT and state=dfuERROR (dnload=0)
(dfu-tool:3158957): FuPluginDfu-DEBUG: 11:55:05.624: refreshed status=errSTALLDPKT and state=dfuERROR (dnload=0)
Device stalled an unexpected request
```

**Expected behavior**
Expected the flashing process to be complete as with the smaller dfu file

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.5.1
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.1
$ dfu-tool --version  
fwupd 1.5.1
```

installed using source

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
G5 5590
[... 121 lines cut ...]
│                         302b14e9-d004-5371-b1fa-553d56913b9d
│                         12c60b98-52cd-500a-b99e-087aa6b47f0b
│                         db677673-1102-5481-a11e-6f397e123ced
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─WDC WD10SPZX-75Z10T2:
      Device ID:          5dbeb140337f610d54913c0f43d15fdaa0eafb51
      Summary:            ATA Drive
      Current version:    03.01A03
      Vendor:             Western Digital (ATA:0x101C)
      GUIDs:              febf540f-1153-54e3-ac56-ca0d9af64b30
                          ff3edd69-3118-58ea-b493-48ae036d2185
                          014f4a74-fe12-56de-8f58-e61b45232cb2
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

**Additional questions**
- Operating system and version: archlinux
- Have you tried rebooting? yes
- Is this a regression? no

hughsie, 2020-12-22:
> errSTALLEDPKT 

This means `Device stalled an unexpected request.` Looking at table A.2.6 in the DFU spec I see either:

 * _Host and device are not synchronized with respect to the quantity of data to be downloaded. The host must initiate recovery procedures. Device stalls the control pipe._
 * _Receipt of any other DFU class-specific request_

I suspect what's happening is that the device is out of space, and it's not letting you overwrite the bootloader that's located at the end of the flash device. Quite why it works with dfu-util I have no idea. Could you compare what dfu-tool and dfu-util are doing please? Thanks.

amfern, 2020-12-26:
Thanks, I did a quick comparison, and didn't find any difference, will dig into the code more thoroughly when have more time

amfern, 2021-01-05:
comparing the code of both tools, i can't spot any difference, 

### dfu-tool
```bash
$ dfu-tool -v write bin/KAKUTEF7.dfu
(dfu-tool:100636): FuCommon-DEBUG: 02:25:23.368: reading bin/KAKUTEF7.dfu with 522260 bytes
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.378: STM-DFU support
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.378: using DFU transfer size 0x0040 bytes
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.378: STM-DFU support
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.378: using DFU transfer size 0x0040 bytes
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.378: STM-DFU support
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.378: using DFU transfer size 0x0040 bytes
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.378: STM-DFU support
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.378: using DFU transfer size 0x0800 bytes
(dfu-tool:100636): FuDevice-DEBUG: 02:25:23.378: using b0a78eb71f4eeea7df8fb114522556ba8ce22074 for usb:01:00:01
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.380: Zone:0, Sec#:0, Addr:0x08000000, Size:0x8000, Caps:0x7 [REW]
Zone:0, Sec#:0, Addr:0x08008000, Size:0x8000, Caps:0x7 [REW]
Zone:0, Sec#:0, Addr:0x08010000, Size:0x8000, Caps:0x7 [REW]
Zone:0, Sec#:0, Addr:0x08018000, Size:0x8000, Caps:0x7 [REW]
Zone:0, Sec#:1, Addr:0x08020000, Size:0x20000, Caps:0x7 [REW]
Zone:0, Sec#:2, Addr:0x08040000, Size:0x40000, Caps:0x7 [REW]
Zone:0, Sec#:2, Addr:0x08080000, Size:0x40000, Caps:0x7 [REW]
Zone:0, Sec#:2, Addr:0x080c0000, Size:0x40000, Caps:0x7 [REW]
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.380: Zone:0, Sec#:0, Addr:0x1fff0000, Size:0x0020, Caps:0x3 [RW]
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.381: Zone:0, Sec#:0, Addr:0x1ff0f000, Size:0x0400, Caps:0x3 [RW]
Zone:0, Sec#:1, Addr:0x1ff0f400, Size:0x0020, Caps:0x3 [RW]
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.381: Zone:0, Sec#:0, Addr:0xffff0000, Size:0x0004, Caps:0x3 [RW]
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.381: refreshed status=errFIRMWARE and state=dfuERROR (dnload=0)
(dfu-tool:100636): FuDevice-DEBUG: 02:25:23.381: installing onto b0a78eb71f4eeea7df8fb114522556ba8ce22074:
FuFirmware:
  FuFirmwareImage:
  Data:                 0x7f814

(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.382: refreshed status=errFIRMWARE and state=dfuERROR (dnload=0)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.382: clearing error errFIRMWARE
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.391: downloading to target: @Internal Flash  /0x08000000/04*032Kg,01*128Kg,03*256Kg
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.391: downloading element at 0x8000000
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.391: marking sector 0x8000000-8008000 to be erased
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.391: setting action device-erase
device-erase:	0%
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.391: erasing sector at 0x8000000
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.392: refreshed status=OK and state=dfuDNBUSY (dnload=100)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.392: sleeping for 100ms…
device-write:	0%
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.749: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.749: doing actual check status
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.749: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.749: setting percentage 100% of device-erase
device-write:	100%
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.749: setting action idle
idle:	100%
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.749: setting action device-write
device-write:	100%
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.749: setting address to 0x8000000
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.750: refreshed status=OK and state=dfuDNBUSY (dnload=104)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.750: sleeping for 104ms…
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.854: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.854: doing actual check status
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.854: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.854: writing sector at 0x8000000 (0x1460)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.857: refreshed status=OK and state=dfuDNBUSY (dnload=104)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.857: sleeping for 104ms…
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.962: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.962: refreshed status=OK and state=dfuDNLOAD-IDLE (dnload=0)
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.962: setting percentage 0% of device-write
device-write:	0%
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.962: setting percentage 100% of device-write
device-write:	100%
(dfu-tool:100636): FuPluginDfu-DEBUG: 02:25:23.962: setting action idle
idle:	100%
[... 1838 lines cut ...]
 Download from image offset 00075000 to memory 08085000-080857ff, size 2048
 Download from image offset 00075800 to memory 08085800-08085fff, size 2048
 Download from image offset 00076000 to memory 08086000-080867ff, size 2048
 Download from image offset 00076800 to memory 08086800-08086fff, size 2048
 Download from image offset 00077000 to memory 08087000-080877ff, size 2048
 Download from image offset 00077800 to memory 08087800-08087fff, size 2048
[... cut ...]
 Download from image offset 0007d000 to memory 0808d000-0808d7ff, size 2048
 Download from image offset 0007d800 to memory 0808d800-0808dfff, size 2048
 Download from image offset 0007e000 to memory 0808e000-0808e7ff, size 2048
 Download from image offset 0007e800 to memory 0808e800-0808efff, size 2048
 Download from image offset 0007f000 to memory 0808f000-0808f122, size 291
Done parsing DfuSe file
Transitioning to dfuMANIFEST state
dfu-util: can't detach
Resetting USB to switch back to runtime mode```

hughsie, 2021-01-05:
A timing issue perhaps? How strange! Maybe the inav developers might have some ideas?

amfern, 2021-01-11:
I went over the INav configurator and there is a minor difference.

INav configurator divides the DFU firmware into blocks of 65536 bytes then it flashes the blocks with 2048 chunks, before flashing the next block, it updates the DFU write address and resets the wBlockNum to 2, so the wBlockNum value goes between 2 and 33 and never above 33 and never bellow 2.
The wBlockNum represents the chunk's write offset from the address we told DFU to write.

fwupd doesn't divide the firmeware into block, it just writes it as a single block, thus the wBlockNum goes above 256 and wraps around because of cast to (uint8) at line https://github.com/fwupd/fwupd/blob/96c079d9f0323f3ae745757b98e8b4a4b02dd20d/plugins/dfu/dfu-target-stm.c#L367
Which results in 0 which isn't allowed operation resulting in stalled response from the device. Removing the cast to uint8 solves the problem. 

dfu-util doesn't divide the firmware into blocks, similar to fwupd it writes it all as one block but uses int for the wBlockNum.

I opened a PR for this https://github.com/fwupd/fwupd/pull/2744

hughsie, 2021-01-11:
On Mon, 11 Jan 2021 at 16:18, Ilya guterman <notifications@github.com> wrote:
> INav configurator divides the DFU firmware into blocks of 65536 bytes then it flashes the blocks with 2048 chunks, before flashing the next block, it updates the DFU write address and resets the wBlockNum to 2, so the wBlockNum value goes between 2 and 33 and never above 33 and never bellow 2.

That's valid I guess. More USB traffic, but in the grand scheme of
things not even measurable.

> fwupd doesn't divide the firmeware into block, it just writes it as a single block, thus the wBlockNum goes above 256 and wraps around because of cast to (uint8) at line https://github.com/fwupd/fwupd/blob/96c079d9f0323f3ae745757b98e8b4a4b02dd20d/plugins/dfu/dfu-target-stm.c#L367

I have no idea why that cast is there; that makes no sense to me at all.

> Which results in 0 which isn't allowed operation resulting in stalled response from the device. Removing the cast to uint8 solves the problem.

Excellent, I'll merge it when all the CI tests have completed. Thanks
so much for debugging this!

Richard.

amfern, 2021-01-11:
Thank you, it was fun to debug :)

closed 2021-01-11 by amfern