fwupd/fwupd #2697: USB NVME enclosure

reagentoo, 2020-12-21:
**Describe the question**
I'm using IB-1817M-C31 + Samsung 970 pro ssd + linux machine (kernel v5.10.1).
Is it possible to use fwupd for it? Unfortunately, `get-devices` doesn't show my device in the list:
`fwupdmgr get-devices --show-all-devices`

**Additional info**
lsusb:
```
Bus 006 Device 002: ID 152d:0580 JMicron Technology Corp. / JMicron USA Technology Corp. USB 3.1 Storage Device
```

dmesg:
```
[  117.520380] usb 6-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[  117.554812] usb 6-2: New USB device found, idVendor=152d, idProduct=0580, bcdDevice=31.03
[  117.554814] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  117.554815] usb 6-2: Product: USB 3.1 Storage Device
[  117.554816] usb 6-2: Manufacturer: JMicron
[  117.554816] usb 6-2: SerialNumber: ----------------------
[  117.574392] scsi host8: uas
[  117.574721] scsi 8:0:0:0: Direct-Access     Samsung  SSD 970 PRO      3103 PQ: 0 ANSI: 6
[  117.575206] sd 8:0:0:0: Attached scsi generic sg3 type 0
[  119.923599] sd 8:0:0:0: [sdd] 2000409264 512-byte logical blocks: (1.02 TB/954 GiB)
[  119.923601] sd 8:0:0:0: [sdd] 4096-byte physical blocks
[  119.923715] sd 8:0:0:0: [sdd] Write Protect is off
[  119.923716] sd 8:0:0:0: [sdd] Mode Sense: 5f 00 00 08
[  119.923876] sd 8:0:0:0: [sdd] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[  119.924673] sd 8:0:0:0: [sdd] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes)
[  119.961194] sd 8:0:0:0: [sdd] Attached SCSI disk
```

S.M.A.R.T.:
```
# smartctl --all -d sntjmicron /dev/sdd
smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.10.1-gentoo-r1-x86_64] (local build)
Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       Samsung SSD 970 PRO 1TB
Serial Number:                      ---------------------------
Firmware Version:                   1B2QEXP7
PCI Vendor/Subsystem ID:            0x144d
IEEE OUI Identifier:                0x002538
Total NVM Capacity:                 1,024,209,543,168 [1.02 TB]
Unallocated NVM Capacity:           0
Controller ID:                      4
Number of Namespaces:               1
Namespace 1 Size/Capacity:          1,024,209,543,168 [1.02 TB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            002538 5501424998
Local Time is:                      Mon Dec 21 16:39:26 2020 MSK
Firmware Updates (0x16):            3 Slots, no Reset required
Optional Admin Commands (0x0037):   Security Format Frmw_DL Self_Test Directvs
Optional NVM Commands (0x005f):     Comp Wr_Unc DS_Mngmt Wr_Zero Sav/Sel_Feat Timestmp
Maximum Data Transfer Size:         512 Pages
Warning  Comp. Temp. Threshold:     81 Celsius
Critical Comp. Temp. Threshold:     81 Celsius

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     6.20W       -        -    0  0  0  0        0       0
 1 +     4.30W       -        -    1  1  1  1        0       0
 2 +     2.10W       -        -    2  2  2  2        0       0
 3 -   0.0400W       -        -    3  3  3  3      210    1200
 4 -   0.0050W       -        -    4  4  4  4     2000    8000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         0

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        43 Celsius
Available Spare:                    100%
Available Spare Threshold:          10%
Percentage Used:                    0%
Data Units Read:                    539,138 [276 GB]
Data Units Written:                 2,000,410 [1.02 TB]
Host Read Commands:                 4,212,272
Host Write Commands:                15,628,198
Controller Busy Time:               15
Power Cycles:                       9
Power On Hours:                     7
Unsafe Shutdowns:                   8
Media and Data Integrity Errors:    0
Error Information Log Entries:      0
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Temperature Sensor 1:               43 Celsius
Temperature Sensor 2:               50 Celsius

Error Information (NVMe Log 0x01, max 64 entries)
No Errors Logged
```

**fwupd version information**
```shell
# fwupdmgr --version
client version:	1.5.4
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.4
```

hughsie, 2020-12-21:
> USB 3.1 Storage Device

It looks like it exports a USB3 interface, not a PCI bridge. I don't think we can do anything in this case unfortunately.  You could ask the enclosure vendor if the USB converter chip has updates, but I don't think you'll be able to update the NVMe drive with this hardware, sorry.

closed 2020-12-21 by hughsie

reagentoo, 2020-12-23:
> USB 3.1 Storage Device

@hughsie Yes, but still it's a scsi_disk device. Therefore, scsi api is available to ssd.

hughsie, 2020-12-23:
I don't think the SCSI protocol includes support for firmware updates, correct? Even if it did, does NVMe support upgrades over SCSI?

reagentoo, 2020-12-23:
@hughsie To be honest, I am not competent in this matter.
I just assumed that the scsi protocol is enough to update the firmware. In the same way, it is enough for sata-devices.

mirh, 2021-01-07:
I just updated my Sabrent EC-DFLT enclosure (not sure if with JMS561 or JMS578, device id literally changed in the process), and in windows's procmon I spotted lots of `IOCTL_SCSI_PASS_THROUGH_DIRECT` commands. 
So, well, maybe it's vendor specific (haven't really read the spec tbf) but it can be done. 

I wouldn't know about the drive inside in its turn.

hughsie, 2021-01-07:
Ahh! You want to update the enclosure, not the NVMe disk perhaps?

mirh, 2021-01-07:
Idk, I'm not OP (which titled the issue that way though). 
Anyway, some/many/most nvme drives are actually supposed to receive firmware updates over SCSI. 
https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/upgrading-firmware-for-an-nvme-device

reagentoo, 2023-07-28:
@hughsie 
The specification of the controllers explicitly indicates the possibility of flashing the disk.
ASM2364 is ASMedia:
```
NVM Express Features:
...
    Support NVMe Firmware Update Process
```
https://www.asmedia.com.tw/product/BD5YqfdsPDqXFqi3/BF2yq24XzDuS5Tr4

I'll try to test the box with ASM2364 when i get it.
But currently fwupdmgr doesn't detect my Samsung 970 pro ssd inside JMicron JMS583 box.
(JMS583 is also not detected)