fwupd/fwupd #2698: Firmware update failed: system will not boot

madscientist, 2020-12-21:
**Describe the bug**
I have a Dell Mobile Precision 5520 XCTO laptop for a couple of years now.  It's currently running up-to-date Ubuntu 20.04.1.  I have done many firmware updates with fwupd with  no problems over the years.  I recently was notified of a new firmware update so I installed it and rebooted.

After the reboot it tried to apply the firmware update and got a failure message.  The system doesn't wait for any type of acknowledgement of the error so I had no opportunity to write it down and I don't know what it said.

The system tried to reboot and now it never succeeds: it appears that it can't find the boot image and drops into hardware check mode (no hardware issues are found).  I managed to take pictures then transcribe them below

My boot order is:
* Linux Firmware Updater -- filename \EFI\ubuntu\shimx64.efi
* Linux-Firmware-Updater \fwupx64.efi -- filename \EFI\ubuntu\shimx64.efi

The first boot gives the following errors:
```
Booting in insecure mode
Found update fwupd-ux-capsule
Found update fwupd-34578c72-11dc-4378bc7f-b643866f5590c-0
WARNING: No updates to process.  Called in error?
start_image() returned Invalid Parameter
```

the second boot gives these errors:
```
Booting in insecure mode
Failed to open \EFI\ubuntu\fwupx64.efi - Not Found
Failed to load image \EFI\ubuntu\fwupx64.efi: Not Found
start_image() returned Not Found
```

If I "add Boot Option" I can view the EFI files and I see that the \EFI\ubuntu\fwupx64.efi file does exist there.  Adding a new boot entry to use fwupx64.efi doesn't make any difference (it prints nothing just goes directly to hardware check).  I see a \EFI\ubuntu\grubx64.efi as well but if I add that into the boot order it doesn't do anything (just goes directly into hardware check).

Since I can't boot anymore I can't provide any more details such as versions, etc. except to say I'm using whatever the currently latest versions are on Ubuntu 20.04.x (my system was up to date before I rebooted).  Is there any sort of troubleshooting guide or any way to get my system to boot again?  I checked the website and didn't find anything like that.

madscientist, 2020-12-21:
OK I got my system to boot.  I guessed that the shim was finding the name of the next boot image from the name of the boot image, so I created a new boot option that had the name Linux-Firmware-Upda \grubx64.efi (I wasn't able to type the last 3 chars without shortening the initial name... not sure how the one that's there currently managed to work).

When I boot this it did allow my system to boot.  Obviously, something is still wrong here with the firmware updates.  I'm happy to help debug this if someone can give me guidance.  I can see I'm using fwupdate 1.3.11.

hughsie, 2020-12-21:
@superm1 can debug this after Christmas; I'm glad you got it limping along in the interim. It sounds more like a shim/bootloader issue rather than the Dell firmware but Mario might have more ideas.

madscientist, 2020-12-21:
One thing I should have mentioned: my disk is encrypted using full-disk encryption.  I'm not sure whose responsibility it is to ask for my encryption passphrase and set that up; if it is done via the shim that may explain why attempting to boot directly to grubx64.efi failed.  Or maybe it's something completely different.

superm1, 2021-01-04:
The capsule update is intentionally stored in the EFI system partition which is not encrypted with FDE as configured by the OS installer.  If you have a SED that's a different story..  So can you confirm if there is a SED in the box?

Can you please share your `efibootmgr -v` output so we can better understand what you have set?

madscientist, 2021-01-04:
I don't know what you mean by a SED so I can neither confirm nor deny it :).  I will say that I've had this system encrypted this way ever since I got it in 2018 and performed a number of firmware updates on it, so whatever setup it has, has worked in the past.  This Dell laptop was delivered with Ubuntu installed but I wiped it and reinstalled from scratch when I first got it, using Ubuntu's full disk encryption facility that is offered when installing.

I will also say that there is one difference between the last time I did a firmware upgrade and this time: I needed to install virtualbox on this system (to allow minikube to work) and when I installed it, it did give me a bunch of details about secure boot etc. which I didn't fully understand (bad, I know, but why does everything have to be so complicated??!?!?  I just want to run virtualbox images I don't want to change the host OS boot security).  The system continued to boot so I didn't pay too much attention but maybe there's some kind of partly-completed configuration or something?

Here is the bootmgr output.  Note that the boot image is the one I tricked out to invoke grub directly rather than the shim:
```
BootCurrent: 000A
Timeout: 0 seconds
BootOrder: 0000,000A,0001,0002,0009
Boot0000  Linux Firmware Updater        HD(1,GPT,b1074c59-665e-4a2d-8063-beb97048d7b7,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0001  Linux-Firmware-Updater \fwupx64.efi   HD(1,GPT,b1074c59-665e-4a2d-8063-beb97048d7b7,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0002  test-fwup     PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-03-00-3B-6B-77)/HD(1,GPT,b1074c59-665e-4a2d-8063-beb97048d7b7,0x800,0x100000)/File(\EFI\ubuntu\fwupdx64.efi)
Boot0003* USB Storage Device    BBS(USB,USB Storage Device,0x0)..BO
Boot0004* CD/DVD/CD-RW Drive    BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0005* Onboard NIC   BBS(Network,Onboard NIC,0x0)..BO
Boot0006* USB NIC       BBS(128,Realtek PXE B00 D14,0x0)..BO
Boot0007* UEFI: KXG50ZNV512G NVMe TOSHIBA 512GB, Partition 1    HD(1,GPT,ea59192c-bceb-4a2e-8a2a-718a539ffb13,0x800,0x177000)/File(EFI\boot\bootx64.efi)..BO
Boot0008* M.2 PCIe SSD  BBS(HD,P0: KXG50ZNV512G NVMe TOSHIBA 512GB,0x0)..BO
Boot0009  ubuntu-grub   PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-03-00-3B-6B-77)/HD(1,GPT,b1074c59-665e-4a2d-8063-beb97048d7b7,0x800,0x100000)/File(\EFI\ubuntu\grubx64.efi)
Boot000A* Linux-Firmware-Upda \grubx64.efi      PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-03-00-3B-6B-77)/HD(1,GPT,b1074c59-665e-4a2d-8063-beb97048d7b7,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
```

madscientist, 2021-01-04:
Just to be clear, then *test-fwup* and *ubuntu-grub* boots were ones I added to try to get things working, that never worked.

superm1, 2021-01-04:
> I don't know what you mean by a SED so I can neither confirm nor deny it :). I will say that I've had this system encrypted this way ever since I got it in 2018 and performed a number of firmware [...]

From what you describe, it sounds like you're using the FDE that the Ubuntu installer offers not a Self Encrypting Drive (SED).  OK.

> I will also say that there is one difference between the last time I did a firmware upgrade and this time: I needed to install virtualbox on this system (to allow minikube to work) and when I [...]

Did you actually enable secure boot in BIOS setup at this time too? 

>BootCurrent: 000A
Timeout: 0 seconds
BootOrder: 0000,000A,0001,0002,0009
Boot0000  Linux Firmware Updater        HD(1,GPT,b1074c59-665e-4a2d-8063-beb97048d7b7,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0001  Linux-Firmware-Updater \fwupx64.efi   HD(1,GPT,b1074c59-665e-4a2d-8063-beb97048d7b7,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0002  test-fwup     PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-03-00-3B-6B-77)/HD(1,GPT,b1074c59-665e-4a2d-8063-beb97048d7b7,0x800,0x100000)/File(\EFI\ubuntu\fwupdx64.efi)
Boot0003* USB Storage Device    BBS(USB,USB Storage Device,0x0)..BO
Boot0004* CD/DVD/CD-RW Drive    BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0005* Onboard NIC   BBS(Network,Onboard NIC,0x0)..BO
Boot0006* USB NIC       BBS(128,Realtek PXE B00 D14,0x0)..BO
Boot0007* UEFI: KXG50ZNV512G NVMe TOSHIBA 512GB, Partition 1    HD(1,GPT,ea59192c-bceb-4a2e-8a2a-718a539ffb13,0x800,0x177000)/File(EFI\boot\bootx64.efi)..BO
Boot0008* M.2 PCIe SSD  BBS(HD,P0: KXG50ZNV512G NVMe TOSHIBA 512GB,0x0)..BO
Boot0009  ubuntu-grub   PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-03-00-3B-6B-77)/HD(1,GPT,b1074c59-665e-4a2d-8063-beb97048d7b7,0x800,0x100000)/File(\EFI\ubuntu\grubx64.efi)
Boot000A* Linux-Firmware-Upda \grubx64.efi      PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-03-00-3B-6B-77)/HD(1,GPT,b1074c59-665e-4a2d-8063-beb97048d7b7,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)

My suspicion is there is some de-duplication happening at BIOS level that is confusing things since you have a bunch of entries pointing to the same files.  

The most concerning for this project are `Boot0000` and `Boot0001`.  A long time ago we transitioned from Linux-Firmware-Updater to "Linux Firmware Updater".  The former only used by systems that had a potentially broken BIOS (you can see in the flags [here](https://github.com/fwupd/fwupd/blob/6a489a9eaf623a61df72e951a4df000468b8caff/plugins/uefi/fu-uefi-device.c#L597)).

We used to clean up the entries automatically after updates.  If you ran an update using fwupd 1.3.7 or older you should have had both entries removed:
https://github.com/fwupd/fwupd/commit/aa6498866ce4d1dd8068f359fd5dabbe343d394c

However in fwupd 1.3.8 and later we stopped doing that: https://github.com/fwupd/fwupd/commit/60373e03fd0668d379ba0d7b90a6f02ba7d87478#diff-2d35a59cd85a8a03c44537f89bf59f53ce8590367983def82400c1bf79c7521c

So I hypothesize you did an update with a really old fwupd, older than 1.1.0.  Then you didn't do any update between 1.2.4 and 1.3.8.  Then you did another update after 1.3.8.  So you ended up with both "Linux Firmware Updater" and "Linux-Firmware-Updater" entries.

>BootCurrent: 000A

I notice you're actually booting into 000A, which is funny in it's own too.  From the fwupd perspective the only thing we really can do to help this situation is to add some code back in to remove the old `Linux-Firmware-Updater` entry if you're supposed to be using `Linux Firmware Updater`.  However I'm not sure it's worth the bytes since this is a corner case.

So I think my suggestion now is you manually do this:
1. Remove `Boot0000`, `Boot0001`, `Boot0001`, `Boot0002`, `Boot0009`, `Boot000A` using `efibootmgr`
2. Run `sudo update-grub` and it should make you a new entry for Ubuntu.  If it doesn't run `sudo grub-install` and it should
3. Reboot and make sure you can boot properly.
3. fwupd will make it's own new entry on your next firmware update.

@hughsie what you think?

madscientist, 2021-01-04:
Correct I don't believe I have a self-encrypting drive.

Note boot 000A is one I added by hand, from the UEFI boot menu provided by the Dell firmware (not using efibootmgr, because my system wouldn't boot so I couldn't use efibootmgr).  Same with 0002 and 0009.  The arguments that are present in 0002, 0009, and 000A are included by default by the Dell firmware and I was not given any way to edit them.

This is also why the name on 000A starts with **Linux-Firmware-Upda**,  because the UEFI boot creation menu wouldn't allow the full string as a name for the boot option so I had to truncate it.

Regarding the provenance of this system: it was shipped to me from Dell with Ubuntu 16.04 LTS on it.  When I received it I wiped the disk and installed Ubuntu 18.04 LTS, with FDE, from an ISO image.  Since then (sometime last summer) I performed an in-place upgrade from Ubuntu 18.04 LTS to Ubuntu 20.04 LTS (not a wipe/reinstall).  It's possible that one of these processes didn't update the boot configuration as expected.

I can't be 100% certain but I _think_ I have successfully installed firmware updates since I upgraded to 20.04.  Is there anything that tracks this I could query?

Regarding enabling secure boot: to the best of my understanding I did not enable secure boot.  When I boot it says "booting in insecure mode" and it asks for my disk encryption passphrase but doesn't ask for any other password info.  Is there a way to check the state of secure boot?

I will wait a bit to see if @hughsie has thoughts before performing the steps you suggest.

Thanks!

PS. I should mention I see another firmware update ready to download but I'm not going to do that until I get my system booting using "standard" fwupd :)

superm1, 2021-01-04:
> Regarding the provenance of this system: it was shipped to me from Dell with Ubuntu 16.04 LTS on it. When I received it I wiped the disk and installed Ubuntu 18.04 LTS, with FDE, from an ISO image. [...]

Ubuntu 16.04 shipped with fwupd 0.7.x and upgraded in the field to 0.8.x.  Ubuntu 18.04 shipped with fwupd 1.0.6 and later upgraded to 1.2.10 (much later).  Ubuntu 20.04 shipped with 1.3.9 and upgraded to 1.3.11 a few months after launching.  So If I was to guess you ran updates on Ubuntu 18.04 while it still had 1.0.6, and upgraded to 20.04 some time before 1.2.10 was backported.  Then that's how you hit the corner case.  Someone installing Ubuntu 18.04 right now and then upgrading to ubuntu 20.04 would get 1.2.10 in 18.04 and then 1.3.11 in 20.04.

>Regarding enabling secure boot: to the best of my understanding I did not enable secure boot. When I boot it says "booting in insecure mode" and it asks for my disk encryption passphrase but doesn't ask for any other password info. Is there a way to check the state of secure boot?

OK, so secure boot doesn't play into it then.

>I can't be 100% certain but I think I have successfully installed firmware updates since I upgraded to 20.04. Is there anything that tracks this I could query?

Looking at the versions available in the Ubuntu archives solidifies my hypothesis in my mind.

>PS. I should mention I see another firmware update ready to download but I'm not going to do that until I get my system booting using "standard" fwupd :)

Well that's great.  I think if you manually clean up all the entries we can see if everything is in good shape for you.

hughsie, 2021-01-04:
> @hughsie what you think?

Sounds sane to me.

madscientist, 2021-01-06:
Thanks all; I will get around to this sometime in the next couple of days.  Stay tuned!

madscientist, 2021-01-17:
Sorry for the delay!  A few days turned into a week, etc.

I followed the instructions above and all worked well.  Note that `update-grub` didn't install a new boot entry, but `grub-install` created an `ubuntu` entry.

Rebooting worked fine using that, and I then installed the latest firmware update and that applied OK as well and my system continues to boot normally.

Thanks!

closed 2021-01-17 by madscientist

othonb, 2021-09-30:
> OK I got my system to boot. I guessed that the shim was finding the name of the next boot image from the name of the boot image, so I created a new boot option that had the name Linux-Firmware-Upda [...]

Thank you very much! It solved a related problem in my Dell Latitude here. \EFI\ubuntu\grubx64.efi saved my day...