fwupd/fwupd #2708: Please keep https://github.com/fwupd/fwupd/issues/391# open until fixed permanently

axtimwalde, 2020-12-29:
This closed bug https://github.com/fwupd/fwupd/issues/391# keeps happening.

axtimwalde, 2020-12-29:
As of now works again.  That was fast!  Thanks a lot!

ben2talk, 2020-12-30:
```
ournalctl -b -u fwupd
-- Logs begin at Sun 2020-12-27 12:01:01 +07, end at Wed 2020-12-30 16:19:41 +07. --
Dec 30 16:03:42 Evo systemd[1]: Starting Firmware update daemon...
Dec 30 16:03:42 Evo fwupd[10958]: Failed to load engine: Failed to load config: Key file does not start with >
Dec 30 16:03:42 Evo systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
Dec 30 16:03:42 Evo systemd[1]: fwupd.service: Failed with result 'exit-code'.
Dec 30 16:03:42 Evo systemd[1]: Failed to start Firmware update daemon.
Dec 30 16:18:29 Evo systemd[1]: Starting Firmware update daemon...
Dec 30 16:18:29 Evo fwupd[20455]: Failed to load engine: Failed to load config: Key file does not start with >
Dec 30 16:18:29 Evo systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
Dec 30 16:18:29 Evo systemd[1]: fwupd.service: Failed with result 'exit-code'.
Dec 30 16:18:29 Evo systemd[1]: Failed to start Firmware update daemon.

```

hughsie, 2020-12-30:
@axtimwalde I did nothing; the CDN just took some time to invalidate. If you use a newer fwupd version we use JCat to avoid this race.

@ben2talk you need to check /etc/fwupd/daemon.conf and open a new bug if it looks like a valid config file.

closed 2020-12-30 by hughsie

ben2talk, 2020-12-30:
@hugsie You were right. I restored one from two months ago, added the line to disable UEFI checks and it ran smoothly.

saschaarthur, 2021-01-12:
+1 its bugged again..

superm1, 2021-01-12:
The currently agreed upon policy for Ubuntu fwupd support is documented here: https://wiki.ubuntu.com/firmware-updates
Changing that will require the technical board approval.  In this case the new dependency `libjcat` needs to also be promoted to _"main"_ in Ubuntu 20.04 for fwupd 1.4.x or 1.5.x to be brought in.  It's currently in _"universe"_ for 20.04.  It's in _"main"_ in 20.10 and newer.

So in summary yes fwupd 1.3.x is what's in Ubuntu 20.04 and will be for now.  If you're on Ubuntu 20.04 (or fwupd 1.3.x  or older elsewhere) there is nothing that can be done other than wait a few minutes and retry.