fwupd/fwupd #2713: OUI quirk required

heynnema, 2020-12-31:
Ubuntu 20.10

external USB disk enclosure with SSD

Dec 31 10:36:59 Satellite-E55 fwupd[3401]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
Dec 31 10:36:59 Satellite-E55 fwupd[3401]: OUI quirk required, please see https://github.com/fwupd/fwupd/wiki/ATA-Disk:-OUI-Quirk-Required!
Dec 31 10:36:59 Satellite-E55 fwupd[3401]: ---
Dec 31 10:36:59 Satellite-E55 fwupd[3401]: [DeviceInstanceId=OUI\888914]
Dec 31 10:36:59 Satellite-E55 fwupd[3401]: Vendor = FIXME
Dec 31 10:36:59 Satellite-E55 fwupd[3401]: VendorId = ATA:UNKNOWN
Dec 31 10:36:59 Satellite-E55 fwupd[3401]: ---

hughsie, 2021-01-01:
Did you try https://github.com/fwupd/fwupd/wiki/ATA-Disk:-OUI-Quirk-Required ?

heynnema, 2021-01-01:
Yes. That's what got me here. *"If you know how, please submit a pull request with the oui quirk added in the right place in plugins/ata/ata.quirk -- the file should be roughly ordered according to the OUI ID value. If this sounds too complicated, please just open an issue here https://github.com/fwupd/fwupd/issues and paste everything between the --- marks."*

I couldn't find the proper information, and I didn't know what file to edit. The specified location isn't a full path to the file.

heynnema, 2021-01-01:
88-89-14   (hex)		All Components Incorporated
888914     (base 16)		All Components Incorporated
				828 New Meister Lane
				Pflugerville  TX  78660
				US

heynnema, 2021-01-04:
@hughsie just a ping

hughsie, 2021-01-04:
@heynnema we've trying to make this manual step unnecessary here: https://github.com/fwupd/fwupd/pull/2716

hughsie, 2021-01-10:
I don't think we need this now we've merged https://github.com/fwupd/fwupd/pull/2716 and also updated the wiki. Thanks tho!

closed 2021-01-10 by hughsie