fwupd/fwupd #2714: FuPluginUefi Error: opening directory “/sys/firmware/efi/esrt/entries”: No such file or directory

LinuxOnTheDesktop, 2021-01-03:
**Describe the bug**
Perhaps no bug as such, but I see this in the syslog:

```
03:51:12:000 systemd[1]  Starting Firmware update daemon...
03:51:12:0859 FuPluginUefi Error opening directory “/sys/firmware/efi/esrt/entries”: No such file or directory
03:51:12:0883 FuEngine device 602b0a6cc821d155208724f0e22f8d111542b74c [INTEL SSDSC2BF240A4L] does not define a vendor-id!
```

**Steps to Reproduce**
Unknown.

**Expected behavior**
I expect no error message.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
2325L19
│
├─3rd Gen Core processor Graphics Controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     09
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               07f8a988-047c-58f7-98b2-f848a5075c94
│                          a0805f65-b696-58cb-b474-1d805df497e8
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─INTEL SSDSC2BF240A4L:
│     Device ID:           602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:             ATA Drive
│     Current version:     LSTi
│     GUIDs:               fa9716f0-3995-5125-b1e7-a1f985da75c3
│                          8d8de2b2-a27a-51d8-be86-27be50338782
│                          2055db9a-2f41-5027-9a78-a0c5cf3c4033
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
│ │   Current version:     G2ETB6WW (2.76 )
│ │   Vendor:              LENOVO
│ │   Update Error:        UEFI Capsule updates not available or enabled
│ │   GUID:                230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │   Device Flags:        • Internal device
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │ 
│ └─Intel AMT [unprovisioned]:
│       Device ID:         e2623122c99d58220498aacbfcfdb1baebbae3c5
[... 357 lines cut ...]
│       └── zstd.mod
├── initrd.img -> initrd.img-5.9.16-050916-generic
├── initrd.img-5.4.0-58-generic
├── initrd.img-5.8.0-33-generic
├── initrd.img-5.8.18-050818-generic
├── initrd.img-5.9.16-050916-generic
├── initrd.img.old -> initrd.img-5.8.18-050818-generic
├── System.map-5.4.0-58-generic
├── System.map-5.8.0-33-generic
├── System.map-5.8.18-050818-generic
├── System.map-5.9.16-050916-generic
├── vmlinuz -> vmlinuz-5.9.16-050916-generic
├── vmlinuz-5.4.0-58-generic
├── vmlinuz-5.8.0-33-generic
├── vmlinuz-5.8.18-050818-generic
├── vmlinuz-5.9.16-050916-generic
└── vmlinuz.old -> vmlinuz-5.8.18-050818-generic
```

**Additional questions**
- Operating system and version:
- Have you tried rebooting? No - for, if I should not see the message then I should _never_ see it.
- Is this a regression? Unknown.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? No.

hughsie, 2021-01-03:
> Error opening directory “/sys/firmware/efi/esrt/entries”: No such file or directory

Is _UEFI Capsule Update_ enabled in your BIOS?

LinuxOnTheDesktop, 2021-01-03:
The computer is a ThinkPad X230. Its BIOS seems to lack that option. The BIOS does have (under Security -> UEFI BIOS Update Option') 'Flash BIOS Updating by End-Users', which is set to 'Enabled', and also (on the same 'page') 'Secure RollBack Prevention', also set to 'Enabled'.

hughsie, 2021-01-03:
Could you see if the /sys/firmware/efi/esrt/entries appears if you use a Fedora 33 LiveUSB disk? The BIOS options sound fine, so it's probably a kernel or kernel config problem.

LinuxOnTheDesktop, 2021-01-03:
Does that LiveUSB include fwupd? Either way, it is quite a bit of hassle; but I will do it if you really think I should.

Also, I can tell you the following. 

- I have the problem on kernel 5.9.16.
- I have the problem on kernel 5.8.18.
- The message appears after `fwupdmgr refresh`.
- I see a message at boot - using either kernel - saying 'couldn't parse dbx signatures'.

hughsie, 2021-01-03:
> Does that LiveUSB include fwupd

Yes.

> I see a message at boot - using either kernel - saying 'couldn't parse dbx signatures'.

Hmm, I wonder if the firmware is borked somehow.

LinuxOnTheDesktop, 2021-01-03:
I have booted with the live Fedora image. I am having some trouble determining whether that image generates the error or not. For, initially, the image was set to the wrong time zone; I see the old error in the log, marked as occurring last night, but upon first booting the image all the times in the log were wrong. Still, I have run `fwupdmgr refresh --force` a few times and I see the error message only once. Also, though, fwupdate says, of the Fedora system, that I have zero local devices supported, whereas previously - under Mint - the tally was one.

As to firmware being borked: I find it is very easy, in trying to either get secure boot working or to disable it, for its inscrutable systems to get in a terrible mess. That might have happened on this machine. I tried just now to fix things by clearing the secure boot keys in the BIOS, and perhaps changing some setting using the 'Mokutil' (that utility does not always tell the user what an option's current state is - and that is not the grossest of its UI failings). And now (as a result of that, I presume) I no longer see  'couldn't parse dbx signatures' at boot; but every fwupd 'refresh' still generates the 'No such file or directory' error.

hughsie, 2021-01-04:
> I am having some trouble determining whether that image generates the error or not

Does the live Fedora image have /sys/firmware/efi/esrt/entries ?

LinuxOnTheDesktop, 2021-01-04:
No: the Fedora live image _lacks_ /sys/firmware/efi/esrt. It does have /sys/firmware/efi.

So far as I can tell, the Fedora live image does _not_, after `refresh`, write any error messages to the log.

Running under Mint, there is no  /sys/firmware/efi/esrt, let alone /sys/firmware/efi/esrt/entries.

hughsie, 2021-01-10:
>  I expect no error message.

closed 2021-01-10 by hughsie