fwupd/fwupd #2731: Dell WD19 TB - upgrade report which got uploaded says update failed, but it updated just fine

madsl, 2021-01-05:
**Describe the bug**
Updated my WD19TB today successfully, but I noticed that the report that got uploaded said that the thunderbolt controller did not successfully update, although it did.

**Report from fwupdmgr**
```
~ # fwupdmgr get-history
WARNING: This package has not been validated, it may not work properly.
XPS 15 9570
│
├─System Firmware:
│ │   Device ID:          66352e92895dc6d018071ad4ecfae249b0b8af0d
│ │   Previous version:   1.16.2
│ │   Update State:       success
│ │   Last modified:      2020-12-31 14:33
│ │   GUID:               e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─XPS 15 9570/Precision 5530 System Update:
│       New version:      1.17.1
│       Remote ID:        lvfs
│       Summary:          Firmware for the Dell XPS 15 9570/Precision 5530
│       License:          Proprietary
│       Size:             18,6 MB
│       Created:          2020-07-09
│       Urgency:          Critical
│       Vendor:           Dell Inc.
│       Description:      
│       This stable release fixes the following issues:
│       
│       • Firmware updates to address security advisories
│       • Fixed the issue where the system Master Password does not work.
│       
│       Security issues fixed:
│       
│       • CVE-2020-0531
│       • CVE-2020-0532
│       • CVE-2020-0534
│       • CVE-2020-0535
│       • CVE-2020-0536
│       • CVE-2020-0537
│       • CVE-2020-0538
│       • CVE-2020-0539
│       • CVE-2020-0540
│       • CVE-2020-0541
│       • CVE-2020-0542
│       • CVE-2020-0594
│       • CVE-2020-0595
│       • CVE-2020-0596
│       • CVE-2020-8674
│       • CVE-2020-0528
│       • CVE-2020-0529
│       • CVE-2020-0543
│       • CVE-2020-0548
│       • CVE-2020-0549
│       • CVE-2020-5362
│       • CVE-2020-8758
│       • INTEL-SA-00295
│       • INTEL-SA-00322
│       • INTEL-SA-00320
│       • INTEL-SA-00329
│       • INTEL-TA-00404
│     
├─Package level of Dell dock:
[... 261 lines cut ...]

**fwupd version information**
Please provide the version of the daemon and client.
```
~ # fwupdmgr --version
client version: 1.5.4
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.4
```

**Dock SKU**
Please mention which module is installed in your WD19.

- [ ] WD19 (Single-C)
- [ ] WD19TB (Thunderbolt)
- [X] WD19DC (Dual-C)

**Peripherals connected to the dock**
- Monitor running 1080p over DP
- Logitech keyboard (unifying receiver, did not receive an update during this session)
- Microsoft mouse

Running Gentoo Linux, kernel 5.9.16.

superm1, 2021-01-05:
Do you know for sure the update was performed while fwupd 1.5.4 was installed?  Or is it possible it was something older?  (It was supposed to be fixed in 1.5.3..: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-WD19TB:-Device-version-not-updated-on-success)

madsl, 2021-01-05:
```
# LC_ALL=C date -d @$(</var/db/pkg/sys-apps/fwupd-1.5.4/BUILD_TIME)
Tue Dec 29 18:15:11 CET 2020
```

superm1, 2021-01-05:
Hmm OK.  So something else we're missing still for this.  @hughsie any more ideas?  It's the whole delayed upgrade flow.

madsl, 2021-01-05:
The fwupd daemon got unresponsive after the update, by the way, I just recalled. Sorry, I should've written that in the original bug report. If I remember right I had to kill -9 fwupd and then systemctl restart fwupd to get it up again (didn't respond to systemctl stop fwupd)

madsl, 2021-01-05:
The logs don't say much, here you see that I killed it with -9 when it didn't respond to stop:
```
jan. 05 12:17:32 latentcall systemd[1]: Starting Firmware update daemon...
jan. 05 12:17:34 latentcall systemd[1]: Started Firmware update daemon.
jan. 05 12:17:37 latentcall fwupd[6450]: 11:17:37:0371 FuPluginLinuxLockdown could not open /sys/kernel/security/lockdown: Error opening file /sys/kernel/security/lockdown: No such file or>
jan. 05 12:19:33 latentcall fwupd[6450]: 11:19:33:0433 FuDevice             disabling polling: failed to open /dev/hidraw3
jan. 05 12:25:11 latentcall systemd[1]: Stopping Firmware update daemon...
jan. 05 12:26:13 latentcall systemd[1]: fwupd.service: Main process exited, code=killed, status=9/KILL
jan. 05 12:26:13 latentcall systemd[1]: fwupd.service: Failed with result 'signal'.
jan. 05 12:26:13 latentcall systemd[1]: Stopped Firmware update daemon.
jan. 05 12:26:13 latentcall systemd[1]: fwupd.service: Consumed 7.123s CPU time.
```

superm1, 2021-01-05:
>The fwupd daemon got unresponsive after the update, by the way, I just recalled. Sorry, I should've written that in the original bug report. If I remember right I had to kill -9 fwupd and then systemctl restart fwupd to get it up again (didn't respond to systemctl stop fwupd)

Oh really?  Darn, I would have liked to see a backtrace where it was spinning if it was still spinning on something.

> Hmm OK. So something else we're missing still for this. @hughsie any more ideas? It's the whole delayed upgrade flow.

I think I might have caught it.  I'll send up some PR.  As @madsl  found It's a purely cosmetic error.

commit 3165e94f (2021-01-06): trivial: fu-engine: correct a logic error in the daemon for thunderbolt

WD19TB uses skip-restart in some cases, but not all.
The matrix of cases is enumerated in 834b28009d7e8fab9d37621456973a51a521c284

Unfortunately in the most common case now - new kernel and new daemon
`skip-restart` *isn't* used.  The device should be left in a `needs-activation`
state though.

Use this to skip the trigger of failed upload report.

Fixes: #2731
closed 2021-01-06 by hughsie