fwupd/fwupd #2748: fix config section in example uefi_capsule.conf

gdamjan, 2021-01-12:
**Describe the bug**
 The config section name in the example [uefi_capsule.conf](https://github.com/fwupd/fwupd/blob/master/plugins/uefi-capsule/uefi_capsule.conf) file needs to changed to `[uefi_capsule]`, since the change of the plugin name.

**Steps to Reproduce**

Install fwupd, edit `/etc/fwupd/uefi_capsule.conf` to un-comment (for ex. `DisableShimForSecureBoot=true`), the configuration is not honored.

**fwupd version information**

```shell
$ fwupdmgr --version
client version: 1.5.5
compile-time dependency versions
        gusb:   0.3.5

Unknown                  [***************************************]
daemon version: 1.5.5
```

```
$ pacman -Qs fwupd
local/fwupd 1.5.5-1
    Simple daemon to allow session software to update firmware
```

superm1, 2021-01-12:
@gdamjan thanks for filing. Would you mind submitting a PR?
@hughsie Bessides the starting template conf, do we need to think about this for upgrades at all you think?  Some sort of migration from `[uefi]` to `[uefi_capsule`?

closed 2021-01-12 by hughsie

hughsie, 2021-01-12:
@superm1 I think migrations like this are probably handled better by the packagers rather than "upstream" -- and certainly a `sed` could be done in the debian scripts.

PR #2749 (2021-01-12): rename config section in uefi_capsule.conf to plugin name
in ee2e2c36749298e58b34dca163ea48a7fc925da6 the plugin name was changed
from uefi to uefi_capsule. while the config file name was changed, the
section name should also be changed.

fixes #2748

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit bb8c8040 (2021-01-12): rename config section in uefi_capsule.conf to plugin name

in ee2e2c36749298e58b34dca163ea48a7fc925da6 the plugin name was changed
from uefi to uefi_capsule. while the config file name was changed, the
section name should also be changed.

fixes #2748
commit 7c91b8a6 (2021-01-12): trivial: debian: migrate uefi->uefi_capsule in uefi.conf

Fixes: #2748
commit 70f6595b (2021-01-12): trivial: debian: migrate uefi->uefi_capsule in uefi.conf

Fixes: #2748