fwupd/fwupd #2755: Can't create deb files

Sean-StarLabs, 2021-01-12:
Far from major but can't create deb packages:

```
[ ! -d debian/tmp/usr/lib/modules-load.d ] || dh_install -pfwupd usr/lib/modules-load.d
dh_missing -a --fail-missing
dh_missing: warning: lib/modules-load.d/fwupd-msr.conf exists in debian/tmp but is not installed to anywhere 
```

superm1, 2021-01-12:
Can you please provide full build output?  Need to see what you're building from.  I'm not sure what for you they are getting put in `/lb/modules-load.d` instead of `usr/lib/modules-load.d`

Sean-StarLabs, 2021-01-13:
Build log:
https://launchpadlibrarian.net/516966047/buildlog_ubuntu-groovy-amd64.fwupd_1.5.5+r17+g6c48e67d_BUILDING.txt.gz

I get the same locally, just don't know how to send that (comments have a character limit, who knew...)

closed 2021-01-13 by superm1

Sean-StarLabs, 2021-01-13:
That did the trick, thank you :+1: 

Can't get the service to start with the built version:
```
● fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: exit-code) since Wed 2021-01-13 19:32:36 GMT; 1s ago
       Docs: https://fwupd.org/
    Process: 2245 ExecStart=/usr/libexec/fwupd/fwupd (code=exited, status=1/FAILURE)
   Main PID: 2245 (code=exited, status=1/FAILURE)

Jan 13 19:32:36 sean-LabTop systemd[1]: Starting Firmware update daemon...
Jan 13 19:32:36 sean-LabTop fwupd[2245]: /usr/libexec/fwupd/fwupd: /usr/local/lib/x86_64-linux-gnu/libjcat.so.1: version `LIBJCAT_0.1.3' not found (required by /usr/libexec/fwupd/fwupd)
Jan 13 19:32:36 sean-LabTop systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
Jan 13 19:32:36 sean-LabTop systemd[1]: fwupd.service: Failed with result 'exit-code'.
Jan 13 19:32:36 sean-LabTop systemd[1]: Failed to start Firmware update daemon.

```
It exists...
```
sean@sean-LabTop:/usr/local/lib/x86_64-linux-gnu$ ls -l
total 296
drwxr-xr-x 2 root root   4096 Jan 13 19:23 fwupd-plugins-3
drwxr-xr-x 2 root root   4096 Jan 13 19:23 girepository-1.0
lrwxrwxrwx 1 root root     12 Dec 30 14:36 libjcat.so -> libjcat.so.1
lrwxrwxrwx 1 root root     16 Jan 13 19:36 libjcat.so.1 -> libjcat.so.1.0.0
-rwxr-xr-x 1 root root 287976 Dec 30 14:36 libjcat.so.1.0.0
drwxr-xr-x 2 root root   4096 Jan 13 19:23 pkgconfig

```

Might be related as installing from source is fine, it's only the .debs that can't start the service.

superm1, 2021-01-13:
You shouldn't have that in `/usr/local`.  Use the packaged version http://launchpad.net/ubuntu/+source/libjcat

Sean-StarLabs, 2021-01-13:
It's straight from the groovy repo

```
$ apt-cache showpkg libjcat1
Package: libjcat1
Versions: 
0.1.3-2 (/var/lib/apt/lists/gb.archive.ubuntu.com_ubuntu_dists_groovy_main_binary-amd64_Packages) (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/apt/lists/gb.archive.ubuntu.com_ubuntu_dists_groovy_main_binary-amd64_Packages
                  MD5: f288bc99c94ed9e6085612ae807f4bff
 Description Language: en
                 File: /var/lib/apt/lists/gb.archive.ubuntu.com_ubuntu_dists_groovy_main_i18n_Translation-en
                  MD5: f288bc99c94ed9e6085612ae807f4bff

Reverse Depends: 
  libfwupd2,libjcat1 0.1.0
  libfwupdplugin1,libjcat1 0.1.0
  libfwupd2,libjcat1 0.1.0
  fwupd,libjcat1 0.1.3
  libjcat-tests,libjcat1 0.1.3-2
  jcat,libjcat1 0.1.1
  libjcat-dev,libjcat1 0.1.3-2
  libfwupdplugin1,libjcat1 0.1.0
  fwupd,libjcat1 0.1.3
  gir1.2-jcat-1.0,libjcat1 0.1.3
Dependencies: 
0.1.3-2 - libc6 (2 2.4) libglib2.0-0 (2 2.61.2) libgnutls30 (2 3.6.12) libgpg-error0 (2 1.14) libgpgme11 (2 1.2.0) libjson-glib-1.0-0 (2 1.2.0) 
Provides: 
0.1.3-2 - 
Reverse Provides: 

```

superm1, 2021-01-13:
1) Remove /usr/local/*
2) Install `libjcat-dev`.
3) Rebuild your package
4) It should work

Sean-StarLabs, 2021-01-13:
Just did!

Works, but see a strange unknown bar for any `fwupd` related command:
```
$ fwupdmgr --version 
client version:	1.5.6
compile-time dependency versions
	gusb:	0.3.5

Unknown                  [***************************************]
daemon version:	1.5.6
```

superm1, 2021-01-13:
Can you please open new bug for that behavior?

commit 1a66c1c8 (2021-01-13): trivial: debian: fix modules-load.d directory

Fixes: #2755
PR #2758 (2021-01-13): trivial: debian: fix modules-load.d directory
Fixes: #2755

Need to double check the snap and older debian/ubuntu before merging this.

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit dcd15488 (2021-01-13): trivial: debian: fix modules-load.d directory

Fixes: #2755