fwupd/fwupd #2760: The default archive_size_max (512MB) is small for some Poly device's firmware.

zhj8407, 2021-01-13:
**Describe the bug**
The size of Poly Studio's firmware archive exceeds the default archive_size_max (512MB). 

**Steps to Reproduce**

1. Plug in the Poly Studio.
2. Run "sudo fwupdtool install poly_studio_fw_1_4_0_000770.bin.cab".
3. Error message "archive too large (583.6 MB, limit 536.9 MB)" is reported.

**Expected behavior**
The firmware upgrade can run successfully.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt install fwupd

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
─Poly Studio:
│     Device ID:             ac8542369709970eaeabdca7d4f6e16aec31ad37
│     Current version:    2.7
│     Vendor:                 Poly, Inc. (USB:0x095D)
│     GUIDs:                  ef8d95ed-479a-5a0a-98cb-1799a11b7b25
│                                   d22c7eca-b220-5070-8844-c1f34ef56af0
│     Device Flags:        • Updatable
```

**Additional questions**
- Operating system and version: Ubuntu 20.04
- Have you tried rebooting? Yes
- Is this a regression? No

superm1, 2021-01-13:
Seems it's been set to 512MB for a while: https://github.com/fwupd/fwupd/commit/c7bbbc246b89df3d121e745a052322773a74e3ff

I guess we should bump up to 1GB?  Can you submit PR?

zhj8407, 2021-01-13:
Sure, no problem. I can submit the PR.

hughsie, 2021-01-13:
Could we do something smarter? e.g. set it to 50% of the total RAM or something?

superm1, 2021-01-13:
Lets see, the reason this was originally done was to prevent DoS the daemon with big files.
https://github.com/fwupd/fwupd/pull/353

I don't think a heuristic makes it any less likely to DoS the daemon - you can still pass big files and block it from being use if it's in use.  So perhaps need to think about what damage a DoS attack can do?  I almost wonder if we should have the default set to `-1` and let user decide to change the default if they are worried about a certain attack on their machine.

hughsie, 2021-01-13:
> I almost wonder if we should have the default set to -1

That would be fine for me. There are plenty of other easier ways to DoS a system these days..

closed 2021-01-14 by hughsie