fwupd/fwupd #2776: Firmware notifications contain a URL that points to localhost

Sean-StarLabs, 2021-01-15:
* Star Lite Mk III Device Update version 1.1.0

Please log into the LVFS and view the failures: https://localhost/lvfs/firmware/5960/analytics/reports/3

superm1, 2021-01-21:
Would you mind copying the failure here?  I can't see your failure instance.

Sean-StarLabs, 2021-01-21:
You mean this? https://fwupd.org/lvfs/firmware/5960/analytics/reports/3

Sean-StarLabs, 2021-01-21:
`

BootTime | 1607624168
-- | --
cf4b0cbd825a282787e418b5470697119680c909
12
1.5.3
0.3.5
x86_64
fedora
workstation
33
35.5 kB (35,464 bytes)
/boot/efi
574619947
d9d7b13b-e4db-4f91-8bf6-8952a9caa82a
I3
Lite
I3
Star Labs
5.9.12-200.fc33.x86_64
0x0
0x1000000
False
uefi
2.4
12
0.7.14
1.5.3
0.3.5
5.9.12-200.fc33.x86_64
Disabled
0x00000008 c42fedad268200cb1d15f97841c344e79dae3320 [..kT..U@..N....:] 0x00000001 36d77105edeb10bd732525b769d75747d53d6224 [..........*.....] 0x00000004 9069ca78e7450a285173431b3e52c5c25299e473 PCR0: cf4b0cbd825a282787e418b5470697119680c909 PCR0: 80fe4abbc6c9c5fcd3df6c3b82cbb8b4ebe2723580e9ddce5136d23f8244e514
2.0
Enabled
failed to run update on reboot
1.1.0
1.0.0

`

superm1, 2021-01-21:
So where does it say localhost?

Sean-StarLabs, 2021-01-21:
In the email notification (text and link):

[image]

It has:
https://localhost/lvfs/firmware/5960/analytics/reports/3

Should be:
https://fwupd.org/lvfs/firmware/5960/analytics/reports/3

superm1, 2021-01-21:
Ohhhhh.  I see.  it's not the report; it's an LVFS problem.  Report it here instead: https://gitlab.com/fwupd/lvfs-website

closed 2021-01-21 by superm1

Sean-StarLabs, 2021-01-21:
That's where it got moved too! Google didn't know :)