fwupd/fwupd #2782: WARNING: This package has not been validated, it may not work properly.

mendel5, 2021-01-17:
Today I have updated fwupd from version 1.5.1 to 1.5.5. My install method is snap. When using fwupd I get the error `WARNING: This package has not been validated, it may not work properly.` as an output at the top for most commands. There is also an output that says `Unknown                  [***************************************]` which I have not seen before.

Here is some info:
```
$ sudo fwupdmgr --version
client version:	1.5.5
compile-time dependency versions
	gusb:	0.3.5

Unknown                  [***************************************]
daemon version:	1.5.5
```
I think in previous version of fwupd there was no `Unknown` printed out. See for example the same command in my issue at https://github.com/fwupd/fwupd/issues/2487 (bottom of post 1).

```
$ fwupdmgr get-devices
Unknown                  [***************************************]
WARNING: This package has not been validated, it may not work properly.
20QD00M4GE
│
├─Thunderbolt host controller:
│     Device ID:          470906fa3412b62a24358dda0865908d5969c99b
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    44.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              74cdddba-f7b6-5a31-a3dc-64186b5a5491
│                         18bb00e3-5d7a-55d3-831e-0e000c184b0e
│                         c7920601-0cda-507f-851e-92229eb1d370
│                         96c6512b-0138-51da-b02f-f44f2c4c7744
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          c085a3cbf4dc1d60d110b7616816e2ffd279b64e
│     Current version:    0.1.20
│     Minimum Version:    0.1.20
│     Vendor:             DMI:LENOVO
│     GUIDs:              f72e048b-65bd-4e71-9071-1ac7045223e5
│                         9cd54062-d764-51a7-a780-51a3695beae2
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          0ec6a2d51c2c55a649247d119ffb8e229168d6c9
│     Current version:    192.64.1551
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│                         35cbf8ae-bee1-52fa-bf0d-11f1fc4c12f5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f
│ │   Current version:    0.1.40
│ │   Minimum Version:    0.0.1
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   GUIDs:              4bbc40fa-f81e-4206-bc70-a1f7b744d964
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       171d8529-50f6-550b-9db6-7f21af7414ef
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  238
│       Minimum Version:  238
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341
│                         5971a208-da00-5fce-b5f5-1234342f9cf7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
├─Touchpad:
│     Device ID:          91961e6ddc950eb21e71db6f33ec09a133933a95
│     Current version:    1.3.3013412
│     Bootloader Version: 8.0.0
│     Vendor:             Synaptics (HIDRAW:0x06CB)
│     GUIDs:              10010313-27c0-5c7f-bd99-955028fd3a01
│                         71463ccd-7553-5d7d-9fac-7ed8073a7853
│                         981824ee-4283-595b-a2e4-46eae92dba40
│                         0957b08b-b580-5c3a-8073-98b9bb15bc7f
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Supported on remote server
│   
├─UEFI Device Firmware:
│     Device ID:          b596f8dec94a959248534b35641fe2341e27ed06
│     Current version:    16785559
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     GUIDs:              9045ae09-34ab-46d6-826d-b02f859dd20b
│                         56aa226c-4831-532f-a1c4-896f57b8e8d8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          f61a138e8240c473a88d703babf7be5bc0e098e6
│     Current version:    4784136
│     Minimum Version:    4784136
│     Vendor:             DMI:LENOVO
│     GUIDs:              db246b7d-a882-4e0e-984f-23ac1b0270db
│                         0253277e-3c60-502c-8423-74b9e15950b5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          97d441c3228fe8615f883b166a8809389803f9e2
│     Current version:    26403
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     GUIDs:              68d3b08f-6ed2-43ca-86f0-f1ad6e41b390
│                         64c2832d-137c-5b2f-af14-028dc25b659a
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─WDC PC SN730 SDBQNTY-512G-1001:
      Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Summary:            NVM Express Solid State Drive
      Current version:    11170101
      Vendor:             Sandisk Corp (NVME:0x15B7)
      GUIDs:              7b974856-317d-5538-a915-de0a6353750f
                          12c86995-0b90-5ec5-98f3-7a6ed4ca50e0
                          a39943dd-3afb-54f8-b110-c5a21f071200
                          fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2
                          06b4e2aa-91af-508b-b06e-65e3b3189e97
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```
Here you can see `Unknown` and `WARNING` at the top.

```
$ sudo fwupdmgr refresh --force
Unknown                  [***************************************]
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 6 local devices supported
```
Here as well.

```
$ sudo fwupdmgr report-history
Unknown                  [***************************************]
WARNING: This package has not been validated, it may not work properly.
No reports require uploading
```
And here too.

Do you why it says `Unknown` and `WARNING`?

hughsie, 2021-01-17:
It's intended behaviour; we do almost no QA on the nightly auto-generated snap. You need to use the distro-supported package to get rid of the warning. The 'Unknown' status message was also fixed in master recently.

mendel5, 2021-01-18:
> The 'Unknown' status message was also fixed in master recently.

Alright!

> It's intended behaviour; we do almost no QA on the nightly auto-generated snap. You need to use the distro-supported package to get rid of the warning.

Ok, I will try installing fwupd using `apt`.

closed 2021-01-18 by hughsie

ferki, 2021-02-09:
I'm also having the same with 1.5.5 (but not with 1.5.4) when installing from the default package sources on Gentoo. Should I open a new issue about it?

hughsie, 2021-02-09:
> when installing from the default package sources on Gentoo

If you're compiling from source then the message is correct, and not a bug...

LinuxOnTheDesktop, 2021-08-04:
> It's intended behaviour; we do almost no QA on the nightly auto-generated snap. You need to use the distro-supported package to get rid of the warning.

It seems to me that the message itself should say something to that effect. Otherwise affected users have to search the web to discover the meaning of the message. For, the message itself is cryptic: what 'package'? what sort of 'validation'? (Nor even is the message grammatical: commas are not conjunctions; treating them as if they are can and here does leave the relation between clauses unclear.) I commend rewriting the message along the following lines.

> WARNING: this version of the fwupd tool is an interim release. Use 'apt' or similar to obtain a better-tested release.

LinuxOnTheDesktop, 2023-12-05:
&nbsp;&nbsp;&nbsp;&nbsp;The following confusing and indeed ungrammatical message remains present within recent versions of 'fwupd'.

> This package has not been validated, it may not work properly.

Here is a relevant way of conceiving  grammatical problems: an ungrammatical sentence resembles code that will not compile. Indeed: as with code that will not compile, normally there is good reason to refuse to allow ungrammatical text (though one would have thought that this argument had been won long ago).

clipcarl, 2025-07-04:
I'm getting the message with the package installed by my distribution (Alpine Linux).

The grammar of the message can be fixed by changing the comma (,) to a semicolon (;) or to a period (.).

However, the message is still confusing and should be changed to actually tell the user something useful and understandable. With no context of what "validated" or "properly" mean this message is cryptic and meaningless.

superm1, 2025-07-06:
> I'm getting the message with the package installed by my distribution (Alpine Linux).
> 

You should report this to your distribution provider.

> The grammar of the message can be fixed by changing the comma (,) to a semicolon (;) or to a period (.).
>
> However, the message is still confusing and should be changed to actually tell the user [...]

Please file a pull request with a proposal for clearer text and we can discuss it there rather than on a previously closed bug.

superm1, 2025-07-06:
> > Please file a pull request with a proposal for clearer text and we can discuss it there rather than on a previously closed bug.
> 
> I have no idea what this error message is trying to convey so I can't propose new wording. Presumably the developers know what the point of this error message is so they would be the best people to come up with new wording that actually communicates its purpose to the users.

The basic message it conveys is that this build has had no validation. Packagers are supposed to set it when they have done some level of testing and know it works.