fwupd/fwupd #2792: Best practices for designing a firmware update interface.

bwidawsk, 2021-01-20:
My team (@djbw @stellarhopper) is working on [enabling](https://gitlab.com/bwidawsk/linux/-/tree/cxl-2.0v4) CXL 2.0 for Linux kernel. Part of the [CXL specification](https://www.computeexpresslink.org/download-the-specification) outlines updating device firmware. Vendors are free to implement whatever scheme they want for firmware update if they eschew the spec'd way, but we will be providing a well defined interface to do it (via IOCTL) the spec'd way. We will also ship a library, [libcxl](https://github.com/pmem/ndctl/tree/cxl-2.0v1), that is able to perform the firmware updates for the root user (we will also have an application that can do it).

What I'd like to ask this project is, are there preferred ways to design the interface to be easy to use with fwupd? Are there any gotchas we should be aware of? What else should we know in order to be able to easily integrate our work into fwupd?

Thanks.
Dan & Ben & Vishal

hughsie, 2021-01-20:
Hi all! First, thanks for opening this bug; I've been asked the same thing before and I've polished up some of that discussion into some a draft MR for lvfs -- this might help you a bit: https://gitlab.com/fwupd/lvfs-website/-/merge_requests/889/diffs -- especially the part about depending on a new library :/

I'd love to see how little code you could transplant from libcxl to a fwupd plugin; typically the firmware update parts of a library like this are pretty self contained and simple. e.g. just copy and pasting https://github.com/pmem/ndctl/blob/cxl-2.0v1/ndctl/firmware-update.h and the other core bits needed for that.

bwidawsk, 2021-01-20:
@hughsie libcxl will be part of ndctl, which should exist in every major distro. As for the reset of the advice, it looks straightforward.

Thanks for the quick response. For now, @stellarhopper is doing most of the work on the userspace side, so I'll leave this open until he feels like he knows what he needs to do.

hughsie, 2021-01-20:
@bwidawsk it exists in Ubuntu, but it's not in main: https://packages.ubuntu.com/xenial/libdevel/libcxl-dev -- it's also only built for ppc64le everywhere I look -- is that expected? I really think not depending on additional libraries and just copying the small amount of code you need into fwupd would be the path of least resistance if you want it enabled by default on all distros.

superm1, 2021-01-20:
It seems like it was torn out of Debian too, I can't find it there.  Probably because the package hasn't been touched since 2018 even in Ubuntu: https://launchpad.net/ubuntu/+source/libcxl

I'm in alignment, lets try to carry a little bit of code in fwupd from libcxl.  Without yet digging into the code or any specs at a high level you expect that there are a few basic steps to updating firmware:
1. Put the device in  a state to accept firmware
2. Put the firmware somewhere to stage it
3. Write to a SPI part
4. Activate the new firmware

If you want a very simple approach is low touch in userspace you can model off thunderbolt/usb4.  All of those steps are in two simple sysfs actions.  Everything else is abstracted to the kernel.

If you want a more in depth approach you can have a few ioctls to the kernel for those steps, you can look at nvme for an idea how this works.

stellarhopper, 2021-01-20:
@hughsie @superm1 Ah that Ubuntu library is an unfortunate name collision with Open CAPI :\
The ndctl packages for Ubuntu provide libndctl and libdaxctl (e.g. https://packages.ubuntu.com/hirsute/libndctl6).

The new 'libcxl' is not yet in any release, so won't be in any distro today, but once available, there is should already a packaging 'pathway' for it since it will be part of the 'ndctl' project.

I think the reasons for not depending on a new library for fwupd make sense. The fw update process should only need to issue some ioctls, or read/write some sysfs files. It sounds reasonable to replicate that code in a fwupd plugin.

The actual firmware update has to be coordinated by the kernel - the userspace part will likely be limited to setting up the binary, and saying 'go', and the kernel does the rest. We just starting to flesh out those interfaces, but they should remain stable.

I propose lets figure out the kernel interfaces, and implement the libcxl side first. And then see actually how much (or how little) code we're talking about replicating - I expect it to be very manageable.

bwidawsk, 2021-01-20:
> The fw update process should only need to issue some ioctls, or read/write some sysfs files. It sounds reasonable to replicate that code in a fwupd plugin.

The only argument I have is if/when vendors decide not to use the spec'd way to update the firmware. I think we said we wouldn't support this in libcxl fwiw.

hughsie, 2021-01-20:
I think "you don't need to write any code to get firmware on the LVFS" would be a pretty convincing reason! From the OEMs I've worked with it seems the less development they have to do the better.

djbw, 2021-01-20:
One blind-spot I have is whether fwupd has the concept of activation methods and batched updates. For example, the firmware update ABI for NVDIMMs recently grew the capability for "online activate" rather than a reboot. However, since that activation makes memory unavailable for milliseconds (quiesce period) the driver arranges to activate multiple devices at once under one quiesce period with the OS suspended. I expect CXL memory devices will need to contend with similar details.

superm1, 2021-01-20:
> One blind-spot I have is whether fwupd has the concept of activation methods and batched updates. For example, the firmware update ABI for NVDIMMs recently grew the capability for "online activate" [...]

It sure does support this.  There is an activate sub command specifically for that purpose:
https://github.com/fwupd/fwupd/blob/master/src/fu-util.c#L2945

By default if no one calls activate but something needs activation it's done at shutdown.  Some OSes (such as ChromeOS) will call activate at logout.

djbw, 2021-01-20:
Cool thanks, will take a look at how to publish an activation method for a class of devices.

bwidawsk, 2021-01-20:
> I think "you don't need to write any code to get firmware on the LVFS" would be a pretty convincing reason! From the OEMs I've worked with it seems the less development they have to do the better.

I don't know why they wouldn't use the spec'd mechanism, and so far nobody necessarily has, but it's technically possible.

closed 2021-01-26 by superm1