fwupd/fwupd #2801: Failed to run update on reboot

physkets, 2021-01-23:
**Describe the bug**
The system restarts after performing a `$ fwupdmgr update`, but does not perform the update. Upon checking the update history, this is what it says in the relevant section of the relevant firmware, the update of which was attempted:
```shell
$ fwupdmgr get-history
  │   Update State:       failed
  │   Update Error:       failed to run update on reboot
```

**Steps to Reproduce**
I am using the system with SecureBoot enabled, and my own keys. I have set `DisableShimForSecureBoot=true` in `/etc/fwupd/uefi_capsule.conf`.

**Expected behavior**
Upon restart, fwupd should initiate a UEFI flash.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version: 1.5.5
compile-time dependency versions
        gusb:   0.3.5

Unknown                  [***************************************]
daemon version: 1.5.5
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):  
`pacman`

**fwupd device information**

```shell
$ fwupdmgr get-devices --show-all-devices
20UECTO1WW
│
├─AMD Ryzen 7 PRO 4750U with Radeon Graphics:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         997104da-ec11-5e04-9d94-ac5501fbf609
│                         1f2e0c2f-e078-5fcb-a8de-f78eff9a470a
│                         5fd5eb41-dd4a-5742-8448-861eaaa61cf8
│     Device Flags:       • Internal device
│   
├─SAMSUNG MZVLB512HBJQ-000L7:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express Solid State Drive
│     Current version:    4M2QEXF7
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:              0b4d773a-7ac3-58c1-a541-e22ef1cdfe02
│                         c9d531ea-ee7d-5562-8def-c64d0d144813
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d
│                         47335265-a509-51f7-841e-1c94911af66b
│                         f4ceb2a8-09c4-55c0-af2a-13ccd63925b1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:          3f10cbb75c39627fc0d14eddd274374207a5c753
│     Current version:    0.1.25
│     Vendor:             LENOVO (DMI:LENOVO)
│     Update State:       success
│     GUIDs:              01453b71-da0c-4832-9f4f-e378245339c7
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a
│                         38ad8832-d322-5d73-aff7-77b8ae0f1a6f
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
20UECTO1WW
│
├─AMD Ryzen 7 PRO 4750U with Radeon Graphics:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         997104da-ec11-5e04-9d94-ac5501fbf609
│                         1f2e0c2f-e078-5fcb-a8de-f78eff9a470a
│                         5fd5eb41-dd4a-5742-8448-861eaaa61cf8
│     Device Flags:       • Internal device
│   
├─SAMSUNG MZVLB512HBJQ-000L7:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express Solid State Drive
│     Current version:    4M2QEXF7
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:              0b4d773a-7ac3-58c1-a541-e22ef1cdfe02
│                         c9d531ea-ee7d-5562-8def-c64d0d144813
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d
│                         47335265-a509-51f7-841e-1c94911af66b
│                         f4ceb2a8-09c4-55c0-af2a-13ccd63925b1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:          3f10cbb75c39627fc0d14eddd274374207a5c753
│     Current version:    0.1.25
│     Vendor:             LENOVO (DMI:LENOVO)
│     Update State:       success
│     GUIDs:              01453b71-da0c-4832-9f4f-e378245339c7
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a
│                         38ad8832-d322-5d73-aff7-77b8ae0f1a6f
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
```

**System UEFI configuration**

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0002,0000,0017,0018,0019,001A,001B,001C
Boot0000* Linux Boot Manager    HD(1,GPT,e29be295-b656-3045-98f8-8798bd26ec1c,0x800,0x200000)/File(\EFI\systemd\systemd-bootx64.efi)
Boot0001* Archlinux     HD(1,GPT,e29be295-b656-3045-98f8-8798bd26ec1c,0x800,0x200000)/File(\EFI\Linux\linux-vnl.efi)
Boot0002* Archlinux LTS HD(1,GPT,e29be295-b656-3045-98f8-8798bd26ec1c,0x800,0x200000)/File(\EFI\Linux\linux-lts.efi)
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  ThinkShield secure wipe       FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
Boot0015  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PXE BOOT      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Regulatory Information        FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot001E* Boot Next Boot Option VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,a7ca6d35b2c2684783721826a7404894)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-01453b71-da0c-4832-9f4f-e378245339c7-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
```

```shell
$ tree /boot
/boot
├── amd-ucode.img
├── EFI
│   ├── arch
│   │   ├── fw
│   │   │   ├── fwupd-01453b71-da0c-4832-9f4f-e378245339c7.cap
│   │   │   └── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │   └── fwupdx64.efi
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   ├── Linux
│   │   ├── linux-lts.efi
│   │   └── linux-vnl.efi
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   └── arch-lts.conf
│   ├── loader.conf
│   └── random-seed
├── vmlinuz-linux
└── vmlinuz-linux-lts
```

**Additional questions**
- Operating system and version: *Archlinux*
- Have you tried rebooting? *Yes*
- Is this a regression? *Don't know. Updating system-firmware for the first time*
- Are you using an NVMe disk? *Yes*
- Is secure boot enabled? *Yes; with custom keys.*

hughsie, 2021-01-23:
If you're using `DisableShimForSecureBoot=true` and you're using a custom key don't you also need to resign `fwupdx64.efi` too?

physkets, 2021-01-23:
Yes, and I use [sbctl](https://github.com/Foxboron/sbctl/) to manage signing. According to it, it is signed, and in the EFI partition:

```
$ sudo sbctl verify 
==> Verifying file database and EFI images in /boot...
  -> /boot/EFI/BOOT/BOOTX64.EFI is signed
  -> /boot/EFI/systemd/systemd-bootx64.efi is signed
  -> /boot/vmlinuz-linux is signed
  -> /boot/vmlinuz-linux-lts is signed
  -> /usr/lib/fwupd/efi/fwupdx64.efi.signed is signed
  -> /boot/EFI/Linux/linux-lts.efi is signed
  -> /boot/EFI/Linux/linux-vnl.efi is signed
  -> /boot/EFI/arch/fwupdx64.efi is signed
```

hughsie, 2021-01-23:
@superm1 any ideas?

superm1, 2021-01-23:
Try an update one more time and right before the reboot provide the efibootmgr output again. It seems like an entry didn't create but we should confirm that is the actual problem.

Other idea is Lenovo boot lock thing.

physkets, 2021-01-23:
This is the output after the update, but before the restart:
```
$ efibootmgr -v
BootNext: 0003
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0002,0000,0017,0018,0019,001A,001B,001C,0003
Boot0000* Linux Boot Manager    HD(1,GPT,e29be295-b656-3045-98f8-8798bd26ec1c,0x800,0x200000)/File(\EFI\systemd\systemd-bootx64.efi)
Boot0001* Archlinux     HD(1,GPT,e29be295-b656-3045-98f8-8798bd26ec1c,0x800,0x200000)/File(\EFI\Linux\linux-vnl.efi)
Boot0002* Archlinux LTS HD(1,GPT,e29be295-b656-3045-98f8-8798bd26ec1c,0x800,0x200000)/File(\EFI\Linux\linux-lts.efi)
Boot0003* Linux-Firmware-Updater        HD(1,GPT,e29be295-b656-3045-98f8-8798bd26ec1c,0x800,0x200000)/File(\EFI\arch\fwupdx64.efi)
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  ThinkShield secure wipe       FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
Boot0015  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PXE BOOT      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Regulatory Information        FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot001E* Boot Next Boot Option VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,a7ca6d35b2c2684783721826a7404894)
```
There is a new entry for the firmware updater that has been designated `BootNext`. So that seems correct, no?

> Other idea is Lenovo boot lock thing.

What is that?

physkets, 2021-01-23:
Okay, so that was it. I looked through the UEFI settings, and found one called `Boot Order Lock`, and disabled it. Upon restart it did then go on to update the firmware.

Although it did do an odd thing, where towards the end of the update, the fan rose to maximum RPM, and the machine rebooted to show a screen that said something like "Self-Healing backup", so I thought that it did not succeed, but when I check the version of the firmware now, it seems to have been upgraded from 0.1.25 to 0.1.29 . So it seems to have worked.

Thanks a lot!

closed 2021-01-23 by physkets

superm1, 2021-01-23:
@hughsie I think you should modify the template for uefi bugs, this firmware behavior comes up too much

hughsie, 2021-01-23:
Maybe on reboot we could also test if the boot entry still exists? We don't delete it on efi mode now, and if boot order lock is on our changes will have been "reverted". Hopefully when we get the bios interface stuff in we can check in prepare, not after reboot.

superm1, 2021-01-23:
Sounds good to me