fwupd/fwupd #2803: EFI partition not found

computerix, 2021-01-24:
fwupd 1.5.5 on a system without udisks
EFI partition mounted as /boot before starting fwupd the first time
OverrideESPMountPoint=/boot in /etc/fwupd/uefi.conf

Each time I execute fwupdmgr, I get an error message
UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.

Why? If /boot is already mounted before the daemon is started, 
it should just work and not depend on udisks in any way?

hughsie, 2021-01-24:
Seems sane, patches most certainly welcome if this doesn't happen already. All my systems (even the embedded ones!) have udisks installed.

donnex, 2021-02-07:
I've got the same issue as well.

superm1, 2021-02-10:
I think what's going on here is `fu_common_get_esp_for_path` fails because it can't get volumes without udisks (IE here: https://github.com/fwupd/fwupd/blob/67dda6b2e93245dea296c44a4db9086bc3d9b9da/libfwupdplugin/fu-common.c#L2779)

Someone affected, I would suggest to share verbose daemon output to confirm (`/usr/libexec/fwupd/fwupd --verbose`)

computerix, 2021-02-10:
I can't test, I've installed udisks in the meantime.

closed 2021-02-10 by superm1

adrian5, 2021-02-19:
Commit be220a4 made it into 1.5.6, right?

I'm using the current Arch [fwupd 1.5.6-1](https://archlinux.org/packages/community/x86_64/fwupd/) package without udisks, and the issue persists for me.

**/etc/fwupd/uefi_capsule.conf**

```ini
[uefi]
OverrideESPMountPoint=/boot/efi
```

superm1, 2021-02-19:
>EDIT: Renaming the heading in the config file to [uefi_capsule] seems to work! I guess this is an oversight made downstream.

Yup, something like [this](https://github.com/fwupd/fwupd/blob/master/contrib/debian/fwupd.postinst#L18) needs to be put into the `postinst`.  @FFY00 feel free to add something like that for a packaging change.

adrian5, 2021-02-19:
Ah great! I was about to file a request on the Arch bug tracker, but pinging him here is more straight forward.

superm1, 2021-02-19:
(Well email notification isn't the best tracker, you might still file something in case he misses it or is busy and forgets)

adrian5, 2021-02-19:
Blimey, it looks like a mistake on my part actually. I must have touched (though not modified) that file in the past, and the package manager created a sibling (`uefi_capsule.conf.pacnew`) that contains the correct heading. Didn't think to check for that possibility. All resolved now, **false alarm**.

FFY00, 2021-02-19:
Thank you, glad it is resolved. I'll keep this in mind in the future.

commit 0507f3cf (2021-02-10): trivial: fu-common: fix fallback for esp-path if udisks missing

Fixes: #2803
commit be220a4f (2021-02-10): trivial: fu-common: fix fallback for esp-path if udisks missing

Fixes: #2803