fwupd/fwupd #2806: vli: Hyper discrepancy

memily, 2021-01-26:
**Describe the bug**
Previously, we were able to upgrade/downgrade the entire Hyper .cab firmware package with fwupdmgr and now we're failing on PD firmware (Hub firmware is pass). We’re wondering if there is a *bug*? 

**Steps to Reproduce**
Testing Process: on a VL817+VL103 Share SPI configured board we were able to use 1.5.5 fwupdmgr to update/downgrade Travel Hub Gen2’s firmware and Mini Dock. With that same configuration board we were not able to upgrade/downgrade LVFS uploaded Hyper .cab firmware [VL817+VL103 Share SPI]. 

Results:
Hyper Hub firmware updates/downgrades pass and PD firmware update/downgrade fails. 

Additional note: When we run `python3 fwupd/data/device-tests/hardware.py` it returns success for both tests even if PD did not upgrade/downgrade on the device after calling fwupdmgr get-devices. 

**Expected behavior**
We expect the correct Hub and PD versions from the update/downgrade Hyper .cab file called in `./builddir/src/fwupdmgr install HyperXXX.cab`

**fwupd version information**
client+daemon version: 1.5.5

**Closing**
We're hoping its a simple problem we overlooked somehow! We also checked the tier1 and tier3 meta files and they match the bin files. Thanks.

hughsie, 2021-01-26:
Have you tested this on actual Hyper hardware or just the devboard? IIRC, the reset mechanism was different.

memily, 2021-01-27:
>Have you tested this on actual Hyper hardware...

We are testing fwupd on boards sent to us by Hyper with firmware uploaded to LVFS for Hyper. The reset mechanism is manual replug, which we also tried but PD still fails to come back updated/downgraded afterwards.

hughsie, 2021-01-27:
> but PD still fails to come back updated/downgraded afterwards

So hyper-no-sku-vli-tier3.json succeeds but the version number is invalid? Any chance you could send me a PCB to try here? It would be useful to have in the pre-release device tests if nothing else.

hughsie, 2021-02-04:
Hi Emily. I got the package today, many thanks. A quick check: the firmware is here https://fwupd.org/lvfs/components/6942 which has a tier3 (PD) GUID of `0a120f99-b0f4-52af-9af9-e13155634370`. Looking at the device itself I see:

    │ └─VL103:
    │       Device ID:        fe008de085345975906d64be2af7cc99f36724ca
    │       Summary:          USB-C Power Delivery Device
    │       Current version:  138.36.7.55
    │       Vendor:           Gopod (USB:0x2D01)
    │       Install Duration: 15 seconds
    │       GUIDs:            dbbf1af1-7df4-5715-b8c0-29045c73160e ← USB\VID_2109&PID_0103&APP_37
    │                         109f8fec-bb56-562d-9fae-15e5e8281b89 ← USB\VID_2109&PID_0103&DEV_VL103
    │                         e35736c4-d683-5bfc-bf0b-004d6e2f78f5 ← USB\VID_2109&PID_0103
    │       Device Flags:     • Updatable
    │                         • Cryptographic hash verification is available
    │                         • Device can recover flash failures

I did expect to see `dbbf1af1-7df4-5715-b8c0-29045c73160e` on the LVFS...

memily, 2021-02-08:
Thank the royal mail it arrived! A calendar aside: the Chinese continent is going on lunar new years break midweek so we apologize for possible delays going into next week.

> I did expect to see dbbf1af1-7df4-5715-b8c0-29045c73160e on the LVFS...

What does this mean? Is the PD suppose to be hidden?

hughsie, 2021-02-08:
> Is the PD suppose to be hidden

No. The GUID in the metainfo.xml needs to match one of the GUIDs in the `fwupdmgr get-devices` output.

memily, 2021-02-09:
>No. The GUID in the metainfo.xml needs to match one of the GUIDs in the fwupdmgr get-devices output.

The GUID in the metainfo.xml is correct and matches VID/PID “USB\VID_2D01&PID_0380”. Your `fwupdmgr get-device` result is surprising to us as well, the FW team says they did put PD VID//PID 2D01/0380 FW onto the board before shipping it to you... I'll have to check with the team to see what is the problem with the VID/PID in the FW. Are you able to see the other billboard on the device with `fwupdmgr get-device` [aside from the Hub and PD]?

hughsie, 2021-02-09:
I've attached my fwupdtool output: [fwupdtool.txt](https://github.com/fwupd/fwupd/files/5950283/fwupdtool.txt)

memily, 2021-02-19:
Why howdy, we're back from break!

Does fwupd store the get-devices in a cache perhaps? Is get-devices reading the usb descriptor for VL103 PD from Share SPI? Running `lsusb –v` and win10's USB View we get the correct VID/PID for VL103 PD (attached at the end for reference).

>my fwupdtool output

We see `VendorId:             USB:0x2D01` at the end of the txt output for VL103, which is correct; even if the VID in the GUID is incorrect. We're wondering if its a read issue or a cache issue...

[image]

[image]

hughsie, 2021-02-19:
> Why howdy, we're back from break!

Great, happy new year! I hope you had a great break.

> Is get-devices reading the usb descriptor for VL103 PD from Share SPI

Exactly that; https://github.com/fwupd/fwupd/blob/master/plugins/vli/fu-vli-usbhub-pd-device.c#L39 -- is the PD device not share SPI? i.e. does it have it's own SPI chip?

My suspicion is the firmware needs to have the correct VID/PID written into the shared SPI firmware.

memily, 2021-02-19:
Thanks! Everyone got much needed rest.

>My suspicion is the firmware needs to have the correct VID/PID written into the shared SPI firmware. 

Yes, PD is share SPI; but the way the usb descriptor should be fetched for PD is not from Hub shared SPI. We've confirmed from the firmware department that the VID/PID on the firmware is correct for PD and using `lsusb` validates VID/PID for Hub and PD --on our test device that is identical to the device we sent you.

hughsie, 2021-02-20:
> but the way the usb descriptor should be fetched for PD is not from Hub shared SPI

We currently have two different PD devices we can represent using the quirk files:

 * `FuVliPdDevice`: This is a "standalone" PD with it's own SPI chip, using GPIO_3 to disable the WP and then either erasing the entire SPI and writing the whole image, or for dual image devices writing fw1/fw2 as appropriate. This has to be exposed as an enumerated USB device.

* `FuVliUsbhubPdDevice`: The "shared" PD that shares the SPI with the USB HUB. The SPI is erased at the offset specified for the device kind and then the entire blob is written at that offset. The USB descriptor is not available as it's not always enumerated and so we read the descriptor from the SPI at address `FLASHMAP_ADDR_PD_LEGACY`, falling back to `FLASHMAP_ADDR_PD` if invalid. I think that not enumerating the PD device was a deliberate decision on some hardware.

If I understand correctly Hyper is a new type of PD device that is enumerated as a USB device but actually shares the SPI with the USB HUB? In which case we don't support that with the existing code, and I really think we should just use `FuVliUsbhubPdDevice` for all PD devices that share SPI with the USB hub, regardless of whether they are enumerated or not. That does mean the USB descriptor stored at offset `FLASHMAP_ADDR_PD` needs to match the descriptor that is being exported.

I really don't think it makes sense to define a 3rd way to update the PD device, but if you had to have this it would be something like `FuVliPdUsbHubDevice` -- but I really think just fixing the firmware to have matching VID/PIDs compared to the descriptor is the simplest thing to do.

memily, 2021-02-22:
>If I understand correctly Hyper is a new type of PD device that is enumerated as a USB device but actually shares the SPI with the USB HUB? 

Working? On a Saturday? A crime! Hyper is not a new type of PD device, the hardware and FW update method is the exact same as Lenovo TR Gen2 [Share SPI, VL817 can self-reset, VL103 requires replug]. We tested TR Gen 2 device and it was able to upgrade/downgrade PD. And yes, not enumerating the PD device was a deliberate decision on some hardware design. 

>That does mean the USB descriptor stored at offset `FLASHMAP_ADDR_PD` needs to match the descriptor that is being exported.

We will follow up to check where the descriptor is being exported. The firmware department assures us the firmware inside your device does have matching VID/PIDs with the cab on LVFS.

hughsie, 2021-02-22:
> The firmware department assures us the firmware inside your device does have matching VID/PIDs with the cab on LVFS

Dumping 8 bytes of firmware at 0x21003 I get: `8a24 0737 0921 0301` which corresponds to a version of `138.36.7.55` and a VID of `2109` and PID of `0103`

memily, 2021-02-25:
>Dumping 8 bytes of firmware at 0x21003

We got the updated, correctly written VID/PID to 0x21003 from the fw department and have made the update to LVFS and json files. Sorry for the trouble...

hughsie, 2021-02-25:
old firmware:

	VL103
	  DeviceId:             fe008de085345975906d64be2af7cc99f36724ca
	  ParentDeviceId:       da77984c82b59c6fc69516431f467fd9a8d39a7f
	  Status:               device-write
	  Guid:                 dbbf1af1-7df4-5715-b8c0-29045c73160e <- USB\VID_2109&PID_0103&APP_37
	  Guid:                 109f8fec-bb56-562d-9fae-15e5e8281b89 <- USB\VID_2109&PID_0103&DEV_VL103
	  Guid:                 e35736c4-d683-5bfc-bf0b-004d6e2f78f5 <- USB\VID_2109&PID_0103
	  Summary:              USB-C Power Delivery Device
	  Plugin:               vli
	  Protocol:             com.vli.usbhub
	  Flags:                updatable|registered|can-verify|can-verify-image|self-recovery|no-guid-matching
	  Vendor:               Gopod
	  VendorId:             USB:0x2D01
	  Version:              138.36.7.55
	  VersionFormat:        quad
	  VersionRaw:           0x8a240737
	  Icon:                 audio-card
	  InstallDuration:      15
	  Created:              2021-02-25
	  Modified:             2021-02-25
	  PhysicalId:           usb:04:00:01
	  LogicalId:            PD
	  Order:                18446744073709551615
	  DeviceKind:           VL103
	  FwOffset:             0x20000
	  FwSize:               0x8000

new firmare:

	VL103
	  DeviceId:             fe008de085345975906d64be2af7cc99f36724ca
	  ParentDeviceId:       da77984c82b59c6fc69516431f467fd9a8d39a7f
	  Guid:                 d5a80de1-4491-5cb6-8e47-f5739b13950b <- USB\VID_2D01&PID_0380&APP_37
	  Guid:                 1cb900a9-178b-5d34-a2ed-dde670049dcd <- USB\VID_2D01&PID_0380&DEV_VL103
	  Guid:                 0a120f99-b0f4-52af-9af9-e13155634370 <- USB\VID_2D01&PID_0380
	  Guid:                 887fe9cc-6a40-532c-958d-548b73a96a7b
	  Summary:              USB-C Power Delivery Device
	  Plugin:               vli
	  Protocol:             com.vli.usbhub
	  Flags:                updatable|registered|can-verify|can-verify-image|self-recovery|no-guid-matching
	  Vendor:               Gopod
	  VendorId:             USB:0x2D01
	  Version:              138.36.9.55
	  VersionFormat:        quad
	  VersionRaw:           0x8a240937
	  Icon:                 audio-card
	  InstallDuration:      15
	  Created:              2021-02-25
	  PhysicalId:           usb:04:00:01
	  LogicalId:            PD
	  Order:                18446744073709551615
	  DeviceKind:           VL103
	  FwOffset:             0x20000
	  FwSize:               0x8000

device test:

	$ ./hardware.py devices/hyper-no-sku-vli-tier*
	devices/hyper-no-sku-vli-tier1.json:
	Running test on Hyper USB-C Hub [VL817+VL103]
	[INFO]     Current version 80.83
	[INFO]     Installing version 90.83
	[SUCCESS]  Installed 90.83
	[INFO]     Installing version 80.83
	[SUCCESS]  Installed 80.83
	devices/hyper-no-sku-vli-tier3.json:
	Running test on Hyper USB-C Hub [VL817+VL103]
	[INFO]     Current version 138.36.9.55
	[INFO]     Installing version 154.36.9.55
	[SUCCESS]  Installed 154.36.9.55
	[INFO]     Installing version 138.36.9.55
	[SUCCESS]  Installed 138.36.9.55

closed 2021-02-25 by hughsie