fwupd/fwupd #2812: fuzzy search for firmware files to install

CragW, 2021-01-27:
**Describe the bug**
fwupdtool can't install firmware of given fuzzy path/to/*.cab; The course of operation does work properly if full filename is specified e.g. path/to/test.cab.

**Steps to Reproduce**
```
# fwupdtool install /tmp/*.cab
Loading…                 [***************************************] ]
WARNING: This package has not been validated, it may not work properly.
Failed to open file “/tmp/*.cab”: No such file or directory
Segmentation fault
```
**Expected behavior**
The result of fuzzy matching is saved to a list for install attempt

**fwupd version information**
```shell
$ fwupdmgr --version
client version: 1.5.5-94-gda7b898f
compile-time dependency versions
        gusb:   0.3.5
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
source

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
# skipped
```

**Additional questions**
- Operating system and version: Ubuntu 20.04.1 LTS (Focal Fossa)
- Have you tried rebooting? Yes
- Is this a regression? No

CragW, 2021-01-27:
filename = /tmp/*.cab
`fu_common_get_contents_bytes` finds no content for ***.cab** therefore returned back to the caller `fu_util_cmd_arrary_run` callback.

```
	filename = fu_util_download_if_required (priv, values[0], error);
	if (filename == NULL)
		return FALSE;

	/* parse silo */
	blob_cab = fu_common_get_contents_bytes (filename, error);
	if (blob_cab == NULL) {
		fu_util_maybe_prefix_sandbox_error (filename, error);
		return FALSE;
	}
```

hughsie, 2021-01-27:
The crash is at least fixed by https://github.com/fwupd/fwupd/pull/2814/commits/46764bd83f0827587ad77ccc866c95b050f9bbdc

hughsie, 2021-01-27:
Shouldn't the shell do the glob expansion?

CragW, 2021-01-27:
Yes, you are right. I think fuzzy search is in fact working properly, no issues in that part.  The fuzzy files **\*.cab** can be expanded correctly with its presence, and install worked properly if only 1 cab file exists.

I think the real issues are:
1. No **\*.cab** found at all, i.e.: `filename = /tmp/*.cab`, ultimately install failed (zero content)
2. Multiple **\*.cab** found: could be device no found if results = 2, also could be `Invalid arguments` if results > 2.

I will update the bug description.

hughsie, 2021-01-27:
I think `fwupdtool` isn't going to love multiple files and that's okay. Remember fwupdtool is really only for fwupd development and debugging -- most users should be using `fwupdmgr` and the dameon.

CragW, 2021-01-27:
Sure, I'm going to close this as won't fix.

closed 2021-01-27 by CragW