fwupd/fwupd #2830: SuperIO Offline Updates get "stuck"

Sean-StarLabs, 2021-01-29:
If any update is scheduled for SuperIO, and something changes (such as entering S3, disconnecting the charger) before the update is installed, then the update will get "stuck".

The update then no longer appears in GNOME Software, and trying `fwupdmgr update` will show:
```
1f664ad149f2151fb750b52751dbf9ba045a7846 is already scheduled to be updated
```
It requires running `sudo fwupdmgr clear-offline` followed by a `fwupdmgr update` to get it to install. 

I say this only knowing about the 3 plugins we use but would it not make sense to clear the pending updates each time the fwupd.service starts?

hughsie, 2021-01-29:
@superm1 makes sense to me; spot any logic problems? I guess we also need to check the boot time to avoid clearing it on idle quit.

superm1, 2021-01-29:
When does superio update get installed?  It's only in a clean rebooted session right?

Yes, I think checking for a changed boot is a good enough criteria to mark any pending updates like that.

hughsie, 2021-01-30:
Does https://github.com/fwupd/fwupd/pull/2835 work?

Sean-StarLabs, 2021-02-03:
I have not been able to fault it :)

closed 2021-02-03 by hughsie