fwupd/fwupd #2836: install_dell_bios_exe.py: ValueError: Namespace Fwupd not available

paulmenzel, 2021-02-01:
On Debian sid/unstable, trying to follow https://github.com/fwupd/firmware-dell/blob/master/README.md, the command fails.

```
$ dpkg -l fwupd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  fwupd          1.5.5-2      amd64        Firmware update daemon
$ fwupdmgr --version
client version:	1.5.5
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.5
$ sudo /usr/share/fwupd/install_dell_bios_exe.py XPS_13_9310_2-in-1_2.0.0.exe
Traceback (most recent call last):
  File "/usr/share/fwupd/install_dell_bios_exe.py", line 13, in <module>
    gi.require_version('Fwupd', '2.0')
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 126, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Fwupd not available
```

superm1, 2021-02-01:
Do you have the gir package installed too?

paulmenzel, 2021-02-01:
> Do you have the `gir1.2-fwupd-2.0` package installed too?

It was not installed. `apt install gir1.2-fwupd-2.0` fixed the issue. It’s not listed in the package metadata:

    Recommends: python3, bolt, dbus, secureboot-db, udisks2, fwupd-signed

There is no `Suggests` entry.

superm1, 2021-02-01:
Yeah we don't want it installed by default, so not in `Recommends` I think is the correct solution.  Suggests seems like a good consolation.

paulmenzel, 2021-02-01:
Sounds good.

Additionally, is there a way to catch this error, and print instructions to install the required packages?

superm1, 2021-02-01:
The package name is going to vary from OS to OS, I guess we can list a few things to try though maybe.

paulmenzel, 2021-02-01:
Indeed. Or link to some Wiki page or so.

commit c5f67747 (2021-02-01): trivial: Make it easier to know what packages to install to let
helper scripts run.

Fixes: #2836
PR #2837 (2021-02-01): trivial: Make it easier to know what packages to install to let
helper scripts run.

Fixes: #2836

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ ] Feature
- [x] Documentation
commit c794b754 (2021-02-01): trivial: Make it easier to know what packages to install to let
helper scripts run.

Fixes: #2836
closed 2021-02-01 by superm1