fwupd/fwupd #2840: Dell XPS 13 9310 2-in-1: `install_dell_bios_exe.py XPS_13_9310_2-in-1_2.0.0.exe` fails with `Couldn't find any UEFI devices`

paulmenzel, 2021-02-04:
As reported in https://github.com/fwupd/firmware-dell/issues/50#issuecomment-772423059 the script `install_dell_bios_exe.py` does not work on the Dell XPS 13 9310 2-in-1.

    $ fwupdmgr --version
    client version:	1.5.5
    compile-time dependency versions
    	gusb:	0.3.5
    
    daemon version:	1.5.5
    $ md5sum XPS_13_9310_2-in-1_2.0.0.exe
    dacf140b6bc3dc46993d000ea8c1e345  XPS_13_9310_2-in-1_2.0.0.exe
    $ sudo /usr/share/fwupd/install_dell_bios_exe.py XPS_13_9310_2-in-1_2.0.0.exe
    Couldn't find any UEFI devices

superm1, 2021-02-04:
As mentioned there - is it installed in UEFI mode?  Are UEFI capsule updates turned on in BIOS setup?
`fwupdmgr get-devices` output should indicate both of these.

paulmenzel, 2021-02-04:
$ ls /sys/firmware/efi
    config_table  efivars  esrt  fw_platform_size  fw_vendor  runtime  runtime-map  systab

paulmenzel, 2021-02-04:
> As mentioned there - is it installed in UEFI mode? Are UEFI capsule updates turned on in BIOS setup?
> `fwupdmgr get-devices` output should indicate both of these.

```
$ fwupdmgr get-devices
XPS 13 9310 2-in-1
│
├─KBG40ZPZ256G NVMe KIOXIA 256GB:
│     Device ID:          c430a03ca2a65dfe2412ff950c79c51f6aec1317
│     Summary:            NVM Express Solid State Drive
│     Current version:    10400105
│     Vendor:             KIOXIA Corporation (NVME:0x1E0F)
│     GUIDs:              fbef9958-2179-5fb1-9e91-31c22cc29b2d
│                         b2cca4ac-d65f-e811-9c2d-fa7ae01bbebc
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│   
├─System Firmware:
│ │   Device ID:          3799e725ec98d58dcf68eea254c50ef7dd261a4f
│ │   Current version:    65539
│ │   Minimum Version:    65539
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              a472a89c-42ab-c171-228e-e182e424af76
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       055d2140-2ac7-5568-ae08-d78b141978ae
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  83
│       Minimum Version:  83
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
├─Touchpad:
│     Device ID:          6042b531410f9a5c52c89376621f5382911db5ab
│     Current version:    1.3.3260652
│     Bootloader Version: 54.0.0
│     Vendor:             Synaptics (HIDRAW:0x06CB)
│     GUIDs:              c9d02f56-521a-5797-8842-255abc8b9187
│                         613c9012-bcb6-5809-9aaa-8e244cf2de0e
│                         bbfe9e12-4f9d-5bb7-b1fa-56c05ace5393
│                         181b0f39-26f3-54a8-a2b4-ceb67afc791b
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─UEFI Device Firmware:
│     Device ID:          6b9d757c1f64d696e7af69942c2d6ed3993479d4
│     Current version:    96
│     Minimum Version:    96
│     Vendor:             DMI:Dell Inc.
│     GUIDs:              176e090d-0ddb-495e-8173-bc998ccfecd0
│                         3e1d83c5-31af-55ac-9f88-8be02bda9596
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          7e3dfff3a5ac665b2b413a1a1a3babdadc1c3784
│     Current version:    272630021
│     Minimum Version:    272630021
│     Vendor:             DMI:Dell Inc.
│     GUIDs:              b2cca4ac-d65f-e811-9c2d-fa7ae01bbebc
│                         0360499c-14d8-53b8-938b-37de40587af4
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Wacom AES Device:
      Device ID:          dbbf40284cb602275c039924b932efc748233670
      Current version:    3006.00
      Vendor:             HIDRAW:0x056A
      GUIDs:              f2e31e0f-ab3b-5cd2-8a2f-462ada7ed50a
                          5f02cd21-7586-5650-95e9-388fdf0b6e63
                          a6531f36-4cd5-5878-a8f4-9efc1b7702b3
      Device Flags:       • Internal device
                          • Updatable
```

paulmenzel, 2021-02-04:
[sudo strace -s1000 /usr/share/fwupd/install_dell_bios_exe.py XPS_13_9310_2-in-1_2.0.0.exe](https://github.com/fwupd/fwupd/files/5926802/install_bios_exe.py-strace.txt)

paulmenzel, 2021-02-05:
Unfortunately, that didn’t fix the behavior. I print `item.get_name()` in the error:

    $ grep uefi-capsule /usr/share/fwupd/install_dell_bios_exe.py
        if item.get_plugin() == 'uefi' or item.get_plugin() == 'uefi-capsule':
    $ sudo /usr/share/fwupd/install_dell_bios_exe.py XPS_13_9310_2-in-1_2.0.0.exe 
    Couldn't find any UEFI devices; item.get_name() = Wacom AES Device

paulmenzel, 2021-02-05:
The plugin name is `wacom_raw`:

With

    print("Couldn't find any UEFI devices; item.get_name() = %s; item_get_plugin() = %s" % (item.get_name(), item.get_plugin()))

I get:

    Couldn't find any UEFI devices; item.get_name() = Wacom AES Device; item_get_plugin() = wacom_raw

paulmenzel, 2021-02-05:
With `print("deviceid = %s" % deviceid)`, the type of `deviceid` is `None`. I guess, a different format string needs to be used to print `deviceid`.

    $ sudo /usr/share/fwupd/install_dell_bios_exe.py XPS_13_9310_2-in-1_2.0.0.exe 
    deviceid = None
    Couldn't find any UEFI devices; item.get_name() = Wacom AES Device; item_get_plugin() = wacom_raw; item.get_id() = dbbf40284cb602275c039924b932efc748233670

PR #2841 (2021-02-04): trivial: contrib/install_dell_bios_exe.py: recognize the plugin rename
uefi was renamed to uefi-capsule and now this script forgot the change.
Allow both to work so this script runs on new or older fwupd.

Fixes: #2840

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit b5481f24 (2021-02-04): trivial: contrib/install_dell_bios_exe.py: recognize the plugin rename

uefi was renamed to uefi-capsule and now this script forgot the change.
Allow both to work so this script runs on new or older fwupd.

Fixes: #2840
commit b71b273c (2021-02-08): trivial: install_dell_bios_exe: fixup uefi-capsule -> uefi_capsule
Fixes: #2840
closed 2021-02-08 by superm1