fwupd/fwupd #2843: Fwupdmgr doesn't understand multiple ESPs

hpvb, 2021-02-05:
**Describe the bug**
I have a RAID1 setup with a systemd timer job to synchronize the two ESPs. Fwupdmgr does not enjoy this situation.

**Steps to Reproduce**
Run fwupdmgr and observe the following:
```
$ fwupdmgr 
Unknown                  [***************************************]
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Command not found
```
Setting `OverrideESPMountPoint=/boot/efi` does not resolve the situation (I tried both the `/etc/fwupd/uefi.conf` and `/etc/fwupd/uefi.conf`. Running `sudo /usr/bin/fwupdtool esp-list --verbose` lets me choose between my two ESPs.
```
$ sudo /usr/bin/fwupdtool esp-list --verbose
02:35:22:0151 FuDebug              Verbose debugging enabled (on console 1)
02:35:22:0168 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:35:22:0169 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:35:22:0172 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:35:22:0173 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
02:35:22:0174 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:35:22:0176 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
Choose a volume:
0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
2.	/org/freedesktop/UDisks2/block_devices/nvme1n1p1
1
/boot/efi/EFI/BOOT/BOOTIA32.EFI
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/BOOT/fbia32.efi
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/fedora/fonts/unicode.pf2
/boot/efi/EFI/fedora/shimia32-fedora.efi
/boot/efi/EFI/fedora/BOOTIA32.CSV
/boot/efi/EFI/fedora/BOOTX64.CSV
/boot/efi/EFI/fedora/gcdia32.efi
/boot/efi/EFI/fedora/gcdx64.efi
/boot/efi/EFI/fedora/grubia32.efi
/boot/efi/EFI/fedora/grubx64.efi
/boot/efi/EFI/fedora/mmia32.efi
/boot/efi/EFI/fedora/mmx64.efi
/boot/efi/EFI/fedora/shim.efi
/boot/efi/EFI/fedora/grubenv
/boot/efi/EFI/fedora/fw/fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
/boot/efi/EFI/fedora/fw/fwupd-74992bad-d49d-4f82-bb77-bbe865bea34e.cap
/boot/efi/EFI/fedora/fwupdx64.efi
/boot/efi/EFI/fedora/shimia32.efi
/boot/efi/EFI/fedora/shimx64-fedora.efi
/boot/efi/EFI/fedora/shimx64.efi
/boot/efi/EFI/fedora/grub.cfg
/boot/efi/EFI/Dell/Bios/Recovery/BIOS_PRE.rcv
/boot/efi/EFI/Dell/Bios/Recovery/BIOS_CUR.RCV
/boot/efi/System/Library/CoreServices/SystemVersion.plist
/boot/efi/mach_kernel
```

**Expected behavior**
For fwupdmgr to at least honor `OverrideESPMountPoint=/boot/efi` even if there's more than one ESP

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.5
compile-time dependency versions
	gusb:	0.3.5

Unknown                  [***************************************]
daemon version:	1.5.5
```
[... 131 lines cut ...]
│       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:     • Internal device
│     
└─UHD Graphics 630 (Mobile):
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    02
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              7993964e-33c2-59e5-b138-c69dc7971ebe
                          4687e99d-d897-5846-b959-902eee4bb1ed
                          db2a81f9-8177-5e23-8c0b-809f7b3314e3
                          db677673-1102-5481-a11e-6f397e123ced
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available

```

**Additional questions**
- Operating system and version:
Fedora 33 / 5.10.10-200.fc33.x86_64

- Have you tried rebooting?
Yes, also restarting fwupdmgr after changing settings

- Is this a regression?
I certainly have installed firmware updates through fwupmgr before. And judging by my `rpmsave` files I had `OverrideESPMountPoint=/boot/efi` in `/etc/fwupd/uefi.conf` before it got removed and presumably moved into `uefi_capsule.conf`. So I do suspect this is a regression.

hpvb, 2021-02-05:
Since I maybe left this a bit ambiguous earlier: The ESPs are *not* on RAID1. They are separate partitions entirely and are merely kept in sync from `/boot/efi` to `/boot/efi_back` with a simple timer:
```
#!/bin/bash

if grep -q '/boot/efi ' /proc/mounts; then
  rsync -a /boot/efi/ /boot/efi_back/
fi
```
Ran every 15 minutes

hughsie, 2021-02-05:
I wonder if you're hitting this (fixed) bug: https://github.com/fwupd/fwupd/commit/bb8c8040c722013645a6abaf45876ea60f87ae0a

i.e. the file needs to be called uefi_capsule.conf and the section header needs to be`[uefi_capsule]`

hpvb, 2021-02-05:
Ah yes! Indeed the shipped `uefi_capsule.conf` had a section header that was still `[uefi]` after correcting this the problem went away. Thank you!

closed 2021-02-05 by hpvb