fwupd/fwupd #2844: Q: updating monoitor firmware over HDMI

kloczek, 2021-02-05:
Some monirtors allows update firware over HDMI.
On Windowsk my Iiyama monitro allows to do something like this but I dpon't see fwupdmgr details bout monitor.
```
# fwupdmgr get-devices
Unknown                  [***************************************]
SYS-7039A-I
│
├─INTEL SSDPEKNW020T8:
│     Device ID:          73e5c64ab94eb335fb590af99ecd15b5ff191bfa
│     Summary:            NVM Express Solid State Drive
│     Current version:    002C
│     Vendor:             Intel Corporation (NVME:0x8086)
│     Serial Number:      BTNH90420DAE2P0C
│     GUIDs:              c5fe8b70-dc9a-5c3b-9634-659091d29812 ← NVME\VEN_8086&DEV_F1A8&SUBSYS_8086390D&REV_03
│                         1122104f-b10a-5f32-bc13-7a1ac0f52ea2 ← NVME\VEN_8086&DEV_F1A8&SUBSYS_8086390D
│                         c6cd9ab0-8f20-512e-9e1f-1af55b8454b9 ← NVME\VEN_8086&DEV_F1A8&REV_03
│                         82741c78-f5dc-5c23-a152-00de5799edc8 ← NVME\VEN_8086&DEV_F1A8
│                         7c0505be-b609-566e-bf94-561e441a9171 ← INTEL SSDPEKNW020T8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
└─Samsung SSD 860 EVO 250GB:
      Device ID:          a0dcbc176905a16a8b5b6e7374729284e7305f2d
      Summary:            ATA Drive
      Current version:    RVT02B6Q
      Vendor:             Samsung (ATA:0x144D|OUI:002538)
      Serial Number:      S3YJNX0M131858K
      GUIDs:              95a1ee2f-402f-570b-9c51-fde95db6849a ← IDE\Samsung_SSD_860_EVO_250GB_______________RVT02B6Q
                          5b23aa59-694d-5622-a0b6-d294b9821711 ← IDE\0Samsung_SSD_860_EVO_250GB_______________
                          00641f76-401e-5a04-8167-4c567e3116f8 ← Samsung SSD 860 EVO 250GB
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

hughsie, 2021-02-05:
> Some monirtors allows update firware over HDMI.

Are there any specifications about how to do this? I think it would require some development help from Iiyama.

kloczek, 2021-02-05:
Looking on what google shows on https://www.google.com/search?&q=update+monitor+firmware+hdmi+specification seems like more and more vendors provides such possibility so I'm guessing that it must be somehow part of the HDMI protocon specyyfication from some version.
HDMI as standard AFAIK is derivate of the ethernet ..

hughsie, 2021-02-05:
@superm1 do you have any knowledge about this? @kloczek without a vendor volunteering specifications, or someone volunteering to reverse engineer the protocol I'm afraid fwupd won't know how to do this.

superm1, 2021-02-05:
The aux channel that we updated mst hub is the way that this stuff works too. But yes we need a specification to make it happen. It's vendor specific.

kloczek, 2021-02-05:
@hughsie I understand.
If I'll find anything will drop infor about that here.
Nevertheless I think that sooner or later it will be possible to use some spec.
As I have Iiyama supprt will try to ask about some documentation ..

superm1, 2021-02-05:
Better than just releasing documentation it would be best to ask them to write a fwupd plugin themselves.  If they don't have the expertise in house, they can work with a consulting house to accomplish it: https://fwupd.org/lvfs/docs/consulting

closed 2021-02-05 by superm1