fwupd/fwupd #2847: What are the main differences vs  fwup-home/fwup?

lobo-tuerto, 2021-02-07:
**Describe the question**
I'm trying to get a Nerves project working on a Manjaro box. It requires me to install `fwup` 1.8.
The only info about installing fwup for Archlinux I did found was `fwupd`.
I see that the lastest version over here is **1.5.5** and the name is not `fwup` but `fwupd`.

I was wondering what are the main differences, between: 
https://github.com/fwup-home/fwup/
and
https://github.com/fwupd/fwupd

And why do we have two `fwup*` projects.

**fwupd version information**

```shell
$ fwupdmgr --version
```
fwupd version: 1.5.5

hughsie, 2021-02-07:
They are completely different things. fwupd is for installing firmware onto EEPROM devices, fwup is for installing the OS image to the target device. Both projects have comprehensive README files. :)

closed 2021-02-07 by hughsie

lobo-tuerto, 2021-02-07:
Thanks for the nice and prompt reply! :)