fwupd/fwupd #2848: Logitech-Unifying-RQR12.10_B0032.cab fails its checksum

liffiton, 2021-02-07:
**Describe the bug**
Logitech-Unifying-RQR12.10_B0032.cab fails its checksum.  The sha1 sum in the metadata doesn't match that of the actual file.

**Steps to Reproduce**
```
$ fwupdmgr refresh
$ fwupdmgr upgrade
Upgrade available for Unifying Receiver from RQR12.03_B0025 to RQR12.10_B0032
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: 
Downloading RQR12.10_B0032 for Unifying Receiver...
Fetching firmware https://fwupd.org/downloads/8d83954fcf79453738dbeba9615a095bae9caed9-Logitech-Unifying-RQR12.10_B0032.cab
Downloading…             [***************************************] Less than one minute remaining…
Checksum invalid, expected afe3c3bff4a35ef556a650f4ef7448455f97f636 got 4854e229502c4d266330a4902451fb7827f1d1d7
```

(I can replicate checksum it finds using `sha1sum` on the command line, and I see the checksum it expects in `/var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz`, updated a few minutes ago.)

**Expected behavior**
I expect it to successfully upgrade the receiver.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):  Installed via `apt`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
[others snipped]
├─Unifying Receiver:
│     Device ID:           612bbfcd50180da24c6cda2f1d3534083494efbe
│     Summary:             A miniaturised USB wireless receiver
│     Current version:     RQR12.03_B0025
│     Bootloader Version:  BOT01.02_B0015
│     Vendor:              USB:0x046D
│     Install Duration:    30 seconds
│     GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
│                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
│                          279ed287-3607-549e-bacc-f873bb9838c4
│     Device Flags:        • Updatable
│                          • Supported on remote server
```

**Additional questions**
- Operating system and version:  Ubuntu 20.04
- Have you tried rebooting?  No.
- Is this a regression?  I don't know.

hughsie, 2021-02-07:
I resigned a ton of firmware a few hours ago. Can you try "fwupdmgr refresh --force" and try again? Thanks.

liffiton, 2021-02-07:
I had tried that right before submitting the issue.  Repeated just now, I get the same result:

```
$ fwupdmgr refresh --force
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Successfully downloaded new metadata: 3 local devices supported

$ fwupdmgr upgrade
Upgrade available for Unifying Receiver from RQR12.03_B0025 to RQR12.10_B0032
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: 
Downloading RQR12.10_B0032 for Unifying Receiver...
Fetching firmware https://fwupd.org/downloads/8d83954fcf79453738dbeba9615a095bae9caed9-Logitech-Unifying-RQR12.10_B0032.cab
Downloading…             [***************************************]
Checksum invalid, expected afe3c3bff4a35ef556a650f4ef7448455f97f636 got 4854e229502c4d266330a4902451fb7827f1d1d7
```

Possibly a CDN update lag issue?  I can try again in a few hours or tomorrow.

hughsie, 2021-02-07:
I just purged the CDN; can you try again?

liffiton, 2021-02-07:
Same two commands, same result, I'm afraid.

hughsie, 2021-02-07:
Okay; I've regenerated stable manually; try one more time pls?

liffiton, 2021-02-07:
Yup, that worked!

[Well, it first made it past the checksum and reported `failed to get device after detach: failed to wait for detach replug: device 612bbfcd50180da24c6cda2f1d3534083494efbe did not come back`.  But then I re-ran it, and it got through "Writing..." and then gave that same failure message.  But after that second time it seems to be updated.  I ran `fwupdmgr get-devices`, and it asked me to submit a report on the failed update, which I did.  But it reports the correct version for that receiver now, and the receiver works.]

Thanks!

closed 2021-02-07 by liffiton