fwupd/fwupd #2850: fwupd general protection fault on goodix module

GoPerry, 2021-02-08:
**Describe the bug**
A clear and concise description of what the bug is.

[  127.233672] traps: fwupd[3143] general protection fault ip:7f8b431c8418 sp:7ffdd9cf5070 error:0 in libfu_plugin_goodixmoc.so[7f8b431c6000+4000]
[  153.110134] traps: fwupd[3430] general protection fault ip:7fb420762418 sp:7fffabbc3cc0 error:0 in libfu_plugin_goodixmoc.so[7fb420760000+4000]
[  187.859349] traps: fwupd[3520] general protection fault ip:7fdfc1514418 sp:7ffdbb81f680 error:0 in libfu_plugin_goodixmoc.so[7fdfc1512000+4000]
[  217.882256] traps: fwupd[3580] general protection fault ip:7f17e7b43418 sp:7ffd29f774b0 error:0 in libfu_plugin_goodixmoc.so[7f17e7b41000+4000]
[  290.685105] traps: fwupd[3946] general protection fault ip:7fa5efd4a418 sp:7ffe3da343c0 error:0 in libfu_plugin_goodixmoc.so[7fa5efd48000+4000]
[  322.316052] traps: fwupd[4111] general protection fault ip:7f5aff4fb418 sp:7fff6482c3e0 error:0 in libfu_plugin_goodixmoc.so[7f5aff4f9000+4000]

**Steps to Reproduce**
Steps to reproduce the behavior.
install  RHEL8.4  on FPG DVT system 

**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
[dell@localhost ~]$ fwupdmgr  --version
client version: 1.5.5
compile-time dependency versions
        gusb:   0.3.0

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
RHEL builtin package
**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```
[root@localhost dell]# fwupdmgr get-devices --show-all-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

**Additional questions**
- Operating system and version: 8.4
- Have you tried rebooting? : each bootimg cycle can reproduce the issue so far.
 
- Is this a regression? : not sure

superm1, 2021-02-08:
@GoPerry by chance can you launch the daemon manually without systemd and see if it reproduces?  If so, can you please get a backtrace?

GoPerry, 2021-02-08:
@superm1 
Here is a new issue found on my testing system FPG-DVT-XXX

`[root@localhost dell]# fwupdmgr get-devices --show-all-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
[root@localhost dell]# journalctl  -ex
                                                               #2  0x00007fc603716e53 fu_goodixmoc_device_setup (libfu_plugin_goodixmoc.so)
                                                               #3  0x00007fc62d5b3b95 fu_device_setup (libfwupdplugin.so.1)
                                                               #4  0x00007fc62d5b3d16 fu_device_open (libfwupdplugin.so.1)
                                                               #5  0x00007fc62d5ace49 fu_device_locker_new_full (libfwupdplugin.so.1)
                                                               #6  0x00007fc62d5c0453 fu_plugin_runner_usb_device_added (libfwupdplugin.so.1)
                                                               #7  0x000055ae1fa22ab4 fu_engine_usb_device_added_cb (fwupd)
                                                               #8  0x00007fc62ca413fd g_closure_invoke (libgobject-2.0.so.0)
                                                               #9  0x00007fc62ca54985 signal_emit_unlocked_R (libgobject-2.0.so.0)
                                                               #10 0x00007fc62ca5da46 g_signal_emit_valist (libgobject-2.0.so.0)
                                                               #11 0x00007fc62ca5e083 g_signal_emit (libgobject-2.0.so.0)
                                                               #12 0x00007fc62c0db575 g_usb_context_idle_signal_cb (libgusb.so.2)
                                                               #13 0x00007fc62c0dc440 g_usb_context_add_device (libgusb.so.2)
                                                               #14 0x00007fc62c0dc628 g_usb_context_rescan (libgusb.so.2)
                                                               #15 0x00007fc62c0dc6e8 g_usb_context_enumerate (libgusb.so.2)
                                                               #16 0x000055ae1fa3286f fu_engine_load (fwupd)
                                                               #17 0x000055ae1fa18052 main (fwupd)
                                                               #18 0x00007fc62b3c9493 __libc_start_main (libc.so.6)
                                                               #19 0x000055ae1fa1853e _start (fwupd)

                                                               Stack trace of thread 24036:
                                                               #0  0x00007fc62b497a31 __poll (libc.so.6)
                                                               #1  0x00007fc62807b5ef linux_udev_event_thread_main (libusb-1.0.so.0)
                                                               #2  0x00007fc62b77314a start_thread (libpthread.so.0)
                                                               #3  0x00007fc62b4a2db3 __clone (libc.so.6)

                                                               Stack trace of thread 24037:
                                                               #0  0x00007fc62b497a31 __poll (libc.so.6)
                                                               #1  0x00007fc628074807 handle_events.isra.2 (libusb-1.0.so.0)
                                                               #2  0x00007fc628075aab libusb_handle_events_timeout_completed (libusb-1.0.so.0)
                                                               #3  0x00007fc628075b03 libusb_handle_events (libusb-1.0.so.0)
                                                               #4  0x00007fc62c0dbac9 g_usb_context_event_thread_cb (libgusb.so.2)
                                                               #5  0x00007fc62c78de1a g_thread_proxy (libglib-2.0.so.0)
                                                               #6  0x00007fc62b77314a start_thread (libpthread.so.0)
                                                               #7  0x00007fc62b4a2db3 __clone (libc.so.6)

                                                               Stack trace of thread 24038:
                                                               #0  0x00007fc62b497a31 __poll (libc.so.6)
                                                               #1  0x00007fc62c765ab6 g_main_context_iterate.isra.21 (libglib-2.0.so.0)
                                                               #2  0x00007fc62c765e72 g_main_loop_run (libglib-2.0.so.0)
                                                               #3  0x00007fc62cd5b59a gdbus_shared_thread_func (libgio-2.0.so.0)
                                                               #4  0x00007fc62c78de1a g_thread_proxy (libglib-2.0.so.0)
                                                               #5  0x00007fc62b77314a start_thread (libpthread.so.0)
                                                               #6  0x00007fc62b4a2db3 __clone (libc.so.6)

                                                               Stack trace of thread 24033:
                                                               #0  0x00007fc62b497a31 __poll (libc.so.6)
[... cut ...]
-- The unit systemd-coredump@10-24040-0.service has successfully entered the 'dead' state.
Feb 07 21:33:28 localhost.localdomain dbus-daemon[981]: [system] Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
Feb 07 21:33:28 localhost.localdomain polkitd[980]: Unregistered Authentication Agent for unix-process:24020:346867 (system bus name :1.1309, object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8) (disconnected fr>
lines 1682-1750/1750 (END)
`

superm1, 2021-02-08:
Can you confirm with VID/PID for this Goodix device on your SUT? (the ones that are supposed to be supported are listed in https://github.com/fwupd/fwupd/blob/master/plugins/goodix-moc/goodixmoc.quirk)

GoPerry, 2021-02-08:
[root@localhost lib]# lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 008: ID 8087:0026 Intel Corp.
Bus 003 Device 005: ID 0c45:672e Microdia
Bus 003 Device 003: ID 27c6:63ac --> this one should be the goodix FP device.
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

GoPerry, 2021-02-08:
Bus 003 Device 003: ID 27c6:63ac
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x27c6
  idProduct          0x63ac
  bcdDevice            1.00
  iManufacturer           1 Goodix Technology Co., Ltd.
  iProduct                2 Goodix USB2.0 MISC
  iSerial                 3 UIDE765EC33_XXXX_MOC_B0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          3 UIDE765EC33_XXXX_MOC_B0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              4 MISC Data
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)

GoPerry, 2021-02-08:
@superm1 
which commands to start the daemon manually not from systemd?
and full logs command to capture logs if needed.

superm1, 2021-02-08:
something like 

```
gdb --args /usr/libexec/fwupd/fwupd --verbose
```

Then run `bt` when it crashed and hopefully we see the details of the thread it's crashing in.

If you can install debug symbols it will be a lot more helpful too.

hughsie, 2021-02-08:
Yup, definitely need the backtrace with fwpd-debuginfo installed please. If this is a system with a Red Hat RHEL subscription please also create a bug in https://bugzilla.redhat.com/ so that it gets prioritised appropriately.

GoPerry, 2021-02-09:
> Yup, definitely need the backtrace with fwupd-debuginfo installed please. If this is a system with a Red Hat RHEL subscription please also create a bug in https://bugzilla.redhat.com/ so that it gets prioritised appropriately.
Agreed, Will create one BZ and upload backtrace log

hughsie, 2021-02-09:
@GoPerry if you can upload the backtrace in the next few hours I might be able to sneak a fix into 8.4.

superm1, 2021-02-09:
I'm afraid@GoPerry is already started CNY, so probably not going to come quickly.

GoPerry, 2021-02-10:
> I'm afraid@GoPerry is already started CNY, so probably not going to come quickly.

> @GoPerry if you can upload the backtrace in the next few hours I might be able to sneak a fix into 8.4.
Hi Richard:
I am busy on some other issue.
Here i create one BZ for you
https://bugzilla.redhat.com/show_bug.cgi?id=1927091

Meanwhile i am installing latest RHEL8.4 nightly build .
Will install the debug package and provide the backtrace later today

superm1, 2021-02-10:
@GoPerry one more thing - it would be useful to set `FWUPD_GOODIXFP_VERBOSE=1` in your environment while getting backtrace and log.

GoPerry, 2021-02-10:
`Reading symbols from /usr/libexec/fwupd/fwupd...Reading symbols from /usr/lib/debug/usr/libexec/fwupd/fwupd-1.5.5-1.el8.x86_64.debug...done.
done.
(gdb) run
Starting program: /usr/libexec/fwupd/fwupd --verbose
Missing separate debuginfos, use: yum debuginfo-install glibc-2.28-145.el8.x86_64
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7fffecaa6700 (LWP 10085)]
[New Thread 0x7fffe7fff700 (LWP 10086)]
13:00:04:0391 FuDebug              Verbose debugging enabled (on console 1)
13:00:04:0391 FuConfig             loading config values from /etc/fwupd/daemon.conf
13:00:04:0391 FuRemoteList         loading /usr/share/fwupd/metainfo
13:00:04:0391 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
13:00:04:0391 XbSilo               file: 67f769dc-a8b3-0341-66a2-9c46de691717, current:67f769dc-a8b3-0341-66a2-9c46de691717, cached: (null)
13:00:04:0391 XbSilo               loading silo with file contents
13:00:04:0391 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
13:00:04:0391 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
13:00:04:0391 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
13:00:04:0391 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
13:00:04:0392 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
13:00:04:0392 FuRemoteList         ordering lvfs-testing=lvfs+1
13:00:04:0392 FuRemoteList         ignoring unfound remote fwupd
13:00:04:0392 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
13:00:04:0392 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
13:00:04:0392 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
13:00:04:0392 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
13:00:04:0392 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
13:00:04:0392 Jcat                 loaded 1 certificates
13:00:04:0392 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
13:00:04:0392 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
13:00:04:0392 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
13:00:04:0392 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
13:00:04:0392 Jcat                 loaded 1 certificates
13:00:04:0392 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
13:00:04:0392 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
13:00:04:0396 FuEngine             client certificate exists and working
13:00:04:0396 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
13:00:04:0396 FuHistory            got schema version of 6
13:00:04:0396 FuIdle               setting timeout to 7200s
13:00:04:0396 FuHwids              smbios property Manufacturer=Dell Inc.
13:00:04:0396 FuHwids              smbios property EnclosureKind=1f
13:00:04:0396 FuHwids              smbios property Family=Precision
13:00:04:0396 FuHwids              smbios property ProductName=Precision 5560
13:00:04:0396 FuHwids              smbios property ProductSku=0A62
13:00:04:0396 FuHwids              smbios property BiosVendor=Dell Inc.
13:00:04:0396 FuHwids              smbios property BiosVersion=0.3.1
13:00:04:0396 FuHwids              smbios property BiosMajorRelease=00
13:00:04:0396 FuHwids              smbios property BiosMinorRelease=03
13:00:04:0396 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
13:00:04:0396 FuHwids              smbios property BaseboardProduct=
13:00:04:0398 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
13:00:04:0398 XbSilo               file: 4ce30941-f85e-640a-647d-2e418e717438, current:4ce30941-f85e-640a-647d-2e418e717438, cached: (null)
13:00:04:0398 XbSilo               loading silo with file contents
13:00:04:0398 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
13:00:04:0398 XbSilo               file: 81f3ea63-9b21-d078-0ff3-472ba35997f7, current:81f3ea63-9b21-d078-0ff3-472ba35997f7, cached: (null)
13:00:04:0398 XbSilo               loading silo with file contents
13:00:04:0398 FuEngine             1 components now in silo
[New Thread 0x7fffe77fe700 (LWP 10087)]
[New Thread 0x7fffe6ffd700 (LWP 10088)]
13:00:04:0401 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_solokey.so)
13:00:04:0402 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_acpi_dmar.so)
13:00:04:0403 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_steelseries.so)
13:00:04:0404 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_acpi_facp.so)
13:00:04:0406 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_superio.so)
13:00:04:0410 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_altos.so)
13:00:04:0411 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_amt.so)
13:00:04:0412 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_ata.so)
13:00:04:0412 FuPlugin             added udev subsystem watch of block
13:00:04:0419 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_uefi_capsule.so)
[... cut ...]
#16 0x00007ffff6474575 in g_usb_context_idle_signal_cb () from /lib64/libgusb.so.2
#17 0x00007ffff6475440 in g_usb_context_add_device () from /lib64/libgusb.so.2
#18 0x00007ffff6475628 in g_usb_context_rescan () from /lib64/libgusb.so.2
#19 0x00007ffff64756e8 in g_usb_context_enumerate () from /lib64/libgusb.so.2
#20 0x000055555558686f in fu_engine_load (self=0x5555557fd0d0, flags=flags@entry=14, error=error@entry=0x7fffffffe1e8) at ../src/fu-engine.c:6510
#21 0x000055555556c052 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:1972
`

GoPerry, 2021-02-10:
[image]

GoPerry, 2021-02-10:
@hughsie @superm1 
I capture some bt for the issue.
I tried two FPG DVT1 system, they all can reproduce the issue.

GoPerry, 2021-02-10:
(gdb) l
59      __NTH (memset (void *__dest, int __ch, size_t __len))
60      {
61        /* GCC-5.0 and newer implements these checks in the compiler, so we don't
62           need them here.  */
63      #if !__GNUC_PREREQ (5,0)
64        if (__builtin_constant_p (__len) && __len == 0
65            && (!__builtin_constant_p (__ch) || __ch != 0))
66          {
67            __warn_memset_zero_len ();
68            return __dest;

GoPerry, 2021-02-10:
`  328.593541] traps: fwupd[4158] general protection fault ip:7fe3d0cf4418 sp:7ffd16fa1260 error:0 in libfu_plugin_goodixmoc.so[7fe3d0cf2000+4000]
[  332.551401] usb 3-10: USB disconnect, device number 2
[  332.894941] usb 3-10: new full-speed USB device number 5 using xhci_hcd
[  333.022157] usb 3-10: New USB device found, idVendor=27c6, idProduct=63ac, bcdDevice= 1.00
[  333.022158] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  333.022159] usb 3-10: Product: Goodix USB2.0 MISC
[  333.022159] usb 3-10: Manufacturer: Goodix Technology Co., Ltd.
[  333.022160] usb 3-10: SerialNumber: UIDC9864493_XXXX_MOC_B0
[  354.329643] traps: fwupd[4450] general protection fault ip:7f68b0efb418 sp:7fffb0e0d760 error:0 in libfu_plugin_goodixmoc.so[7f68b0ef9000+4000]
[  358.287290] usb 3-10: USB disconnect, device number 5
[  358.631338] usb 3-10: new full-speed USB device number 6 using xhci_hcd
[  358.758751] usb 3-10: New USB device found, idVendor=27c6, idProduct=63ac, bcdDevice= 1.00
[  358.758752] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  358.758752] usb 3-10: Product: Goodix USB2.0 MISC
[  358.758753] usb 3-10: Manufacturer: Goodix Technology Co., Ltd.
[  358.758754] usb 3-10: SerialNumber: UIDC9864493_XXXX_MOC_B0
[  617.407710] traps: fwupd[5078] general protection fault ip:7f9086ba0418 sp:7ffdb8383db0 error:0 in libfu_plugin_goodixmoc.so[7f9086b9e000+4000]
[  621.364295] usb 3-10: USB disconnect, device number 6
[  621.708866] usb 3-10: new full-speed USB device number 7 using xhci_hcd
[  621.836381] usb 3-10: New USB device found, idVendor=27c6, idProduct=63ac, bcdDevice= 1.00
[  621.836383] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  621.836383] usb 3-10: Product: Goodix USB2.0 MISC
[  621.836384] usb 3-10: Manufacturer: Goodix Technology Co., Ltd.
[  621.836384] usb 3-10: SerialNumber: UIDC9864493_XXXX_MOC_B0
[ 5949.061648] traps: fwupd[9618] general protection fault ip:7fd0c33da418 sp:7fff45708e00 error:0 in libfu_plugin_goodixmoc.so[7fd0c33d8000+4000]
[ 5953.017723] usb 3-10: USB disconnect, device number 7
[ 5953.356808] usb 3-10: new full-speed USB device number 8 using xhci_hcd
[ 5953.484223] usb 3-10: New USB device found, idVendor=27c6, idProduct=63ac, bcdDevice= 1.00
[ 5953.484224] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5953.484225] usb 3-10: Product: Goodix USB2.0 MISC
[ 5953.484225] usb 3-10: Manufacturer: Goodix Technology Co., Ltd.
[ 5953.484226] usb 3-10: SerialNumber: UIDC9864493_XXXX_MOC_B0
[ 6056.690714] traps: fwupd[9755] general protection fault ip:7f1a8696b418 sp:7ffee60ec170 error:0 in libfu_plugin_goodixmoc.so[7f1a86969000+4000]
[ 6060.648349] usb 3-10: USB disconnect, device number 8
[ 6060.988750] usb 3-10: new full-speed USB device number 9 using xhci_hcd
[ 6061.116005] usb 3-10: New USB device found, idVendor=27c6, idProduct=63ac, bcdDevice= 1.00
[ 6061.116006] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6061.116007] usb 3-10: Product: Goodix USB2.0 MISC
[ 6061.116008] usb 3-10: Manufacturer: Goodix Technology Co., Ltd.
[ 6061.116008] usb 3-10: SerialNumber: UIDC9864493_XXXX_MOC_B0
[ 6144.978732] traps: fwupd[9884] general protection fault ip:7f9cf2fb6418 sp:7ffcbc63cc10 error:0 in libfu_plugin_goodixmoc.so[7f9cf2fb4000+4000]
[ 6148.934850] usb 3-10: USB disconnect, device number 9
[ 6149.276710] usb 3-10: new full-speed USB device number 10 using xhci_hcd
[ 6149.403962] usb 3-10: New USB device found, idVendor=27c6, idProduct=63ac, bcdDevice= 1.00
[ 6149.403963] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6149.403964] usb 3-10: Product: Goodix USB2.0 MISC
[ 6149.403964] usb 3-10: Manufacturer: Goodix Technology Co., Ltd.
[ 6149.403964] usb 3-10: SerialNumber: UIDC9864493_XXXX_MOC_B0
[ 6222.383231] usb 3-10: USB disconnect, device number 10
[ 6222.724695] usb 3-10: new full-speed USB device number 11 using xhci_hcd
[ 6222.851948] usb 3-10: New USB device found, idVendor=27c6, idProduct=63ac, bcdDevice= 1.00
[ 6222.851949] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6222.851950] usb 3-10: Product: Goodix USB2.0 MISC
[ 6222.851950] usb 3-10: Manufacturer: Goodix Technology Co., Ltd.
[ 6222.851950] usb 3-10: SerialNumber: UIDC9864493_XXXX_MOC_B0
[ 6271.928482] usb 3-10: USB disconnect, device number 11
[ 6272.268680] usb 3-10: new full-speed USB device number 12 using xhci_hcd
[ 6272.396054] usb 3-10: New USB device found, idVendor=27c6, idProduct=63ac, bcdDevice= 1.00
[ 6272.396056] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6272.396057] usb 3-10: Product: Goodix USB2.0 MISC
[ 6272.396058] usb 3-10: Manufacturer: Goodix Technology Co., Ltd.
[ 6272.396058] usb 3-10: SerialNumber: UIDC9864493_XXXX_MOC_B0
[ 6598.038940] usb 3-10: USB disconnect, device number 12
[ 6598.380483] usb 3-10: new full-speed USB device number 13 using xhci_hcd
[ 6598.507835] usb 3-10: New USB device found, idVendor=27c6, idProduct=63ac, bcdDevice= 1.00
[ 6598.507836] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6598.507836] usb 3-10: Product: Goodix USB2.0 MISC
[ 6598.507837] usb 3-10: Manufacturer: Goodix Technology Co., Ltd.
[ 6598.507837] usb 3-10: SerialNumber: UIDC9864493_XXXX_MOC_B0
[ 6667.756987] usb 3-10: USB disconnect, device number 13
[... 35 lines cut ...]
[... cut ...]
                                                               #4  0x00007ffff6b26e1a g_thread_proxy (libglib-2.0.so.0)
                                                               #5  0x00007ffff5b0c14a start_thread (libpthread.so.0)
                                                               #6  0x00007ffff583bdb3 __clone (libc.so.6)
-- Subject: Process 10375 (fwupd) dumped core
-- Defined-By: systemd
-- Support: https://access.redhat.com/support
-- Documentation: man:core(5)
--
-- Process 10375 (fwupd) crashed and dumped core.
`

hughsie, 2021-02-10:
Awesome, thanks. I've spotted the problem and will have a test branch ready in an hour or so. Do you need an RPM to test this or are you comfortable with building from source?

GoPerry, 2021-02-10:
> Awesome, thanks. I've spotted the problem and will have a test branch ready in an hour or so. Do you need an RPM to test this or are you comfortable with building from source?

I don`t have the building environment yet.
I can download the rpm from brew

hughsie, 2021-02-10:
@GoPerry I'm just building you an 8.4.0 RPM now. I'll continue RHEL discussions on the RH BZ ticket.

hughsie, 2021-02-10:
@GoPerry RHEL ticket updated with a brew build.

GoPerry, 2021-02-10:
@hughsie 
Yep,testing result updated with positive result in RHBZ.
Thanks for the quick fix patch.

GoPerry, 2021-02-10:
─Fingerprint Sensor:
│     Device ID:          d3612947f029e319ddda982f965ca1c1bd15ab8e
│     Summary:            Match-On-Chip Fingerprint Sensor
│     Current version:    01000254
│     Vendor:             Goodix (USB:0x27C6)
│     Install Duration:   10 seconds
│     GUIDs:              267adb9d-b0eb-5ad0-9a94-db1e15359ebe
│                         6f754a46-4a1d-590e-9c92-2ddafd352253
│     Device Flags:       • Updatable
│                         • Device can recover flash failures

closed 2021-02-10 by hughsie