fwupd/fwupd #2857: Logi-branded Logitech unifying receiver not showing any releases

cassidyjames, 2021-02-09:
**Describe the bug**

My wife’s Logitech unifying receiver came with my wife’s Mac-style keyboard and is stamped with the "logi" mark. It's not a nano-style. It shows in fwupd and GNOME Firmware, but does not show any releases.

Front | Back | End
----- | ---- | ---
[image] | [image] | [image]

**Steps to Reproduce**
Steps to reproduce the behavior.

1. Acquire this unifying receiver
2. Plug it in
3. Look at the device in GNOME Firmware

**Expected behavior**
A clear and concise description of what you expected to happen.

I would hope it would show releases and be updated like my other Logitech Unifying receivers. :)

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Installed from Ubuntu 20.04 LTS archives via `apt`.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
Precision 5530
│
├─Thunderbolt Controller:
│     Device ID:           a173b2bdf1899f27c141c3c16e833da7767e5785
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               adeb70f4-7473-593d-b27b-e65528d1ddab
│                          cbf2f081-b15f-5285-96c1-b273c2a4b762
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─8Bitdo NES30 Pro:
│     Device ID:           3b73641b7c6edee3c153b2fe554e49bde3606a96
│     Summary:             A redesigned classic game controller
│     Current version:     4.10
│     Vendor:              8Bitdo (USB:0x2DC8)
│     Install Duration:    2 minutes
│     GUIDs:               043c7673-bd45-5e2e-b8b9-c22df2dfec2a
│                          b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
│                          9326b31c-db11-5e23-921c-9ab20c21781c
│                          c6566b1b-0c6e-5d2e-9376-78c23ab57bf2
│     Device Flags:        • Updatable
│                          • Supported on remote server
│                          • Requires a bootloader
│                          • Device will not re-appear after update completes
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
[... 133 lines cut ...]
│                          279ed287-3607-549e-bacc-f873bb9838c4
│     Device Flags:        • Updatable
│                          • Supported on remote server
│   
└─Unifying Receiver:
      Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.07_B0030
      Bootloader Version:  BOT03.02_B0009
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server
    
```

It looks like it's `fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027 ` that is not showing any releases. Is this because its version is higher than what is available in the LVFS? If so, I would still hope to see older versions listed here.

**Additional questions**
- Operating system and version: elementary OS 6 (Ubuntu 20.04-based)
- Have you tried rebooting? Yes
- Is this a regression? Not sure

cassidyjames, 2021-02-09:
lsusb information

```shell
$ lsusb -v -s 37

Bus 001 Device 037: ID 046d:c52b Logitech, Inc. Unifying Receiver
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        32
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc52b Unifying Receiver
  bcdDevice           24.07
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      59
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     148
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      98
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               2
```

cassidyjames, 2021-02-09:
@hughsie it does indeed show a release in testing and is upgradable there.

superm1, 2021-02-10:
In that case it's not a fwupd bug, but more a question when Logi will promote it to stable. Closing the issue in fwupd.

closed 2021-02-10 by superm1

RichardFevrier, 2023-04-10:
Bumped [here](https://github.com/Logitech/fw_updates/issues/38)