fwupd/fwupd #2873: plugins/uefi-capsule/efi/meson.build:159:4: ERROR: Problem encountered: -Defi_os_dir is unset

joakim-tjernlund, 2021-02-11:
**Describe the bug**
Just tried to biuld master (been a few weeks since I did that) and got:
```
Configuring vendor.conf using configuration
Configuring vendor-directory.conf using configuration
Configuring fwupdtool using configuration
Configuring fwupdmgr using configuration
Configuring 85-fwupd using configuration
Configuring org.freedesktop.fwupd.service using configuration
Program /usr/bin/vapigen-0.48 found: YES
Program help2man found: YES
Compiler for C supports arguments -fcf-protection: YES 
Program help2man found: YES
Has header "/usr/include/efi/x86_64/efibind.h" : YES 
Message: efi-libdir: "/usr/lib64"
Message: efi-ldsdir: "/usr/lib64"
Message: efi-includedir: "/usr/include/efi"

plugins/uefi-capsule/efi/meson.build:159:4: ERROR: Problem encountered: -Defi_os_dir is unset, see plugins/uefi-capsule/README.md
```

I figure fwupd should not require this by default?

joakim-tjernlund, 2021-02-11:
if I unset supported_build I get this annoying msg:
```
WARNING: This package has not been validated, it may not work properly.
```

How can I keep supported_build? What should I set efi_os_dir to(without breaking anything)?

hughsie, 2021-02-11:
> How can I keep supported_build

You either need `supported_build=false` or you need to define `efi_os_dir` -- it's like that by design for exactly this reason.

joakim-tjernlund, 2021-02-11:
> > How can I keep supported_build
> 
> You either need `supported_build=false` or you need to define `efi_os_dir` -- it's like that by design for exactly this reason.

OK, but I don't know this stuff. What should I set efi_os_dir to without breaking anything ?

superm1, 2021-02-11:
what do you use on the esp?  `/boot/efi/$FOO/`.  It should be whatever `$FOO` evaluates to.

joakim-tjernlund, 2021-02-11:
OK, I guess it should be BOOT then:
```
cd /boot/EFI/BOOT/
# >  ls
./  ../  BOOTX64.EFI*

joakim-tjernlund, 2021-02-11:
Somewhat odd I have to specified that now.
Maybe you should consider dropping this warning instead:
    WARNING: This package has not been validated, it may not work properly.

it is the only reason for me to have supported_build=true

superm1, 2021-02-11:
> OK, I guess it should be BOOT then:
> 
> ```
> cd /boot/EFI/BOOT/
> # >  ls
> ./  ../  BOOTX64.EFI*
> ```

Normally do you set something in `uefi_capsule.conf` for overriding the ESP location?

superm1, 2021-02-11:
> Somewhat odd I have to specified that now.
> Maybe you should consider dropping this warning instead:
> WARNING: This package has not been validated, it may not work properly.
> 
> it is the only reason for me to have supported_build=true

That's entirely the intentional, if automation builds a package and no one has checked it to let people know it could have some problems.

joakim-tjernlund, 2021-02-11:
> > OK, I guess it should be BOOT then:
> > ```
> > cd /boot/EFI/BOOT/
> > # >  ls
> > ./  ../  BOOTX64.EFI*
> > ```
> 
> Normally do you set something in `uefi_capsule.conf` for overriding the ESP location?

No, it just looks like so:
```
[uefi_capsule]

# the shim loader is required to chainload the fwupd EFI binary unless
# the fwupd.efi file has been self-signed manually
#DisableShimForSecureBoot=true

# the EFI system partition path used
# if this is is not /boot/efi, /boot, or /efi
#OverrideESPMountPoint=

# amount of free space required on the ESP, for example using 0x2000000 for 32Mb
#RequireESPFreeSpace=

# with the UEFI removable path enabled, the default esp path is set to /EFI/boot
# the shim EFI binary and presumably this is $ESP/EFI/boot/bootx64.efi
#FallbacktoRemovablePath=false
```

joakim-tjernlund, 2021-02-11:
> That's entirely the intentional, if automation builds a package and no one has checked it to let people know it could have some problems.

I get it is intentional but it is also annoying for people like me to see this warning all the even though all is well.
What if one could just set -Defi_os_dir="" and then have the previous behavior?

superm1, 2021-02-11:
> What if one could just set -Defi_os_dir="" and then have the previous behavior?

OK so let's think all this through.

1. If you don't set `efi_os_dir` then basically this is the codepath that is run:
https://github.com/fwupd/fwupd/blob/6188be1dd4dbd43a50001df3b93a75ab241c1309/plugins/uefi-capsule/fu-uefi-common.c

IE It will try to figure out the os via `/etc/os-release`.  If it can't figure it out from that it will set it to `unknown`.

2. If you *do* set it then we end up with `FallbacktoRemovablePath` not working anymore.  I don't think that's intentional, but we might need to consider it.  This seems like an unintentional bug.

3. If you don't set `efi_os_dir` then what goes into the new SBAT distro id field?  https://github.com/fwupd/fwupd/blob/378b00018b42bdaefe50e70fa546065a41ee0e42/plugins/uefi-capsule/efi/meson.build#L174
This is important for the purpose of chainloading a shim, and we currently expect it matches what's on the ESP.

>No, it just looks like so:

Does UEFI capsule work for you without `efi_os_dir` set?

joakim-tjernlund, 2021-02-11:
> 
> Does UEFI capsule work for you without `efi_os_dir` set?

The I also must set supported_build=false and doing so fwupd seem to work when I do fwupdmgr refresh, fwupdmgr get-updates etc.

There is no new UEFI capsule for my Lenovo tso I cannot really tell for sure.

superm1, 2021-02-11:
So what is in your ESP then right now?  What did it do last time?  The `fwupdx64.efi` app should still be there.

If you run `fwupdmgr reinstall` it will re-stage the update.

joakim-tjernlund, 2021-02-11:
> So what is in your ESP then right now? What did it do last time? The `fwupdx64.efi` app should still be there.
```
# cd /boot/
se-jocke3-lx boot # ls
./   EFI/   intel-uc.img*  lost+found/	System.map-5.10.14-x86_64*  vmlinuz-5.10.14-x86_64*
../  grub/  .keep*	   rescue*	System.map-5.4.87-x86_64*   vmlinuz-5.4.87-x86_64*
se-jocke3-lx boot # ls EFI/BOOT/
./  ../  BOOTX64.EFI*
```
> 
> If you run `fwupdmgr reinstall` it will re-stage the update.

How safe is that? I just have remote access ATM and don't want reinstall every update

superm1, 2021-02-11:
I would suggest to just do `fwupdmgr reinstall` and then delete `BootNext` and don't reboot system in that instance.  But we can see what your system puts in ESP (or if it works).

joakim-tjernlund, 2021-02-11:
> I would suggest to just do `fwupdmgr reinstall` and then delete `BootNext` and don't reboot system in that instance. But we can see what your system puts in ESP (or if it works).

So I did a fwupdmgr reinstall and got this:
```
e-jocke3-lx EFI # ls
./  ../  BOOT/	gentoo/
se-jocke3-lx EFI # ls gentoo/
./  ../  fw/  fwupdx64.efi*
```

Now I just need to figure out how to delete BootNext:
```
efibootmgr -N
Boot entry 0001 does not exist
se-jocke3-lx boot # efibootmgr 
BootNext: 0000
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0010,0011,0012,0013,001D,0019,0017,0018,001A,001B,001C,001E,0023,0000
Boot0000* Linux-Firmware-Updater
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Startup Interrupt Menu
Boot0015  Rescue and Recovery
Boot0016  MEBx Hot Key
Boot0017* USB CD
Boot0018* USB FDD
Boot0019* NVMe0
Boot001A* ATA HDD0
Boot001B* USB HDD
Boot001C* PCI LAN
Boot001D  Other CD
Boot001E  Other HDD
Boot001F* USBR BOOT CDROM
Boot0020* USBR BOOT Floppy
Boot0021* ATA HDD
Boot0022* ATAPI CD
Boot0023* PCI LAN
```

joakim-tjernlund, 2021-02-11:
I guess this means I should just set efi_os_dir=gentoo ?

superm1, 2021-02-11:
Yeah that's what it means.  Or in the ebuild parse `/etc/os-release` (or whatever gentoo provides).
But really we need to think about behavior of using `efi_os_dir` due to my point #2.

joakim-tjernlund, 2021-02-11:
This gentoo dir, does other SW pkgs(like grub) be aware of that dir/name ?

hughsie, 2021-02-11:
@superm1 do you think we should go back to the `efi_sbat_distro_id` separate meson build option?

superm1, 2021-02-11:
> @superm1 do you think we should go back to the `efi_sbat_distro_id` separate meson build option?

That's what I'm starting to think, yes.

closed 2021-02-12 by hughsie