fwupd/fwupd #2878: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

AnimiVulpis, 2021-02-12:
**Describe the bug**
A clear and concise description of what the bug is.
When running `sudo fwupdmgr refresh` I rarely run into this:

```shell
$ sudo fwupdmgr refresh
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

**Steps to Reproduce**

- Run `sudo fwupdmgr refresh` after booting
  - The problem happens rarely and only after a fresh boot (no matter how after the boot though)
  - It works normal again after running `systemctl restart fwupd.service`

**Expected behavior**
Don't timeout ever ^^

**fwupd version information**

```shell
$ fwupdmgr --version
client version:	1.5.5
compile-time dependency versions
	gusb:	0.3.5

Unknown                  [***************************************]
daemon version:	1.5.5
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
- Installed via `pacman`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
Unknown                  [***************************************]
20L50004GE
│
├─USB3.1 Hub:
│ │   Device ID:          23d940eac81a886d3523a7d731f2b68a2bedb0f7
│ │   Summary:            USB 3.x Hub
│ │   Current version:    12.134
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   GUIDs:              227d0f95-a764-598e-85de-3d2bd0a7372f
│ │                       6b74e56f-2d70-59c2-a56a-f09360246022
│ │                       c6381f16-fe6d-5c45-a4ad-b619af4f0f0e
│ │                       fc6cfb35-d4e9-5467-a014-8863de104969
│ │                       23344cfe-9a45-535b-a7a0-af7ecfaffeea
│ │                       ef65710a-3fa0-5330-9a61-5ced5e976be1
│ │                       eaf17a36-88e9-5757-8e99-a14b76fb20f3
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │
│ ├─USB3.1 Hub:
│ │     Device ID:        8965ddc69a0691eb83b5ba8d64c4a992717af997
│ │     Summary:          USB 3.x Hub
│ │     Current version:  12.133
│ │     Vendor:           VIA Labs, Inc. (USB:0x17EF)
│ │     Install Duration: 15 seconds
│ │     GUIDs:            7a216856-8a97-550c-882e-8233751c7cf2
│ │                       949252c2-0971-54f3-b8ac-8782f4ba1436
│ │                       8b2439a7-f319-5583-950a-236eeecfe918
│ │                       41802a3a-da33-52c9-92d0-b3e4c8fc45d8
│ │                       cf5f91c2-cb87-5e19-bef9-20ef14a2bcdb
[... 690 lines cut ...]
Feb 12 12:22:56 dev-110 kernel: usb 1-4.3.3.1: new full-speed USB device number 12 using xhci_hcd
Feb 12 12:22:56 dev-110 kernel: usb 1-4.3.3.1: New USB device found, idVendor=04b4, idProduct=521a, bcdDevice= 0.00
Feb 12 12:22:56 dev-110 kernel: usb 1-4.3.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Feb 12 12:22:56 dev-110 kernel: usb 1-4.3.3.1: Product: USB-I2C Bridge
Feb 12 12:22:56 dev-110 kernel: usb 1-4.3.3.1: Manufacturer: Cypress Semiconductor
Feb 12 12:22:56 dev-110 mtp-probe[6824]: checking bus 1, device 12: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.3/1-4.3.3/1-4.3.3.1"
Feb 12 12:22:56 dev-110 mtp-probe[6824]: bus: 1, device: 12 was not an MTP device
Feb 12 12:22:56 dev-110 mtp-probe[6827]: checking bus 1, device 12: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.3/1-4.3.3/1-4.3.3.1"
Feb 12 12:22:56 dev-110 mtp-probe[6827]: bus: 1, device: 12 was not an MTP device
Feb 12 12:23:20 dev-110 dbus-daemon[682]: [system] Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
Feb 12 12:23:20 dev-110 polkitd[769]: Unregistered Authentication Agent for unix-process:6802:14304 (system bus name :1.39, object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8) (disconnected from bus)
Feb 12 12:23:20 dev-110 audit[6801]: USER_END pid=6801 uid=1000 auid=1000 ses=2 subj==unconfined msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
Feb 12 12:23:20 dev-110 kernel: audit: type=1106 audit(1613129000.535:260): pid=6801 uid=1000 auid=1000 ses=2 subj==unconfined msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
Feb 12 12:23:20 dev-110 kernel: audit: type=1104 audit(1613129000.535:261): pid=6801 uid=1000 auid=1000 ses=2 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
Feb 12 12:23:20 dev-110 audit[6801]: CRED_DISP pid=6801 uid=1000 auid=1000 ses=2 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
Feb 12 12:23:20 dev-110 sudo[6801]: pam_unix(sudo:session): session closed for user root
Feb 12 12:24:25 dev-110 systemd[1]: fwupd.service: start operation timed out. Terminating.
Feb 12 12:24:25 dev-110 systemd[1]: fwupd.service: Failed with result 'timeout'.
Feb 12 12:24:25 dev-110 systemd[1]: Failed to start Firmware update daemon.
Feb 12 12:24:25 dev-110 audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=fwupd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Feb 12 12:24:25 dev-110 kernel: audit: type=1130 audit(1613129065.782:262): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=fwupd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
# [...]
```

I hope that helps pinpointing this elusive problem.

superm1, 2021-02-12:
Thanks for all the work in debugging with adding `--verbose`.
Is this perhaps because of the device it's trying to read timing out?
```
Feb 12 12:22:56 dev-110 fwupd[6810]: 11:22:56:0328 FuUsbDevice          failed to load manufacturer string for usb device 2:4: USB error on device 17ef:a393 : Resource busy [-6]
Feb 12 12:22:56 dev-110 fwupd[6810]: 11:22:56:0328 FuUsbDevice          failed to load product string for usb device 2:4: USB error on device 17ef:a393 : Resource busy [-6]
```

You can see it's not until `Feb 12 12:23:20`.

AnimiVulpis, 2021-02-12:
Thank you for the fast answer.
Could be I guess. But why is this device only timing out on rare occasions?
Or to ask the question a bit differently:
- Can you give me some pointers on how to find out which device is causing this sometimes?
- What could `Resource busy [-6]` mean?
- Is `usb device 2:4` the same as `usb 1-4.3.3.1: USB disconnect, device number 9` (3 lines up)?

Does this help in any way?

```shell
$ sudo lsusb -s 2:4 -v
Bus 002 Device 004: ID 17ef:a393 Lenovo USB3.1 Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x17ef Lenovo
  idProduct          0xa393
  bcdDevice            c.d3
  iManufacturer           1 VIA Labs, Inc.
  iProduct                2 USB3.1 Hub
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes           19
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Feedback
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               8
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x000d
    Per-port power switching
    Compound device
    Per-port overcurrent protection
  bPwrOn2PwrGood      175 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.4 micro seconds
  wHubDelay          2292 nano seconds
  DeviceRemovable    0x08
 Hub Port Status:
   Port 1: 0000.02a0 lowspeed L1
   Port 2: 0000.02a0 lowspeed L1
   Port 3: 0000.02a0 lowspeed L1
   Port 4: 0000.02a0 lowspeed L1
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x002a
  bNumDeviceCaps          3
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat           4 micro seconds
    bU2DevExitLat         231 micro seconds
  Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {30eef35c-07d5-2549-b001-802d79434c30}
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0005
  Self Powered
  U1 Enabled
```

Is there any way to verify that this is causing it? If I get the time I will try to produce the timeout without the hub attached
Any hopes of making `fwupd` more robust against crashing in these cases?

superm1, 2021-02-12:
>Is there any way to verify that this is causing it? 

It's an external hub right?  Try a bunch of times to start and stop the service and confirm it doesn't happen with it not plugged in.

>Any hopes of making fwupd more robust against crashing in these cases?

From what it look like, I don't think it's crashing, but rather it's just that it gets stuck during startup waiting for that device.

AnimiVulpis, 2021-02-17:
Is there a way to handle this timeout more gracefully?
So without the need to `systemctl restart fwupd.service`?

I wasn't yet able to produce a timeout by stopping and starting the service. Seems like if it is happening it always happens on the first start of any boot or not at all (which makes it much harder to investigate :disappointed: )

superm1, 2021-02-17:
Since fwupd is (intentionally) single threaded, we're at the whim of enumerating all the devices before the daemon is ready to accept dbus connections.

@hughsie thoughts on this?  I wonder if we can help this kind of problem by some changes to how libgusb handles stuff getting stuck and letting it give more cycles back into the mainloop?

hughsie, 2021-02-17:
> whim of enumerating all the devices before the daemon is ready to accept dbus connections

I think this is more than enumeration, it seems to be setup() or even probe(). Looking for that VIDPID I see:

    ../plugins/vli/vli-usbhub-lenovo.quirk:[DeviceInstanceId=USB\VID_17EF&PID_A393]

which seems to correlate with a Lenovo Gen2 dock -- is that correct?

> how libgusb handles stuff getting stuck 

I guess the timeout we use for enumeration should be a lot less; it's tricky as if we go through other hubs we need the extra timeout time. How many seconds are we talking about?

> and letting it give more cycles back into the mainloop?

I think it's either 100% sync, async, or threaded. Each seems to have drawbacks. I have been pondering threading for enumeration for some time, maybe we do all hardware access in one thread and all daemon dbus stuff in another. I'm not super keen on putting all coldplug in threads as then it all gets a bit complicated with deps between plugins.

superm1, 2021-02-17:
>maybe we do all hardware access in one thread and all daemon dbus stuff in another

That is what I lean on at the moment.  

>I'm not super keen on putting all coldplug in threads as then it all gets a bit complicated with deps between plugins.

So something that we can think about is a flag for individual plugins to indicate they're done with coldplug, and block any use of a plugin until coldplug is complete.

Something else to consider while we talk about threading possibilities is plugin and daemon permissions which we were talking about offline.  @jlucangelio I think you were thinking about some of that, anything you can share?

AnimiVulpis, 2021-02-17:
> which seems to correlate with a Lenovo Gen2 dock -- is that correct?

Correct: On the underside it's labeled as

```
Lenovo
Thinkpad USB-C Dock Gen2
[...] TYPE: 40AS
```

Listed also [here](https://support.lenovo.com/de/en/solutions/migr-4tpjf4) (with drivers for Windows 10, 8.1, 8, 7)

Should this issue now be closed? WDYT?

hughsie, 2021-02-22:
I've just tested that exact hardware, although with version 13.23 on the VL820Q8 -- it enumerated fine. Would it be possible to update the hub in Windows and then to try again to list the device in Linux? You have 12.211 installed. Thanks!

AnimiVulpis, 2021-02-24:
@hughsie Thank you for that recommendation.
I will try to update the firmware of the hub. Didn't realize that the firmware for the dock can be updated.

Regarding the version number `13.23` where or how would I see that? (output of `lsusb`, `fwupdmgr get-devices`?)
[This](https://support.lenovo.com/us/en/solutions/ACC500106) website only lists `V1.0.6.06161` as the firmware version.

I will try to update and check the versions again

hughsie, 2021-02-24:
> fwupdmgr get-devices

Normally here, although if the device doesn;t enumerate it might be tricky. For this you can look at:

    bcdDevice            c.d3

and just convert from hex to dec. I don't think the Lenovo update lists the versions of the individual components unfortunately.

hughsie, 2021-04-11:
Please re-open if this issue still happens with the new firmware. Thanks!

closed 2021-04-11 by hughsie