fwupd/fwupd #2894: Arch container has been failing to generate

superm1, 2021-02-17:
@FFY00 i've been getting nightly emails about the arch container failing to build.  I thought it was just transient network connectivity issues at first but it seems more sinister.

Can you take a look?

https://github.com/fwupd/fwupd/runs/1914948712

FFY00, 2021-02-17:
Yes, this is a known issue. Essentially, it is an incompatibility in glibc, which IIRC was using syscalls only available on newer kernels, making it impossible to run an arch docker on older distros. This has been fixed in `glibc 2.33-4`, which has been out of testing for a few days, we just need to wait for the docker container to be rebuilt so that it will ship with the fixed glibc.

It seems the docker image was updated 11h ago, which *should* fix this. Can you confirm?

https://bugs.archlinux.org/task/69563
https://github.com/opencontainers/runc/issues/2151

superm1, 2021-02-17:
Ah thanks!  We'll know in about 12 hours when the containers try to generate again.

superm1, 2021-02-18:
Nope, seems it still failed.  Here is the current failure log:
https://github.com/fwupd/fwupd/runs/1923428179?check_suite_focus=true

FFY00, 2021-02-18:
Then it seems like the host is running an old kernel(?), the new glibc should be able to handle up to Linux 4.4. The actual issue is runc returning EPERM instead of ENOSYS when the syscall is not supported, so a newer runc would also solve the problem. See https://github.com/actions/virtual-environments/issues/2658#issuecomment-781141476.

Perhaps an approach like https://github.com/actions/virtual-environments/issues/2698#issuecomment-779262068 would work to fix the building? But we might still have issues running the container, as I understand, we can't follow the glibc path that triggers the issue.

superm1, 2021-02-19:
Certainly open to trying it if you want to!
We can see the last success generation was between 13 and 14 days ago. Is it helpful to track back exact what changed from that?

PR #2922 (2021-02-22): trivial: ci: add a workaround for arch container generation issue
This should be dropped at some point hopefully when the new
docker release has landed.
See: https://github.com/actions/virtual-environments/issues/2658
Fixes: #2894

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 2264a2ef (2021-02-22): trivial: ci: add a workaround for arch container generation issue

This should be dropped at some point hopefully when the new
docker release has landed.
See: https://github.com/actions/virtual-environments/issues/2658
Fixes: #2894
closed 2021-02-22 by superm1