fwupd/fwupd #2900: EFI Capsule On Disk Support

jackpot51, 2021-02-18:
Capsule On Disk is defined in the UEFI Specification, section 8.5.5, Delivery of Capsules via file on Mass Storage device. It is being used on many new designs, such that UpdateCapsule often is a wrapper around it. This is due to memory persistence issues when unlocking the flash and resetting.

A new OsIndications bit is defined in the specification:

```
#define EFI_OS_INDICATIONS_FILE_CAPSULE_DELIVERY_SUPPORTED
0x0000000000000004
```

The OS does the following:
- Check the bit in OsIndicationsSupported, use UpdateCapsule if that is supported instead
- Put signed capsule in \EFI\UpdateCapsule\(anything)
- Set the bit in OsIndications
- Reboot

The firmware then takes over and does the following:
- Detects the bit in OsIndications, unlocks, and resets
- Processes the capsules in case ignored alphabetic order
- Sets Capsule0000, Capsule0001, etc. to an EFI_STATUS for each capsule as it finishes
- Deletes the capsule when it is finished
- Reboots to the OS

The OS can now check the Capsule0000 and so on variables for the result of firmware flashing

superm1, 2021-02-18:
So in this case, we actually wouldn't use `fwupdx64.efi` for updates in your mind of that bit is supported?

jackpot51, 2021-02-18:
Right, and it would reduce the number of reboots by one on supported systems. The reason to support this, for me at least, is we are planning to have System76 open firmware products use capsule updates, but do not want to include the extra code to support UpdateCapsule when it would just be a wrapper around capsule on disk.

superm1, 2021-02-18:
Sounds pretty sensible to me.  Some implementation suggestions for when you submit a PR for this:
1. The spec says it's up to the OS to decide what to name the files on the ESP, so I would say we want the prefix of `fwupd` included in the file name going to to the ESP.
2. I think we'll want to have a boolean in the device attribute to indicate UEFI devices are picking this (IE here: https://github.com/fwupd/fwupd/blob/master/plugins/uefi-capsule/fu-uefi-device.c#L39)
3. You'll want to include debugging in the `to_string` here: https://github.com/fwupd/fwupd/blob/master/plugins/uefi-capsule/fu-uefi-device.c#L126 to indicate this is being used

martinezjavier, 2021-05-20:
another use for this could be to update the U-boot firmware on aarch64 machines that don't have an EFI firmware but that use the U-boot's EFI layer to be [EBBR]( https://arm-software.github.io/ebbr/) complaint.

Right now users have to manually write the U-boot binaries into some internal memory (i.e: SPI flash), for example as explained in @nullr0ute's blog post about [Fedora on the Pinebook Pro](https://nullr0ute.com/2021/05/fedora-on-the-pinebook-pro/).

It would be great if we could use `fwupd` to update the U-boot on these machines, and by reading [this doc](https://source.denx.de/u-boot/u-boot/-/raw/master/doc/board/emulation/qemu_capsule_update.rst)  it seems U-boot already supports Capsule-On-Disk.

In this case `fwupdaa64.efi` won't be used either as @superm1 said, since it's U-boot itself the component that would read the capsule and write it to the SPI flash.

rathann, 2021-05-20:
That would be awesome.

apalos, 2021-06-16:
> another use for this could be to update the U-boot firmware on aarch64 machines that don't have an EFI firmware but that use the U-boot's EFI layer to be [EBBR](https://arm-software.github.io/ebbr/) complaint.

Just FYI all boards that aim for EBBR compliance *must* support EFI

apalos, 2021-06-16:
I'll have a go at it.  The current upstream U-Boot has a broken SMBIOS generation and the needed FMPs are not installed properly.  I've sent patches fixing both, but the main problem is that U-Boot does not support SetVariable at runtime. 

However we do delete the capsule once we apply it, so we might consider just ingoring OsIndications for now

apalos, 2021-07-17:
FYI, all of U-Boot's issues have been addressed and this branch implements capsuleupdate on-disk
https://github.com/fwupd/fwupd/tree/wip/apalos%2Fcapsule-on-disk

closed 2021-07-28 by hughsie