fwupd/fwupd #2901: fwupd[2801]: 17:47:55:0257 FuEngine emmc failed to change udev device

tracing-home, 2021-02-18:
This error is repeated about every 50 seconds in /var/log/syslog

Maybe it is just cosmetic, but as a syslog filler not ideal. By all appearance it tries to register the USB controller as a disk. In the Gnome Desktop Utility "Disks" it shows up as well, but nothing can be done with/about it.

System: Ubuntu 20.04
Hardware: Zotac Zbox CI547 nano

I am not exactly sure when this started. A month or two ago nothing appeared in syslog or the Disks app.  

A suspicion: A few weeks ago I used dd to create a bootable USB stick. As target I selected a USB stick that was not inserted at that time. Where that data went, I have no idea. This is nothing like it in /dev 

Other than that, a good few updates were done in the meantime.

I'm happy to try whatever might be needed. 

Thanks! Martin

Here the error in full:

```
Feb 18 19:20:59 smallbox fwupd[2801]: 18:20:59:0590 FuEngine             emmc failed to change udev device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/host1/target1:0:0/1:0:0:0/block/sdb: cannot ensure ID: FuUdevDevice:
Feb 18 19:20:59 smallbox fwupd[2801]: SD MMC MS PRO
Feb 18 19:20:59 smallbox fwupd[2801]:   Serial:               20120926571200000
Feb 18 19:20:59 smallbox fwupd[2801]:   Flags:                none
Feb 18 19:20:59 smallbox fwupd[2801]:   Vendor:               Generic-
Feb 18 19:20:59 smallbox fwupd[2801]:   Version:              1.00
```
If an USB stick (or similar) is inserted at boot time, it would amount to ../sdc

```
fwupd/focal 1.4.5-1pop0~1611677200~20.04~aaada71 amd64 [upgradable from: 1.3.11-1~focal1]
fwupd/focal-updates,now 1.3.11-1~focal1 amd64 [**installed**,upgradable to: 1.4.5-1pop0~1611677200~20.04~aaada71]
```
I did the update, downgraded again (rebooting in between), no change.

```
$ lspci
..
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
..

$ lspci -v
....
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21) (prog-if 30 [XHCI])
        Subsystem: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller
        Flags: bus master, medium devsel, latency 0, IRQ 126
        Memory at df230000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: <access denied>
        Kernel driver in use: xhci_hcd
        Kernel modules: xhci_pci

00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
        Subsystem: Intel Corporation Sunrise Point-LP Thermal subsystem
        Flags: fast devsel, IRQ 18
        Memory at df24d000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: <access denied>
        Kernel driver in use: intel_pch_thermal
        Kernel modules: intel_pch_thermal
...

$ fwupdmgr update
• Samsung SSD 860 EVO 1TB has no available firmware updates
No updatable devices
```
So the controller is picked up as a drive.

hughsie, 2021-02-20:
Can you attach the full output of `sudo fwupdtool --plugins emmc --verbose get-devices` please -- thanks.

tracing-home, 2021-02-20:
Sure, here it is. However the --plugins parameter does not exist in my version

sudo fwupdtool --version
client version: 1.4.5
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37

I attached the output of three other runs, I can provide more if helpful. I did notice in the get-details emmc an "failed to open emmc" at the end. Though I have no real idea about the relevance, as the .so is loaded well before.

sudo fwupdtool --plugins emmc --verbose get-devices
sudo fwupdtool --verbose get-devices
sudo fwupdtool --verbose get-details emmc

[output-get-devices.txt](https://github.com/fwupd/fwupd/files/6015855/output-get-devices.txt)
[output-emmc.txt](https://github.com/fwupd/fwupd/files/6015856/output-emmc.txt)
[output-hwids.txt](https://github.com/fwupd/fwupd/files/6015857/output-hwids.txt)

hughsie, 2021-02-21:
This was fixed in fwupd 1.5.0 where we stopped unconditionally doing ->setup() and ->rescan() on each changed event. If you can reproduce this issue with fwupd 1.5.x then please reopen. Thanks.

closed 2021-02-21 by hughsie